Marus/cortex-debug #655: Could not find symbol '_SEGGER_RTT' in executable - IAR specific?

TorBlackbeard, 2022-05-17:
Cortex-debug says "Could not find symbol '_SEGGER_RTT' in executable", when using  "address": "auto" configuration.

When using "address": "0x200019a8" it works fine (- but it changes all the time...)

Output from objdump is:
200019a8 g O P2 zi 000000a8 _SEGGER_RTT

Could it be the regex that should find this line fails, because it is IAR specific?

(we're on the 'prerelease' channel of cortex-debug)

haneefdm, 2022-05-17:
Absolutely, the regex is the key. But which version of the objdump are you using? Is it still `arm-none-eabi-objdump` (or any GNU) though. It should work. Whose objdump are you using.

I don't think that line matches anything...The `P2` is problematic. I would use IAR compiler/linker and still use GNU tools for the redt

https://linux.die.net/man/1/objdump

The objdump output is pretty strict and every char. position matters, even a tab

haneefdm, 2022-06-17:
It's been a month and no response/progress. At least an executable has to be provided to move forward.

Also please try (pre)release 1.5.1 already on the Marketplace

closed 2022-06-17 by haneefdm