Marus/cortex-debug #657: Cortex-Debug ends up in a strange state where no commands can be issued

turbofish, 2022-05-18:
Hi,

First, thanks for a great project.

I'm debugging a custom RTOS on an STM32F072 and have managed to force the extension to somehow lose the interactions with gdb.

First, launch.json:

>  {
    "version": "0.2.0",
    "configurations": [
        {
            "name": "QIO",
            "cwd": "${workspaceFolder}",
            "executable": "./builddir/debug/src/master/qio_master_no_boot.elf",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "stlink",
            "gdbPath": "/usr/bin/gdb-multiarch",
            "svdFile": "./doc/STM32F0x2.svd",
            "showDevDebugOutput" : "vscode",
            "showDevDebugTimestamps" :true,
            "preRestartCommands": [ ]
        },
    ]
}

Starting a debug session, stepping, breakpoints etc works fine, but when running and pausing, and then trying to issue a gdb command in the debug console I get the following error:

> 0000076865: Thread Warning: repl: eval. expression 'l' with no thread context. Using default 

After that warning, no commands can be issued, even invalid ones; for example:

> n
From client: evaluate({"expression":"n","context":"repl"})
n
From client: evaluate({"expression":"n","context":"repl"})
fnord
From client: evaluate({"expression":"fnord","context":"repl"})
hello world
From client: evaluate({"expression":"hello world","context":"repl"})

Not even restart works at this point:

> From client: pause({"threadId":1})
From client: reset-device("reset")

The code itself might be a hint. It's a round robin style RTOS. If I create 10 threads and just flip a few pins back and forth, 99.9% of the CPU time will be moving stack pointers around and switching tasks. Then the issue arises right away, If the tasks actually does something it takes significantly longer time. I think the issues come if I pause during a task switch.

I'm not an expert on GDB, but I think the code makes gdb confuse the frame since the odds of me pausing during a switch is very high. I searched the source of the extension and found the following lines in gdb.ts, (3146-3158) which is the lines giving me the first warning:

>// Spec says if 'frameId' is specified, evaluate in the scope specified or in the global scope. Well,
                // we don't have a way to specify global scope ... use current thread then.
                let threadId = this.currentThreadId;
                let frameId = 0;
                if (args.frameId) {     // Should always be valid
                    [threadId, frameId] = GDBDebugSession.decodeReference(args.frameId);
                    if (traceThreads) {
                        this.handleMsg('log', `**** evaluateRequest: ${args.context} '${args.expression}' in thread#${threadId} frame#${frameId}\n`);
                    }
                } else if (!isBusy()) {
                    // In practice, never seen this unless it comes from a custom request
                    this.handleMsg('log', `Thread Warning: ${args.context}: eval. expression '${args.expression}' with no thread context. Using default\n`);
                }

The only way to continue debugging at this point is to end the session and restart.

Is there a reason to not forward the commands to GDB if there is no valid frameid?

Attaching a full debug, there is a two-three pauses and continues before the problem is triggered.

[debugoutput.txt](https://github.com/Marus/cortex-debug/files/8714235/debugoutput.txt)

haneefdm, 2022-05-18:
It is not clear to me what is happening. We always send commands from the REPL window of Debug Console to gdb. Only difference is when debugger is paused, we respect the current thread (we have to) as presented to us by VSCode.

https://github.com/Marus/cortex-debug/blob/4f68b196d02529b5a324dec260d210e2c4276d8f/src/gdb.ts#L3223

Btw, the code you are seeing is not what was released. It has changed since., but not functionally. In your case, follow the logic for context === 'repl'

The reason you are not seeing anything is because we are still waiting for a response for this command

```
0000066443: 58-stack-info-depth --thread 1 10000
```

GDB is hung. We don't send any more commands until the previous one finished. GDB does not behave well if there are no threads. It is not us. We (and VSCode) are fine with zero threads. GDB isn't

haneefdm, 2022-05-18:
Also, the message
```
0000076865: Thread Warning: repl: eval. expression 'l' with no thread context. Using default

```
is totally harmless. `Using default` meaning that whatever gdb thinks is the current thread. Which could be anything depending on the last command executed (there could be hundreds in the background). As a user, you have no control over what the current thread is.

haneefdm, 2022-05-18:
We should focus on why gdb is hung.

If you are writing your own RTOS, are you following the ARM ABI for how stack(s) is managed?

haneefdm, 2022-06-17:
No response, not even sure what the issue is. Closing. Also, we reduced the stack depth query to 1000 for future releases.

closed 2022-06-17 by haneefdm