Marus/cortex-debug #658: Can't Debug if Flash address is different than 0x08000000 in linker script.

burakenez, 2022-05-20:
If i built the elf file with specified memory areas in linker script as below i am able debug the project:
`MEMORY
{
FLASH (rx)                  : ORIGIN = 0x08000000, LENGTH = 484k
RAM1 (xrw)                : ORIGIN = 0x20000008, LENGTH = 0x2FFF8
RAM_SHARED (xrw)   : ORIGIN = 0x20030000, LENGTH = 10K
}`

if i change starting address of flash in linker script as below, i can't debug the project:
`MEMORY
{
FLASH (rx)                  : ORIGIN = 0x08007000, LENGTH = 484k
RAM1 (xrw)                : ORIGIN = 0x20000008, LENGTH = 0x2FFF8
RAM_SHARED (xrw)   : ORIGIN = 0x20030000, LENGTH = 10K
}`

I have been searching for a solution and i couldn't find any. I tried openocd, stutil, stlink servertypes. Each has the same problem. I think there is a configuration for this but because of lack of examples and documentation i failed to find it. My launch config as follows:
`"name": "Debug with ST-Link",
"cwd": "${workspaceFolder}",
"executable": "./Build/Debug/Application/Application.elf",
"request": "launch",
"type": "cortex-debug",
"runToEntryPoint": "main",
"showDevDebugOutput": "none",
"servertype": "stlink",
"device": "STM32WB55RG",
"svdFile": "../../STM32WB55_CM4.svd",
"interface": "swd",
"armToolchainPath": "/opt/gcc-arm-none-eabi/bin",
"stlinkPath": "/opt/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.0.200.202202231230/tools/bin/ST-LINK_gdbserver",
"stm32cubeprogrammer": "/usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin"`

PhilippHaefele, 2022-05-20:
Hi,

This is due to the fact that your STM32WB55RG looks into 0x8000000 for the initial Stack pointer vale and reset vector. If you move your binary there isn't anything valid at this address.

I managed to get a similar case (Code in RAM) by setting the SP and PC via override/after elements. There are maybe better ways of doing it, but this works.

I can lookup the actual commands/launch config if you like?

Best regards
Philipp

haneefdm, 2022-05-20:
FYI: This has nothing to do with us. We don't even look at your memory map. Everything is handed to gdb+openocd and they take care of it and just give us back stack traces if/when program halts

Since you are working with an Cortex-M ST device, it is really their manual you have to follow on how to write code that is not at the default place. All ARM cores start at a fixed Reset-Vector-Table (VT) address. VT address is generally decided by ST (default ARM is 0x0). See

https://developer.arm.com/documentation/dui0497/a/the-cortex-m0-processor/exception-model/vector-table

There are registers you can write to relocate the VT

burakenez, 2022-05-20:
> Hi,
>
> This is due to the fact that your STM32WB55RG looks into 0x8000000 for the initial Stack pointer vale and reset vector. If you move your binary there isn't anything valid at this address.
>
> I [...]

It would be perfect! Thank you.

haneefdm, 2022-05-20:
@PhilippHaefele If you are manually setting the SP/PC, you should also make sure the LR (R14) is 0xFFFFFFFF or you will get corrupt stack traces, gdb crashes, etc. Right after reset, the LR will be correct, but it may not have that reset value if a programming algorithm preceded the fake reset for instance. Definitely something to ponder for a restart.

The SCB->VTOR is a better way I think but it requires a tiny shim at the default boot address. This is because I like the code to execute the same with or without the debugger

To do what @PhilippHaefele suggested, you would use the `overrideLaunchCommands`(also see other`overrideXXXCommands`). For a given gdb-server (OpenOCD vs STLink), see what Cortex-Debug does -- slightly different commands for different gdb-servers but the goal is to
1. Halt and reset the device
2. Program the device
3. Reset and halt at the reset-handler

Specifically, for STLink, we do
```
monitor reset
-target-download
monitor reset
monitor halt
```
Not sure that last line was needed but....you would replace those last two lines with SP/PC/LR overrides as in `set $pc=0x8000XXXX` (use gdb syntax, mine may not be right)

See https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

haneefdm, 2022-05-20:
You can keep posting but, I will close this issue.

closed 2022-05-20 by haneefdm

burakenez, 2022-05-20:
> gdb syntax

thank you so much. i will try this as well. I am also not familiar with gdb syntax. Firtly, i need to get into it.

I also found another work around by altering linker script. Someone else may have same problem. As you mentioned vector table, i put isr vector table to 0x08000000. Following changes needs to be done. Here my solution:

Step 1 - Add ISR_VECTOR to memory:

```
MEMORY
{
ISR_VECTOR (rx)            : ORIGIN = 0x08000000, LENGTH = 0x200
FLASH (rx)                 : ORIGIN = 0x08007000, LENGTH = 484k
RAM1 (xrw)                 : ORIGIN = 0x20000008, LENGTH = 0x2FFF8
RAM_SHARED (xrw)           : ORIGIN = 0x20030000, LENGTH = 10K
}
```

Step 2 - Change position of .isr_vector in SECTIONS from FLASH to ISR_VECTOR

```
SECTIONS
{
  .isr_vector :
  {
    . = ALIGN(8);
    KEEP(*(.isr_vector)) /* Startup code */
    . = ALIGN(8);
  } >ISR_VECTOR

// some other scripts
}
```

haneefdm, 2022-05-20:
Yes, the device will like it because you are using the normal boot process. Many other ways. As you feel the need for bootloaders, you will learn a lot more stuff.