Marus/cortex-debug #659: Extension freezes at "1-gdb-set target-async on"

aviler, 2022-05-21:
Hi, I am having this issue when I hit `Start Debugging` the debug panel shows up and quickly disappears. 
On the `Debug Console`, it freezes after launching. Here's the log:
```
Reading symbols from /opt/gnuarmemb/bin/arm-none-eabi-objdump --syms -C -h -w /home/aviler/asdf/build/zephyr/zephyr.elf
Reading symbols from /opt/gnuarmemb/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/aviler/asdf/build/zephyr/zephyr.elf
Launching GDB: /opt/gnuarmemb/bin/arm-none-eabi-gdb -q --interpreter=mi2 /home/aviler/asdf/build/zephyr/zephyr.elf
1-gdb-set target-async on
```
If I run the commands on my terminal I can see the output of both symbol readings and gdb is also running ok, so not missing `libncurses5`.
Here's my `launch.json`:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug arm-console",
            "cwd": "${workspaceRoot}",
            "executable": "../build/zephyr/zephyr.elf",
            "request": "launch",
            "type": "cortex-debug",
            "serverpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe",
            "servertype": "jlink",
            "device": "nRF5340_xxAA_APP",
            "runToEntryPoint": "main",
            "interface": "swd",
            "armToolchainPath": "/opt/gnuarmemb/bin/",
            "rtos": "Zephyr",
            "svdFile": "../modules/hal/nordic/nrfx/mdk/nrf5340_application.svd",
            "showDevDebugOutput": "raw"
        }
    ]
}
```
As you can see the project is an `nRF5340` one. I'm using `JLink_Linux_V758b_x86_64` and `gcc-arm-none-eabi-9-2019-q4-major-x86_64-linux` which is the recommended version from Nordic documentation. Cortex-Debug version 1.4.4. 
I already tried to downgrade JLink to the version I am using on another laptop with no success.

What am I missing or what else I could check to find a solution? Thanks

haneefdm, 2022-05-21:
Can you run the same gdb from shell and enter the same as what we sent gdb
```
-gdb-set target-async on
```
Maybe it is not gdb. Are able to run the other commands (objdump and nm) as well?

Also, can you try the latest temporarily?
https://github.com/Marus/cortex-debug/releases

aviler, 2022-05-21:
Hi @haneefdm yeah did that. I was in doubt if you had made a typo so I did `1-gdb-set target-async on` and `-gdb-set target-async on` for both commands I got a `1^done` as a response. Was also able to run `objdump` and `nm` with no problems.

Just tried the pre-release `1.5.1-pre5` and got the same situation but this time is a different gdb command `1-gdb-version` probably added as first command in the new release.
Very weird right?

haneefdm, 2022-05-21:
Hmmm. I am on the road so can't spend much time. But yes, 1.5.1 will query the version first and then decide to set the async (old way for old gdb and new way for new gdb) -- so I have to query gdb version first.

Were you able to run objdump and nm the way we are running it? How big was the output of each?

aviler, 2022-05-21:
I ran the exact same commands as the extension says in the `Debug Console`. For `objdump` I got 246.5kB and `nm` got 335.kB both values pretty close to the output I have on another machine where I can debug normally.

aviler, 2022-05-22:
This is awkward but a good lesson. `So, the issue was that I was running VSCode from an unofficial flatpak version` which in turn was running from inside a container and therefore not able to access SDKs on host system. When you open this version of VSCode it automatically opens a file flatpak-warning.txt with more details, it has some workarounds. I've tried to change settings.json with no success. So I'll stick with the official VSCode installation.
I am closing the issue, thank you very much @haneefdm!

closed 2022-05-22 by aviler

tomkulaga, 2022-05-27:
I had a similar issue I was using WSL 2, and turns out i didn't have libncurses5 installed. It would silently fail in VSCode but command line alerted me to the missing .so file