Marus/cortex-debug #660: What command does `cortext-debug` send when you hit Continue?

alxpettit, 2022-05-24:
I have been working on getting `cortex-debug` to cooperate with Rust & RP2040 toolchain, and I've experienced a weird problem with `continue` appearing to have an inconsistent effect depending on where I send it from.

When I press F9, or press the play icon button, the code runs from main onward as normal. However, when I try to continue it by typing `continue` into the debug console, it enters an invalid state of some sort and becomes nonresponsive to attempts to pause the program.

For context, you should note that the RP2040 has a somewhat complicated UF2 bootloader, and the debugger requires some workarounds to skip through to the `main()` function properly.

So, my question is, what would I have to type into the Debug Console to get the same effect as hitting the play button? Or, is this a bug?

haneefdm, 2022-05-24:
You can see the exact commands sent to gdb yourself. Instructions should be in the Debug Console to enable debugging of the debugger itself. To answer your question, Continue is literally `continue` but we use GDB's machine interface which needs the command `-exec-continue`.

But whatever is legal for gdb, you can type in the Debug Console. It is sent directly to gdb. There are no special commands. It is GDB all the way.

But, before you do anything, I would try our latest pre-release. We had some issues with the startup code and VSCode issuing commands

https://github.com/Marus/cortex-debug/releases

We have several people using Rust + gdb + cortex-debug.

alxpettit, 2022-05-25:
> If you want to know how our debugger internal works (will put you to sleep)

I'm actually quite interested in that! :D

So, I tried the latest Github prerelease (`1.5.1-pre5`), as you suggested, and I seem to be experiencing the same oddity.

When I attach to my RP2040 board, and it pauses at `main`, I can hit play/pause or press keyboard buttons and step through the program as I see fit.

_But_, if I type `continue` into the `DEBUG CONSOLE` pane, or tell Cortex-Debug to send it automatically via `postLaunchCommands`, it just seems to enter some kind of broken state where the pause GUI button no longer functions like it's supposed to, and the debugger state displayed in the editor doesn't seem to match the actual state of GDB.

Because of this, I've been forced to hit F5 _and then_ also F9 every time I change my program, and any attempt to script automatic continuing just breaks the debugging tools offered by Cortex-Debug. I imagine I should open issue about that.

haneefdm, 2022-06-02:
In the debug console, there are instructions on how to enable more verbose logging and interactions with gdb. You are not giving us any info. I am going to close this issue because you are not giving me any useful info.

I can reopen when you give me more info to look at.

closed 2022-06-02 by haneefdm

alxpettit, 2022-06-02:
You didn't ask for logs. :P

I was asking a question to clarify my presuppositions about the operation of your program. This is a questions issue, not a bug report issue -- hence the title of my issue. If you want, I can _open_ a bug report issue.