Marus/cortex-debug #661: Some first observations and questions

basprins, 2022-05-26:
Hi,
Using this extension for a week or so now and I have some questions.

First of all, debugging my target (imx rt 1024) with a jlink probe seems somewhat unstable. When I start the debugger (F5) after some time the debugger stops in my main. This is not "super fast". It's also not terribly slow, but there seems some fair amount of dead time which is annoying. I observe this on both ubuntu and windows. Although ubuntu seems to be slower here, for some reason it takes longer in ubuntu to synchronize the symbols. Is this a known issue? Will it be improved is there little you can do about it?

Another thing is when I try to restart my debug session. vscode has a green "power" button which allows me to restart the debug session. Sometimes it fails and it's game over. Something breaks, which results in killing the debug session, and pressing F5 again. 

Also when I am restarting debug sessions over and over again, at some point things get really slow (20seconds before vscode shows my next debug statement). Again, I observe this both on windows and ubuntu.

I don't know if it helps to share the debug console output. I can't make much sense out of it, I guess you guys can :). If you can tell me which of these issues you recognize and which you would like to have more information about, I will share whatever you need.

PS: I read in another ticket that somebody pointed out the somewhat long time before symbols are loaded/synchronized in Cortex-Debug, and that he was asked to share the elf file. I don't know what exactly you need, but I can share my elf file. I guess you also need the code, that might be another story :). But if you can magically do something useful with the elf file I'll happily share that at least.

I hope you recognize some of these issues. More than open for workarounds, or if you tell me to upgrade to some alpha/beta version: say the word.

haneefdm, 2022-05-26:
Could you try the latest?

https://github.com/Marus/cortex-debug/releases

Let me know what you observe and we can go from there. Yes, I can work with just the elf file if you are willing to share.

There is a 'Reset' (belongs to us) button and a 'Restart' (belongs to VSCode). VSCode has been changing what this button does in the past. The Reset button however is just a halt and a reset. Should be super fast -- no program, no re-build, etc. However, it depends if you have a very slow bootloader. For some devices, the program step (on a launch) can take very long but that is between your device and JLink, -- we just wait until it is done.

Enable debug (instructions in the Debug Console) and you can see exactly what is taking time.

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

[image]

[This Wiki could also help](https://github.com/Marus/cortex-debug/wiki)

haneefdm, 2022-05-26:
In the command palette of VSCode, start typing cortex-debug and you will see more commands (including memory window)

basprins, 2022-05-27:
DUDE! 1.5.0 (didn't bother to manually install 1.5.1 yet) is a HUGE improvement. The annoying "synchronizing symbols" delay is completely gone. I will go with this for a while. If there is a good reason to manually update to 1.5.1 I'll do that, but maybe you already released the new stable in the mean time. Many thx!

haneefdm, 2022-05-27:
btw, that was a bug in newer versions of gdb that caused that delay. We had to work around it. Gdb would even run out of whole system memory and crash

closed 2022-05-27 by haneefdm