Marus/cortex-debug #662: Unable to launch second GDB debug session with "chainedConfigurations"

Nekspire, 2022-05-31:
Hello,
I'm trying to debug multicore microcontroller STM32H745ZIT6.  What i would like to achive is to have a separate windows for simultaneously debugging core M7 and core M4.  Right now i'm only able to flash corectlly both of the cores and debug core M7,
but unfortunately i have an issue with debug core M4. 

The debug console says that:

**Failed to launch GDB: localhost:50001: Connection timed out. (from target-select extended-remote localhost:50001)**

launch.json

    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug OpenOCD CM7",
            "servertype": "openocd",
            "gdbPath": "arm-none-eabi-gdb.exe",
            "cwd": "${workspaceRoot}",
            "executable": "./build_cm7/stm32h745zit6_template_cm7.elf",
            "loadFiles": [
                "./build_cm7/stm32h745zit6_template_cm7.elf",
                "./build_cm4/stm32h745zit6_template_cm4.elf"
            ],
            "interface": "swd",
            "device": "STM32H745ZI",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32h7x.cfg"
            ],
            "svdFile": "subprojects/mxgen/STM32H745_CM7.svd",

            "numberOfProcessors": 2,
            "targetProcessor": 0,
            "chainedConfigurations": {
                "enabled": true,
                "waitOnEvent": "postInit",
                "lifecycleManagedByParent": true,
                "launches": [                          
                    {
                        "name": "Debug OpenOCD CM4",
                        "folder": "${workspaceRoot}"
                    }
                ]
            },
        },

        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug OpenOCD CM4",
            "servertype": "openocd",
            "gdbPath": "arm-none-eabi-gdb.exe",
            "cwd": "${workspaceRoot}",
            "executable": "./build_cm4/stm32h745zit6_template_cm4.elf",
            "interface": "swd",
            "device": "STM32H745ZI",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32h7x.cfg"
            ],
            "svdFile": "subprojects/mxgen/STM32H745_CM4.svd",

            "numberOfProcessors": 2,
            "targetProcessor": 1,
        },
    ]

haneefdm, 2022-05-31:
The whole chained configurations and even VSCode's own Launch Groups are for doing it all in one Windows. If you have separate windows, you have to start two separate VSCodes (in separate directories).

Also, look in the OpenOCD output window (gdb-server terminal window) and see what it did. Did it start two servers on 50000 and 50001 respectively? If not something is wrong. Maybe ST doesn't not support this. This part is not up to us. But please attach the output of OpenOCD regardless. 

Finally, If you use chained configs, try doing an `"request": "attach"` instead. But first, we gotta establish connection between gdb and openocd which is what is failing for you.

Nekspire, 2022-06-01:
Hello @haneefdm, thanks for respond. 

> The whole chained configurations and even VSCode's own Launch Groups are for doing it all in one Windows. If you have separate windows, you have to start two separate VSCodes (in separate directories).

Yeah i wanted say that i would like to have just an option do debug both of the cores simultaneously (not in saparate windows), so thats why i'm trying to use chained configuration.

> Also, look in the OpenOCD output window (gdb-server terminal window) and see what it did. Did it start two servers on 50000 and 50001 respectively? If not something is wrong. Maybe ST doesn't not support this. This part is not up to us. But please attach the output of OpenOCD regardless.

The OpenOCD output says that:

```
[2022-06-01T15:51:42.277Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
openocd.exe -c "gdb_port 50000" -c "tcl_port 50002" -c "telnet_port 50004" -s "D:\\programming\\vscode\\embedded\\stm32h745zit6_template" -f "c:/Users/jwoli/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl" -f interface/stlink.cfg -f target/stm32h7x_dual_bank.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 50002 for tcl connections
Info : Listening on port 50004 for telnet connections
Info : clock speed 1800 kHz
Info : STLINK V3J9M3 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.293992
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 50000
Info : Listening on port 50000 for gdb connections
Info : accepting 'gdb' connection on tcp/50000
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0x24000f68
Info : Device: STM32H74x/75x
Info : flash size probed value 2048
Info : STM32H7 flash has dual banks
Info : Bank (0) size is 1024 kb, base address is 0x08000000
Info : Device: STM32H74x/75x
Info : flash size probed value 2048
Info : STM32H7 flash has dual banks
Info : Bank (1) size is 1024 kb, base address is 0x08100000
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
Info : Unable to match requested speed 4000 kHz, using 3300 kHz
Info : Unable to match requested speed 4000 kHz, using 3300 kHz
Info : Padding image section 0 at 0x08000298 with 40 bytes
Info : Padding image section 1 at 0x08014e1c with 4 bytes (bank write end alignment)
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08003198 msp: 0x24080000
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08003198 msp: 0x24080000
Info : Unable to match requested speed 4000 kHz, using 3300 kHz
Info : Unable to match requested speed 4000 kHz, using 3300 kHz
Info : Padding image section 0 at 0x08100298 with 40 bytes
Info : Padding image section 1 at 0x0811285c with 4 bytes (bank write end alignment)
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08003198 msp: 0x24080000
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08003198 msp: 0x24080000
```

> Finally, If you use chained configs, try doing an "request": "attach" instead. But first, we gotta establish connection between gdb and openocd which is what is failing for you.

Surprisingly, today I've tried to run the same configuration on PyOCD with "attach" request and it does work. In case of OpenOCD adding "attach" didin't change anything. But the issue for PyOCD is that it cannot errase the flash after booting it again. It says something like this:

```
0001056 I Target type is stm32h745zitx [board]
0001260 I DP IDR = 0x6ba02477 (v2 rev6) [dap]
0001323 I AHB-AP#0 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
0001325 I AHB-AP#1 IDR = 0x84770001 (AHB-AP var0 rev8) [ap]
[... 62 lines cut ...]
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\core\soc_target.py", line 218, in read_core_register
    return self.selected_core_or_raise.read_core_register(id)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\cortex_m.py", line 971, in read_core_register
    regValue = self.read_core_register_raw(reg_info.index)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\cortex_m.py", line 988, in read_core_register_raw
    vals = self.read_core_registers_raw([reg])
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\cortex_m.py", line 1009, in read_core_registers_raw
    return self._base_read_core_registers_raw(reg_list)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\coresight\cortex_m.py", line 1023, in _base_read_core_registers_raw
    raise exceptions.CoreRegisterAccessError(
pyocd.core.exceptions.CoreRegisterAccessError: cannot read register r0 because core #0 is not halted
[... cut ...]
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\gdbserver\packet_io.py", line 146, in _write_packet
    written = self._abstract_socket.write(packet)
  File "C:\Users\jwoli\AppData\Local\Programs\Python\Python310\lib\site-packages\pyocd\utility\sockets.py", line 58, in write
    return self.conn.send(data)
TypeError: a bytes-like object is required, not 'tuple'
```

I know that it's not related to Cortex Debug extension so i don't expect an anwser for this issue. I need to double check this.

haneefdm, 2022-06-01:
Two problems I found -- both with OpenOCD and specific to how ST configures it.

1. First you have to enable DUAL_CORE. If you look in the stm32h7x.cfg file you will a dependence on that
2. The transport being used `hla_swd` does not support multi-core debug.

Problem 1 is solved by
```
            "openOCDPreConfigLaunchCommands": [
                "set DUAL_CORE 1"
            ],
```
Which results in
```
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Warning : hla does not support multicore debugging
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
```

I cannot solve any of these issues. I don't know what or how you are supposed to STLink-V3 + OpenOCD and have dual core debug working.

Hint: I generally look at STM32CubeIDE as to how it does things and copy paste stuff from there. They seem to be using something very different with their own version of OpenOCD but even with that, I have not been able to get a dual core debug working.

Their tutorial only talks about their own server which is the STLink-Server which is a replacement for OpenOCD. I would try STs own server

I am going to close this because thus far I have not found an issue with Cortex-Debug itself. If we cannot get things running even from the command-line we can't do with this extension. For chained launched to work, all the other SW and HW also has to work.

If you see an issue that I can look at I can re-open this