Marus/cortex-debug #668: Debugging LPC824 with BlackMagicProbe, pausing does not work

Squantor, 2022-06-04:
Hello all,

I am running into an issue while using cortex-debug and pausing the program. 

When I start the program via vscode (pressing F5), it loads fine and breakpoints at main as I configured it. When I continue with F5 the program runs normally but pausing (F6) does not work, it keeps on running and nothing happens.

When I stop the debug session it exits and the program keeps running but the activity led on the blackmagic probe keeps blinking. Like it is while running. Restarting the debug session with F5 fails and I need to reset/powercycle the board.

Hardware setup:
I am using my own design LPC824 board with a replica of the black magic probe. There is some hardware connected to it but I encounter this issue also with a very minimal while(1) __NOP(); loop. I am running the 3ccb0af21c02516e6665c23ee218e5f24be56be4 of the blackmagic firmware (newest at this moment).

Software setup:
I am using Cortex-debug version 1.4.4 with visual studio code 1.67.2 in ubuntu 22.04 LTS
Configuration:
```
{
    "name": "Debug embedded ui mini",
    "cwd": "${workspaceRoot}",
    "executable": "${workspaceFolder}/bin/embedded_ui_mini/debug/embedded_ui_mini.elf",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "bmp",
    "interface": "swd",
    "device": "LPC824",
    "BMPGDBSerialPort": "/dev/ttyACM0",
    "runToEntryPoint": "main",
    "preLaunchCommands": [],
    "preRestartCommands": [],
    "preLaunchTask": "build embedded ui mini",
    "svdFile": "${workspaceFolder}/libMcuLL/svd/nxp/LPC824.svd",
    "debuggerArgs": ["--command=${workspaceFolder}/gdb_scripts/embedded_ui_mini_debug.txt"],
},
```
contents of embedded_ui_mini_debug.txt:
```
set print pretty on
set architecture armv6
file bin/embedded_ui_mini/debug/embedded_ui_mini.elf
target extended-remote /dev/ttyACM0
monitor swdp_scan
attach 1
set mem inaccessible-by-default off
set {int}0x40048000 = 2
load bin/embedded_ui_mini/debug/embedded_ui_mini.elf
set {int}0x40048000 = 2
compare-sections
```

I have tried out the following:
To rule out any hardware/debugger issues. I have tried using a segger and that works without a problem on the same hardware/program. Pausing and resuming works without a problem.
When I hit a bkpt instruction the debugger stops properly at that point.
When I use the same debug commands but from gdb I can debug normally, CTRL-C in gdb work fine. Only when using cortex-debug I cant pause.

Is this a known problem? Can I be of assistance in reproducing this problem?

With kind regards, Bart

PS: Thanks for developing cortex-debug, it works like a champ, love the SVD view!

haneefdm, 2022-06-04:
Can't tell what version of the extension you are using. Could you try the latest pre-release (available from marketplace inside VSCode itself).

If you are doing all the loading initializing yourself, we are also trying to do that and would conflict/redundant. Ideally, all those commands should be in "overrideLaunchCommands" at the end of which we expect the program to halt at the reset vector. Or, create a dummy overrideLaunchCommands that simple does an `echo hello`

In the debug console, there are instructions to create verbose logs. Please enable and attach the output.

Squantor, 2022-06-05:
Hello,

The reason that I need to modify the launching sequence is that the LPC series microcontrollers map rom on flash for the bootloader, when debugging you need to switch it to flash, otherwise you get issues. Hence the ```set {int}0x40048000 = 2``` in the debugging script. I use an external debugging script as then I have just one spot of this script instead of multiple in various configs.

I have done the following:
* updated cortex debug to the latest pre-release (1.5.1 git(89a84bd))
* set showDevDebugOutput to raw
* set overrideLaunchCommands with ["echo hello"]

I am getting this message in the debug window and as a popup now and nothing happens:
```
Failed to generate gdb commands: Cannot change this setting while the inferior is running. (from gdb-set mi-async on)
Cannot change this setting while the inferior is running.
    at i (/home/abilos/.vscode/extensions/marus25.cortex-debug-1.5.1/dist/debugadapter.js:2:158692)
    at Object.handlers.<computed> (/home/abilos/.vscode/extensions/marus25.cortex-debug-1.5.1/dist/debugadapter.js:2:159186)
    at /home/abilos/.vscode/extensions/marus25.cortex-debug-1.5.1/dist/debugadapter.js:2:146629
    at Array.forEach (<anonymous>)
    at v.onOutput (/home/abilos/.vscode/extensions/marus25.cortex-debug-1.5.1/dist/debugadapter.js:2:146062)
    at v.stdout (/home/abilos/.vscode/extensions/marus25.cortex-debug-1.5.1/dist/debugadapter.js:2:145479)
    at Socket.emit (node:events:390:28)
    at addChunk (node:internal/streams/readable:315:12)
    at readableAddChunk (node:internal/streams/readable:289:9)
    at Socket.Readable.push (node:internal/streams/readable:228:10)
```
Full output of Debug Console is attached.
This is my current config:
```
{
    "name": "Debug embedded ui mini",
    "cwd": "${workspaceRoot}",
    "executable": "${workspaceFolder}/bin/embedded_ui_mini/debug/embedded_ui_mini.elf",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "bmp",
    "interface": "swd",
    "device": "LPC824",
    "BMPGDBSerialPort": "/dev/ttyACM0",
    "runToEntryPoint": "main",
    "preLaunchCommands": [],
    "preRestartCommands": [],
    "preLaunchTask": "build embedded ui mini",
    "svdFile": "${workspaceFolder}/libMcuLL/svd/nxp/LPC824.svd",
    "debuggerArgs": ["--command=${workspaceFolder}/gdb_scripts/embedded_ui_mini_debug.txt"],
    "showDevDebugOutput": "raw",
    "overrideLaunchCommands": ["echo hello"],
},
```

I tried upgrading to prerelease before I posted here, and I got the same popup and message, so I went back to the latest release.

What commands do you execute as part of preLaunch/postLaunch? Maybe I can pull apart my current debug script to see how I can fit it in there, but also in what state does gdb need to be for cortex-debug?

Thanks a lot for your help

[cortex-debug-debug-console.txt](https://github.com/Marus/cortex-debug/files/8839598/cortex-debug-debug-console.txt)

haneefdm, 2022-06-05:
1. You need to do everything we do so our interaction with gdb meets expectations. And this can change over time. You have to experiment with it or run a session other than this board to see the full list of commands.
2. **At the end of what every you do, you need to be in a halted state. Or nothing works. This is your main failure.**

If you can do the second thing above properly, then it may just work.

```
Cannot change this setting while the inferior is running.
```
Gdb error message. Basically we cannot talk to gdb at all and hence...

Try `-exec-interrupt` as the last thing you do in your script.

haneefdm, 2022-06-05:
Looking at your script again. You are opening a connection yourself `target extended-remote /dev/ttyACM0`, but we will also be doing that. There is no override for that and hope it works. Instead of `-exec-interrupt` try `monitor halt`

Also, instead of `"request": "launch"`, `"request": "attach"` is more appropriate but again, we need the program to be halted at the end of the startup sequence.

Reading the following may help
https://github.com/Marus/cortex-debug/wiki/Overview

Squantor, 2022-06-05:
I have tried getting the script to work and adding the commands that you advise, but nothing worked.

It seems that running a custom debug script creates a lot of pain on my side. So took a step back, make a configuration without the custom script that will run normally maybe with incorrect settings. When I got this to work I added the LPC824 specific ```set {int}0x40048000 = 2``` and verified that the flash was written correctly with ```compare-sections```. This now works like normal and I can pause and resume like with my segger config.

For future reference, here is my configuration:
```
  {
    "name": "Debug embedded ui mini",
    "cwd": "${workspaceRoot}",
    "executable": "${workspaceFolder}/bin/embedded_ui_mini/debug/embedded_ui_mini.elf",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "bmp",
    "interface": "swd",
    "targetId": 1,
    "device": "LPC824",
    "BMPGDBSerialPort": "/dev/ttyACM0",
    "runToEntryPoint": "main",
    "preLaunchCommands": [],
    "postLaunchCommands": ["set {int}0x40048000 = 2","compare-sections"],
    "preRestartCommands": [],
    "preLaunchTask": "build embedded ui mini",
    "svdFile": "${workspaceFolder}/libMcuLL/svd/nxp/LPC824.svd",
},
```

I will try at a later stage get the debug script to work as I found where in the plugin you do the BlackMagicProbe specific initialization, so at least I can try and imitate the same.

Thanks a lot for your help.

closed 2022-06-05 by Squantor