Marus/cortex-debug #669: ST-Util: GDB Server Quit Unexpectedly (Both Cortex-Debug 1.4.4 as 1.5.1)

BrentVerhaegen, 2022-06-04:
**Setup**

- VSCode 1.67.2
- Windows_NT x64 10.0.22000
- st-util Release 1.7.0 (from [https://github.com/stlink-org/stlink](https://github.com/stlink-org/stlink))
- ARM Toolchain arm-none-eabi 11.2.2022.02
- Tested both Cortex-Debug v1.4.4 (stable) and v1.5.1 (pre-release) 

**Problem**
Error during launch of debugging. The code also does not get loaded in flash of the MCU, although I can see some comunication happening with the ST-Link. Same issue when using `stutil` as `stlink` servertype.
Error: `ST-Util: GDB Server Quit Unexpectedly. See gdb-server output for more details.`
Flashing and running using st-flash works fine. Debugging through STM32CubeIDE as well.
I've attached as much information as possible. All help here is appreciated!

**Launch Config**
```
"configuration": {
    "name": "STlink debug launch",
    "cwd": "C:\\Users\\user\\Desktop\\Nucleo-F030-test",
    "executable": "C:/Users\\user\\Desktop\\Nucleo-F030-test\\build\\Nucleo-F030-test.elf",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "stutil",
    "device": "STM32F030R8",
    "interface": "swd",
    "runToEntryPoint": "main",
    "preLaunchTask": "Compile using make",
    "svdFile": "C:\\Users\\user\\Desktop\\Nucleo-F030-test/.vscode/STM32F0x0.svd",
    "swoConfig": {
        "source": "probe",
        "decoders": []
    },
    "showDevDebugOutput": "raw",
    "__configurationTarget": 5,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "serverpath": "C:\\Program Files (x86)\\stlink\\stlink-1.7.0-x86_64-w64-mingw32\\bin\\st-util.exe",
    "toolchainPath": "C:\\Program Files (x86)\\Arm GNU Toolchain arm-none-eabi\\11.2 2022.02\\bin",
    "toolchainPrefix": "arm-none-eabi",
    "gdbPath": "C:\\Program Files (x86)\\Arm GNU Toolchain arm-none-eabi\\11.2 2022.02\\bin\\arm-none-eabi-gdb.exe",
    "objdumpPath": null,
    "extensionPath": "c:/Users/user/.vscode/extensions/marus25.cortex-debug-1.5.1",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.5.1",
    "__sessionId": "a246a0cc-31f5-46c7-8c54-2a28c5ee96c1"
}
```

**Terminal output**

```
[2022-06-04T22:59:10.300Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Program Files (x86)\\stlink\\stlink-1.7.0-x86_64-w64-mingw32\\bin\\st-util.exe" -p 50000 --no-reset
st-util
[... 9 lines cut ...]

```
Reading symbols from c:/program files (x86)\arm gnu toolchain arm-none-eabi\11.2 2022.02\bin\arm-none-eabi-nm.exe --defined-only -S -l -C -p C:/Users\user\Desktop\Nucleo-F030-test\build\Nucleo-F030-test.elf
Launching GDB: "C:\\Program Files (x86)\\Arm GNU Toolchain arm-none-eabi\\11.2 2022.02\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2 "C:/Users\\user\\Desktop\\Nucleo-F030-test\\build\\Nucleo-F030-test.elf"
1-gdb-version
From client: initialize({"clientID":"vscode","clientName":"Visual Studio Code","adapterID":"cortex-debug","pathFormat":"path","linesStartAt1":true,"columnsStartAt1":true,"supportsVariableType":true,"supportsVariablePaging":true,"supportsRunInTerminalRequest":true,"locale":"nl","supportsProgressReporting":true,"supportsInvalidatedEvent":true,"supportsMemoryReferences":true})
To client: {"seq":0,"type":"response","request_seq":1,"command":"initialize","success":true,"body":{"supportsHitConditionalBreakpoints":true,"supportsConfigurationDoneRequest":true,"supportsConditionalBreakpoints":true,"supportsLogPoints":true,"supportsFunctionBreakpoints":true,"supportsEvaluateForHovers":true,"supportsSetVariable":true,"supportsRestartRequest":true,"supportsGotoTargetsRequest":true,"supportSuspendDebuggee":true,"supportTerminateDebuggee":true,"supportsDataBreakpoints":true,"supportsDisassembleRequest":true,"supportsSteppingGranularity":true,"supportsInstructionBreakpoints":true,"supportsReadMemoryRequest":true}}
From client: launch({"name":"STlink debug launch","cwd":"C:\\Users\\user\\Desktop\\Nucleo-F030-test","executable":"C:/Users\\user\\Desktop\\Nucleo-F030-test\\build\\Nucleo-F030-test.elf","request":"launch","type":"cortex-debug","servertype":"stutil","device":"STM32F030R8","interface":"swd","runToEntryPoint":"main","preLaunchTask":"Compile using make","svdFile":"C:\\Users\\user\\Desktop\\Nucleo-F030-test/.vscode/STM32F0x0.svd","swoConfig":{"source":"probe","decoders":[]},"showDevDebugOutput":"vscode","__configurationTarget":5,"gdbServerConsolePort":55878,"pvtAvoidPorts":[],"chainedConfigurations":{"enabled":false},"debuggerArgs":[],"rttConfig":{"enabled":false,"decoders":[]},"graphConfig":[],"preLaunchCommands":[],"postLaunchCommands":[],"preAttachCommands":[],"postAttachCommands":[],"preRestartCommands":[],"postRestartCommands":[],"serverpath":"C:\\Program Files (x86)\\stlink\\stlink-1.7.0-x86_64-w64-mingw32\\bin\\st-util.exe","toolchainPath":"C:\\Program Files (x86)\\Arm GNU Toolchain arm-none-eabi\\11.2 2022.02\\bin","toolchainPrefix":"arm-none-eabi","gdbPath":"C:\\Program Files (x86)\\Arm GNU Toolchain arm-none-eabi\\11.2 2022.02\\bin\\arm-none-eabi-gdb.exe","objdumpPath":null,"extensionPath":"c:/Users/user/.vscode/extensions/marus25.cortex-debug-1.5.1","registerUseNaturalFormat":true,"variableUseNaturalFormat":true,"pvtVersion":"1.5.1","__sessionId":"a246a0cc-31f5-46c7-8c54-2a28c5ee96c1"})
To client: {"seq":0,"type":"event","event":"output","body":{"category":"stdout","output":"Cortex-Debug: VSCode debugger extension version 1.5.1 git(89a84bd). Usaage info: https://github.com/Marus/cortex-debug#usage"}}
To client: {"seq":0,"type":"event","event":"output","body":{"category":"stderr","output":"\"configuration\": {\n    \"name\": \"STlink debug launch\",\n    \"cwd\": \"C:\\\\Users\\\\user\\\\Desktop\\\\Nucleo-F030-test\",\n    \"executable\": \"C:/Users\\\\user\\\\Desktop\\\\Nucleo-F030-test\\\\build\\\\Nucleo-F030-test.elf\",\n    \"request\": \"launch\",\n    \"type\": \"cortex-debug\",\n    \"servertype\": \"stutil\",\n    \"device\": \"STM32F030R8\",\n    \"interface\": \"swd\",\n    \"runToEntryPoint\": \"main\",\n    \"preLaunchTask\": \"Compile using make\",\n    \"svdFile\": \"C:\\\\Users\\\\user\\\\Desktop\\\\Nucleo-F030-test/.vscode/STM32F0x0.svd\",\n    \"swoConfig\": {\n        \"source\": \"probe\",\n        \"decoders\": []\n    },\n    \"showDevDebugOutput\": \"raw\",\n    \"__configurationTarget\": 5,\n    \"gdbServerConsolePort\": 55878,\n    \"pvtAvoidPorts\": [],\n    \"chainedConfigurations\": {\n        \"enabled\": false\n    },\n    \"debuggerArgs\": [],\n    \"rttConfig\": {\n        \"enabled\": false,\n        \"decoders\": []\n    },\n    \"graphConfig\": [],\n    \"preLaunchCommands\": [],\n    \"postLaunchCommands\": [],\n    \"preAttachCommands\": [],\n    \"postAttachCommands\": [],\n    \"preRestartCommands\": [],\n    \"postRestartCommands\": [],\n    \"serverpath\": \"C:\\\\Program Files (x86)\\\\stlink\\\\stlink-1.7.0-x86_64-w64-mingw32\\\\bin\\\\st-util.exe\",\n    \"toolchainPath\": \"C:\\\\Program Files (x86)\\\\Arm GNU Toolchain arm-no[...]
To client: {"seq":0,"type":"event","event":"output","body":{"category":"stderr","output":"Reading symbols from C:\\Program Files (x86)\\Arm GNU Toolchain arm-none-eabi\\11.2 2022.02\\bin\\arm-none-eabi-objdump.exe --syms -C -h -w C:/Users\\user\\Desktop\\Nucleo-F030-test\\build\\Nucleo-F030-test.elf\n"}}
To client: {"seq":0,"type":"event","event":"output","body":{"category":"stderr","output":"Reading symbols from c:/program files (x86)\\arm gnu toolchain arm-none-eabi\\11.2 2022.02\\bin\\arm-none-eabi-nm.exe --defined-only -S -l -C -p C:/Users\\user\\Desktop\\Nucleo-F030-test\\build\\Nucleo-F030-test.elf\n"}}
To client: {"seq":0,"type":"event","event":"output","body":{"category":"stderr","output":"Launching GDB: \"C:\\\\Program Files (x86)\\\\Arm GNU Toolchain arm-none-eabi\\\\11.2 2022.02\\\\bin\\\\arm-none-eabi-gdb.exe\" -q --interpreter=mi2 \"C:/Users\\\\user\\\\Desktop\\\\Nucleo-F030-test\\\\build\\\\Nucleo-F030-test.elf\"\n"}}
To client: {"seq":0,"type":"event","event":"output","body":{"category":"stderr","output":"1-gdb-version\n"}}
[... cut ...]
Finished reading symbols from nm: Time: 151 ms
Finished reading symbols from objdump: Time: 159 ms
ST-Util: GDB Server Quit Unexpectedly. See gdb-server output for more details.
To client: {"seq":0,"type":"response","request_seq":2,"command":"launch","success":false,"message":"ST-Util: GDB Server Quit Unexpectedly. See gdb-server output for more details.","body":{"error":{"id":103,"format":"ST-Util: GDB Server Quit Unexpectedly. See gdb-server output for more details.","showUser":true}}}
```

</details>

haneefdm, 2022-06-04:
Try running the stutil manually. See what it does. You can see the whole command line in debug console
```
Launching gdb-server: "C:\\Program Files (x86)\\stlink\\stlink-1.7.0-x86_64-w64-mingw32\\bin\\st-util.exe" -p 50000 --no-reset
```
Also, I can only debug/help if you are using the latest 1.5.1. The attached output seems to be a mix of 1.4.x and 1.5.1?

haneefdm, 2022-06-04:
I have zero experience with stutil. Does it need to be told which version of stlink the board has?

haneefdm, 2022-06-04:
Also see if you can start gdb manually. It did not print anything. It should have printed something.

BrentVerhaegen, 2022-06-05:
Thank you for your kind reply @haneefdm 

**st-util**
When running the st-util server manually, I get the following output
```
C:\Users\user> st-util.exe -p 50000 --no-reset
st-util
2022-06-05T14:29:11 INFO common.c: F0xx: 8 KiB SRAM, 64 KiB flash in at least 1 KiB pages.
2022-06-05T14:29:11 INFO gdb-server.c: Listening at *:50000...
```
I can see the light on the debugger flash for a second. Seems like the gdb server is running.
It can detect that there is a STM32F0x with 64kB flash connected (as you can see in above log), so there is communication.

**OpenOCD**
Since you said you don't have much experience with st-util, and I'm not married to it in any way either, I gave OpenOCD a try.

First trying to run OpenOCD from the command line:
```
C:\Users\user> openocd -f C:\Users\user\Desktop\Nucleo-F030-test\.vscode\stlink-v2.cfg -f target/stm32f0x.cfg
Open On-Chip Debugger 0.11.0 (2021-11-18) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : STLINK V2J39M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.253088
Info : stm32f0x.cpu: Cortex-M0 r0p0 processor detected
Info : stm32f0x.cpu: target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for stm32f0x.cpu on 3333
Info : Listening on port 3333 for gdb connections
```
This runs just fine and waits for connections to the server. I can see the light on the STLink flash continously.

Then trying from VSCode using Cortex-Debug 1.5.1:
Terminal output:
```
[2022-06-05T12:38:31.539Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Program Files (x86)\\OpenOCD\\OpenOCD-20211118-0.11.0\\bin\\openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "C:\\Users\\user\\Desktop\\Nucleo-F030-test" -f "c:/Users/user/.vscode/extensions/marus25.cortex-debug-1.5.1/support/openocd-helpers.tcl" -f "C:\\Users\\user\\Desktop\\Nucleo-F030-test/.vscode/stlink-v2.cfg" -f target/stm32f0x.cfg
Open On-Chip Debugger 0.11.0 (2021-11-18) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 1000 kHz
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : STLINK V2J39M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.257819
Info : stm32f0x.cpu: Cortex-M0 r0p0 processor detected
Info : stm32f0x.cpu: target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for stm32f0x.cpu on 50000
Info : Listening on port 50000 for gdb connections
[2022-06-05T12:38:33.986Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

**Debug console**
<details>
<summary>Click to toggle contents of `Debug console`</summary>

[... 57 lines cut ...]
}
Reading symbols from C:\Program Files (x86)\Arm GNU Toolchain arm-none-eabi\11.2 2022.02\bin\arm-none-eabi-objdump.exe --syms -C -h -w C:/Users\user\Desktop\Nucleo-F030-test\build\Nucleo-F030-test.elf
Reading symbols from c:/program files (x86)\arm gnu toolchain arm-none-eabi\11.2 2022.02\bin\arm-none-eabi-nm.exe --defined-only -S -l -C -p C:/Users\user\Desktop\Nucleo-F030-test\build\Nucleo-F030-test.elf
Launching GDB: "C:\\Program Files (x86)\\Arm GNU Toolchain arm-none-eabi\\11.2 2022.02\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2 "C:/Users\\user\\Desktop\\Nucleo-F030-test\\build\\Nucleo-F030-test.elf"
1-gdb-version
From client: initialize({"clientID":"vscode","clientName":"Visual Studio Code","adapterID":"cortex-debug","pathFormat":"path","linesStartAt1":true,"columnsStartAt1":true,"supportsVariableType":true,"supportsVariablePaging":true,"supportsRunInTerminalRequest":true,"locale":"nl","supportsProgressReporting":true,"supportsInvalidatedEvent":true,"supportsMemoryReferences":true})
To client: {"seq":0,"type":"response","request_seq":1,"command":"initialize","success":true,"body":{"supportsHitConditionalBreakpoints":true,"supportsConfigurationDoneRequest":true,"supportsConditionalBreakpoints":true,"supportsLogPoints":true,"supportsFunctionBreakpoints":true,"supportsEvaluateForHovers":true,"supportsSetVariable":true,"supportsRestartRequest":true,"supportsGotoTargetsRequest":true,"supportSuspendDebuggee":true,"supportTerminateDebuggee":true,"supportsDataBreakpoints":true,"supportsDisassembleRequest":true,"supportsSteppingGranularity":true,"supportsInstructionBreakpoints":true,"supportsReadMemoryRequest":true}}
From client: launch({"name":"OpenOCD debug launch","cwd":"C:\\Users\\user\\Desktop\\Nucleo-F030-test","executable":"C:/Users\\user\\Desktop\\Nucleo-F030-test\\build\\Nucleo-F030-test.elf","request":"launch","type":"cortex-debug","servertype":"openocd","config [... cut ...]
Finished reading symbols from nm: Time: 164 ms
Finished reading symbols from objdump: Time: 185 ms
OpenOCD: GDB Server Quit Unexpectedly. See gdb-server output for more details.
To client: {"seq":0,"type":"response","request_seq":2,"command":"launch","success":false,"message":"OpenOCD: GDB Server Quit Unexpectedly. See gdb-server output for more details.","body":{"error":{"id":103,"format":"OpenOCD: GDB Server Quit Unexpectedly. See gdb-server output for more details.","showUser":true}}}
```

</details>

haneefdm, 2022-06-05:
```
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
```
With that nothing works :-) Maybe there is a process running stutil/openocd that has USB still open On maybe improper installation. Don't know why USB would fail and you still get some details on the CPU.

If you can't find who has USB open, then reboot is the only way I know.

Have you trued ST-Link, it is far simpler - maybe some features missing. Try that if you have installed STM32CubeIDE in its normal location and we will find the location of the program for you

BrentVerhaegen, 2022-06-05:
I can't seem to get rid of that USB error. What I've tried:

- Close all other processes/terminals that could've taken the USB port
- Install latest STlink drivers from the ST website
- Other USB cable/other USB port (directly on PC, no hubs)
- Other Nucleo board
- Reboot computer

Ok, Let's try stlink, third time's a charm :-) I have STM32CubeIDE installed, so it's not too difficult.
I've found the STLink GDB server on the following path: `"C:\\Program Files\\ST\\STM32CubeIDE_1.6.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_1.6.0.202101291314\\tools\\bin\\ST-LINK_gdbserver`

The give's the following gdb-server terminal output:
```
[2022-06-05T13:53:59.935Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Program Files\\ST\\STM32CubeIDE_1.6.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_1.6.0.202101291314\\tools\\bin\\ST-LINK_gdbserver" -p 50000 -cp "C:\\Program Files\\STMicroelectronics\\STM32Cube\\STM32CubeProgrammer\\bin" --swd --halt

STMicroelectronics ST-LINK GDB server. Version 5.8.0
Copyright (c) 2020, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 31
        Listen Port Number         : 50000
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled

Target connection mode: Default
Reading ROM table for AP 0 @0xe00fffd0
Hardware watchpoint supported by the target 
COM frequency = 4000 kHz
ST-LINK Firmware version : V2J39M27
Device ID: 0x440
PC: 0x8001a40
ST-LINK device status: HALT_MODE
ST-LINK detects target voltage = 3.26 V
ST-LINK device status: HALT_MODE
ST-LINK device initialization OK
Waiting for debugger connection...
Waiting for connection on port 50000...
[2022-06-05T13:54:02.341Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
``` 

With the following Debug console:
```
"configuration": {
    "name": "STlink debug launch",
    "cwd": "C:\\Users\\user\\Desktop\\Nucleo-F030-test",
    "executable": "C:/Users\\user\\Desktop\\Nucleo-F030-test\\build\\Nucleo-F030-test.elf",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "stlink",
    "device": "STM32F030R8",
    "interface": "swd",
    "runToEntryPoint": "main",
    "preLaunchTask": "Compile using make",
    "svdFile": "C:\\Users\\user\\Desktop\\Nucleo-F030-test/.vscode/STM32F0x0.svd",
    "swoConfig": {
        "source": "probe",
        "decoders": []
    },
    "showDevDebugOutput": "raw",
    "__configurationTarget": 5,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
[... 18 lines cut ...]
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.5.1",
    "__sessionId": "7bd9909e-7d1c-440d-8ce4-af8353f343d9"
}
Reading symbols from C:\Program Files (x86)\Arm GNU Toolchain arm-none-eabi\11.2 2022.02\bin\arm-none-eabi-objdump.exe --syms -C -h -w C:/Users\user\Desktop\Nucleo-F030-test\build\Nucleo-F030-test.elf
Reading symbols from c:/program files (x86)\arm gnu toolchain arm-none-eabi\11.2 2022.02\bin\arm-none-eabi-nm.exe --defined-only -S -l -C -p C:/Users\user\Desktop\Nucleo-F030-test\build\Nucleo-F030-test.elf
Launching GDB: "C:\\Program Files (x86)\\Arm GNU Toolchain arm-none-eabi\\11.2 2022.02\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2 "C:/Users\\user\\Desktop\\Nucleo-F030-test\\build\\Nucleo-F030-test.elf"
1-gdb-version
To client: {"seq":0,"type":"event","event":"output","body":{"category":"stdout","output":"Cortex-Debug: VSCode debugger extension version 1.5.1 git(89a84bd). Usaage info: https://github.com/Marus/cortex-debug#usage"}}
To client: {"seq":0,"type":"event","event":"output","body":{"category":"stderr","output":"\"configuration\": {\n    \"name\": \"STlink debug launch\",\n    \"cwd\": \"C:\\\\Users\\\\user\\\\Desktop\\\\Nucleo-F030-test\",\n    \"executable\": \"C:/Users\\\\user\\\\Desktop\\\\Nucleo-F030-test\\\\build\\\\Nucleo-F030-test.elf\",\n    \"request\": \"launch\",\n    \"type\": \"cortex-debug\",\n    \"servertype\": \"stlink\",\n    \"device\": \"STM32F030R8\",\n    \"interface\": \"swd\",\n    \"runToEntryPoint\": \"main\",\n    \"preLaunchTask\": \"Compile using make\",\n    \"svdFile\": \"C:\\\\Users\\\\user\\\\Desktop\\\\Nucleo-F030-test/.vscode/STM32F0x0.svd\",\n    \"swoConfig\": {\n        \"source\": \"probe\",\n        \"decoders\": []\n    },\n    \"showDevDebugOutput\": \"raw\",\n    \"__configurationTarget\": 5,\n    \"gdbServerConsolePort\": 55878,\n    \"pvtAvoidPorts\": [],\n    \"chainedConfigurations\": {\n        \"enabled\": false\n    },\n    \"debuggerArgs\": [],\n    \"rttConfig\": {\n        \"enabled\": false,\n        \"decoders\": []\n    },\n    \"graphConfig\": [],\n    \"preLaunchCommands\": [],\n    \"postLaunchCommands\": [],\n    \"preAttachCommands\": [],\n    \"postAttachCommands\": [],\n    \"preRestartCommands\": [],\n    \"postRestartCommands\": [],\n    \"serverpath\": \"C:\\\\Program Files\\\\ST\\\\STM32CubeIDE_1.6.0\\\\STM32CubeIDE\\\\plugins\\\\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_1.6.0.202101291314\\\ [... cut ...] ver
Finished reading symbols from nm: Time: 149 ms
Finished reading symbols from objdump: Time: 179 ms
ST-LINK: GDB Server Quit Unexpectedly. See gdb-server output for more details.
To client: {"seq":0,"type":"response","request_seq":2,"command":"launch","success":false,"message":"ST-LINK: GDB Server Quit Unexpectedly. See gdb-server output for more details.","body":{"error":{"id":103,"format":"ST-LINK: GDB Server Quit Unexpectedly. See gdb-server output for more details.","showUser":true}}}
```

BrentVerhaegen, 2022-06-05:
Running the STLink GDB server that is included in STM32CubeIDE on the terminal gives the following, and keeps running like this.
The LED on the STLink blinks 4-5 times.

```
 C:\Program Files\ST\STM32CubeIDE_1.6.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_1.6.0.202101291314\tools\bin> ./ST-LINK_gdbserver.exe -p 50000 -cp "C:/Program Files/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin" --swd --halt

STMicroelectronics ST-LINK GDB server. Version 5.8.0
Copyright (c) 2020, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 31
        Listen Port Number         : 50000
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled

Target connection mode: Default
Reading ROM table for AP 0 @0xe00fffd0
Hardware watchpoint supported by the target
COM frequency = 4000 kHz
ST-LINK Firmware version : V2J39M27
Device ID: 0x440
PC: 0x8001a40
ST-LINK device status: HALT_MODE
ST-LINK detects target voltage = 3.25 V
ST-LINK device status: HALT_MODE
ST-LINK device initialization OK
Waiting for debugger connection...
Waiting for connection on port 50000...
```

haneefdm, 2022-06-05:
With both OpenOCD and STLink, I see the same thing happening. The server opens a TCP port and then immediately, something happens and the whole session terminates. It may be that gdb or we are crashing.

Since you are on Windows, `cd %TEMP%` and look for a file `cortex-debug-server-exiting.log`. Hopefully, it is there (if we are unable to create it, we just move on). In this file, you will see a log of anything exiting, and who is exiting first and if we crashed, you will not see all the exits. It is a collection of all your sessions so look for the last few lines.

haneefdm, 2022-06-05:
In fact, same issue with ST-Util

BrentVerhaegen, 2022-06-05:
Thanks for your reply @haneefdm 
So you're able to reproduce the issue on your side?

Here are the last contents of the `cortex-debug-server-exiting.log` 

**STlink**
```
[2022-06-05T13:53:59.795Z] ppid=18612 pid=18612 STlink debug launch: ******* Starting new session request type="launch"
[2022-06-05T13:53:59.907Z] ppid=18612 pid=21768 GDB started ppid=18612 pid=21768
[2022-06-05T13:54:02.333Z] ppid=18612 pid=21768 GDB: exited
[2022-06-05T13:54:02.334Z] ppid=18612 pid=21768 STlink debug launch: quitEvent: Killing server
[2022-06-05T13:54:02.334Z] ppid=18612 pid=21768 GDBServer(9484): forcing an exit with kill()
[2022-06-05T13:54:02.338Z] ppid=18612 pid=21768 GDBServer(9484): exited code=null signal=SIGTERM
[2022-06-05T13:54:02.361Z] ppid=18612 pid=21768 STlink debug launch: quitEvent: sending VSCode TerminatedEvent
[2022-06-05T13:54:02.381Z] ppid=18612 pid=21768 STlink debug launch: Begin disconnectRequest
```

**OpenOCD**
```
[2022-06-05T13:45:49.564Z] ppid=22724 pid=22724 OpenOCD debug launch: ******* Starting new session request type="launch"
[2022-06-05T13:45:50.075Z] ppid=22724 pid=25664 GDB started ppid=22724 pid=25664
[2022-06-05T13:45:53.955Z] ppid=22724 pid=25664 GDB: exited
[2022-06-05T13:45:53.955Z] ppid=22724 pid=25664 OpenOCD debug launch: quitEvent: Killing server
[2022-06-05T13:45:53.956Z] ppid=22724 pid=25664 GDBServer(10916): forcing an exit with kill()
[2022-06-05T13:45:53.961Z] ppid=22724 pid=25664 GDBServer(10916): exited code=null signal=SIGTERM
[2022-06-05T13:45:53.967Z] ppid=22724 pid=25664 OpenOCD debug launch: quitEvent: sending VSCode TerminatedEvent
[2022-06-05T13:45:54.028Z] ppid=22724 pid=25664 OpenOCD debug launch: Begin disconnectRequest
```

haneefdm, 2022-06-05:
>> So you're able to reproduce the issue on your side?

Not even a chance. This is the first time I even heard of such a thing. From this, I am thinking your gdb is not working at all. A missing dll can do it. Perhaps you missed this from earlier?

> Also see if you can start gdb manually. It did not print anything. It should have printed something.

mildan99, 2022-06-05:
Hello, I encountered the same issue, but it turned out that gdb in latest ARM Toolchain is busted. After replacing it with gdb binary from prev release everything working ok.

BrentVerhaegen, 2022-06-05:
Thanks @mildan99, that was indeed the issue!
I had the  (latest) GNU ARM Embedded Toolchain rev. 11.2 2022.02 installed.
Now installed GNU ARM Embedded Toolchain rev. 10 2021.07 and everything is working as it should. 

Thanks again to both @haneefdm and @mildan99 for your help 👍

closed 2022-06-05 by haneefdm