Marus/cortex-debug #670: Update release on open-vsx

jounathaen, 2022-06-06:
Hi,
This extension provides so many awesome features, however, users of VSCodium cannot enjoy the latest version. The last one on [open-vsx seems to be 1.4.4](https://open-vsx.org/extension/marus25/cortex-debug), whereas the [Microsoft store is on 1.5.1](https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug). 
Maybe it's possible to update it.
Cheers!

haneefdm, 2022-06-06:
Sorry to say that I have no idea how our extension gets published in open-vsx. Maybe they don't support the `pre-release` flag that VSCode now supports.

1.4.4 -- is the official release
1.5.x -- these are pre-releases leading up to 1.6.0.

If the minor version is an odd number, and if it has a pre-release flag it is considered a pre-release by the Marketplace and VSCode.

This convention is established by MS/VSCode/MS-Marketplace. But they are all normal releases just with or without a pre-release flag. Within VSCode, users opt in for pre-releases or switch back and forth.

So, I don't know to answer your question. You may want to enquire with the open-vsx people to see if/when they plan to add support for pre-releases.

haneefdm, 2022-06-06:
Yes, the marketplace has all the releases and one can't even tell if you don't know the convention
[image]

If you never enabled pre-releases, it will look like this

[image]

I would submit a request to open-vsx to support pre-releases. It is an important feature for many users to get preview releases -- some fixes may only be in pre-releases. Yes, pre-releases are considered potentially less stable but that is generally not the case

jounathaen, 2022-06-06:
Alright. Now I get it. Thanks for your clarifications. I don't think there is anything to be done here in this case.

closed 2022-06-06 by haneefdm