Marus/cortex-debug #671: How to debug a relocated app (compile with -fpic, uploaded to an offset in flash)

basprins, 2022-06-07:
Hi,
I have a really hard time debugging my relocated app. I have a custom bootloader which is responsible for loading the vtor table to sram. Afterwards it's jump (depending on configuration) to either the app on its original linked location, or to the app on relocated location (using the global offset table to find global vars and functions). 

I am trying to attach to the CPU, and then 
- add symbols with the desired offset
- halt
- add a breakpoint
- reset

And then I hoped to hit a breakpoint. 

My launch configuration looks as follows:

```
    {
      "name": "relocated iobox-1024",
      "type": "cortex-debug",
      "request": "attach",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceRoot}/iobox-1024/build/iobox-1024.axf",
      "serverpath": "/usr/bin/JLinkGDBServerCLExe",
      "windows": {
        "serverpath": "C:\\Program Files (x86)\\SEGGER\\JLink\\JLinkGDBServerCL.exe",
      },
      "osx": {
        "serverpath": "/Applications/SEGGER/JLink_V764e/JLinkGDBServerCLExe",
      },
      "preAttachCommands": [
        "add-symbol-file ${workspaceRoot}/iobox-1024/build/iobox-1024.axf 0x600c1770",
      ],
      "postAttachCommands": [
        "monitor halt",
        "break ResetISR",
        "monitor reset",
      ],
      "servertype": "jlink",
      "device": "MIMXRT1024xxx5A",
      "preLaunchTask": "iobox-1024 - build",
      "postStartSessionCommands": [],
      "interface": "swd",
      // "showDevDebugOutput": "both",
      // "showDevDebugTimestamps": true,
    },      
```

It "sometimes works" but, it's terribly slow. It takes half a minute before something responds, and most of the time i end up with errors. A non-responsive debugger.

What I "really" want, is instead of an attach configuration, just make a launch configuration which would upload to the "relocated" offset (in my case, an offset of 0xa0000). But I don't know how to instruct cortex-debug to NOT upload the binary to its originally linked location, but upload it to my offset location instead. 

Since I can't get that to work, I continued with the second best alternative I could think of, and that is to make an attach configuration, and use pre/post attach commands. But this works for from reliable (probably because I am doing things wrong).

I also tried to incorporate the `load` in these pre/post commands, but I get an error message that it's not a valid gdb command. 

E.g.
[image]

I am able to share my bootloader.axf (=elf) and iobox-1024.axf if you want. But maybe you can already tell me what I am doing wrong from what I shared.

For what it's worth, the debug output

```
0000000012: Cortex-Debug: VSCode debugger extension version 1.5.1 git(89a84bd). Usaage info: https://github.com/Marus/cortex-debug#usage
0000000015: Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/bp/dev/iobox/firmware/iobox-1024/build/iobox-1024.axf
0000000018: Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/bp/dev/iobox/firmware/iobox-1024/build/iobox-1024.axf
0000000020: Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 /home/bp/dev/iobox/firmware/iobox-1024/build/iobox-1024.axf
0000000023: 1-gdb-version
0000000028: Launching gdb-server: /usr/bin/JLinkGDBServerCLExe -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MIMXRT1024xxx5A
0000000028:     Please check TERMINAL tab (gdb-server) for output from /usr/bin/JLinkGDBServerCLExe
0000000042: -> =thread-group-added,id="i1"
0000000042: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
[... 281 lines cut ...]
0000106745: GDB -> App: {"output":"","token":46,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000106745: 47-exec-interrupt
0000106745: -> 47^done
0000106745: GDB -> App: {"output":"","token":47,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000106745: 48-exec-interrupt
0000106746: -> 48^done
0000106746: GDB -> App: {"output":"","token":48,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000106746: 49-exec-interrupt
0000106747: -> 49^done
0000106747: GDB -> App: {"output":"","token":49,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000106849: Returning dummy thread-id to workaround VSCode issue with pause button not working
0000106853: Returning dummy stack frame to workaround VSCode issue with pause button not working: {"threadId":1,"startFrame":0,"levels":20}
0000106853: 50-exec-continue --all
0000106854: -> 50^running
0000106854: GDB -> App: {"output":"","token":50,"outOfBandRecord":[],"resultRecords":{"resultClass":"running","results":[]}}
0000106854: -> *running,thread-id="all"
0000106854: GDB -> App: {"output":"","outOfBandRecord":[{"isStream":false,"type":"exec","asyncClass":"running","output":[["thread-id","all"]]}]}
0000106854: mi2.status = running
0000106951: Info: Attempt to evalute expression while busy. {"expression":"&flash_peripheral::_norHandle,x","frameId":256,"context":"watch"}
0000106951: Info: Attempt to evalute expression while busy. {"expression":"buffer,x","frameId":256,"context":"watch"}
0000106951: Info: Attempt to evalute expression while busy. {"expression":"&Nor_Flash_Read,x","frameId":256,"context":"watch"}
0000106952: Info: Attempt to evalute expression while busy. {"expression":"flash_peripheral::_norHandle","frameId":256,"context":"watch"}
0000106952: Info: Attempt to evalute expression while busy. {"expression":"_appStartAddress,x","frameId":256,"context":"watch"}

```

haneefdm, 2022-06-08:
Use `"showDevDebugOutput": "raw"` for a much more compact output. Same info.

Several things may be causing problems.

* GDB and us, we should have the same world view of symbols and their addresses. You can still debug but many things may fail otherwise. Instead of using `preAttachCommands`, you can use `symbolFiles` to specify alternate symbol files.
* There is no way to program a hex file with an offset that I know of. To do this, you can create a hex file with an offset already applied and use `loadFiles` to program the device. Neither gdb or Jlink can apply an offset for you. The gdb load command only uses ELF format. You can try the ELF format. But there are many sections in the ELF file. code, data, etc. Which section should the offset be applied to?
* You can always create a script that does the custom programming and launch it use the `preLaunchTask`. Perhaps you can use JFlash and use its -relocate command line option or use a script. See https://www.segger.com/downloads/jlink/UM08003

In the end, it takes more mastery of gdb and JLinkGDBServer than cortex-debug to do what you want. If you cannot do this manually (from command-line), you cannot do it via cortex-debug. I am amazed that you gotten as far as you have gotten.

basprins, 2022-06-08:
Bruuh! I love you. There, i said it! I just fixed the last remaining issue in my app. Some functions that execute in ram were causing issues when relocating my app. With this brilliant addon I managed to find the bug!

closed 2022-06-08 by basprins