Marus/cortex-debug #672: ARM Toolchain path - which shell does cortex debug use?

dsabala, 2022-06-08:
Hi

Cortex-debug works great, however I need to get rid of 
`"armToolchainPath" : "/home/dsa/tools/gcc-arm/gcc-arm-none-eabi-10.3-2021.10/bin",`
from my `launch.json` file since I want to push VSCode config to repo and my reviewer is an Eclipse guy :) 

I am using `zsh` as my default shell. I added 
`export PATH="/home/dsa/tools/gcc-arm/gcc-arm-none-eabi-10.3-2021.10/bin:$PATH"`
to both `.zshrc`, `.bashrc` and `.profile` since I am not experienced in linux and not sure how things like `vscode` or `cortex-debug` use system shell. 

Now I guess I can execute `arm-none-eabi-gdb` from every shell possible and it works:
```
arm-none-eabi-gdb --version
bash
arm-none-eabi-gdb --version
sh
arm-none-eabi-gdb --version
``
My `launch.json`
{
    "version": "0.2.0",
    "configurations": [
      {
      // "showDevDebugOutput": "both",
      "type": "cortex-debug",
      "request": "launch",
      "name": "Debug J-Link",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceRoot}/build-gcc-arm-Debug/ecu-app.elf",
      "serverpath": "JLinkGDBServerCLExe", 
      "servertype": "jlink",
      "interface": "swd",
      "rttConfig": {
        "enabled": true,
        "address": "auto",
        "decoders": [
            {
                "port": 0,
                "type": "console",
                "noprompt": true,
            }
        ]
      },
      "runToEntryPoint": "_start",
      //"armToolchainPath" : "/home/dsa/tools/gcc-arm/gcc-arm-none-eabi-10.3-2021.10/bin",
      "jlinkscript": "debug.jlinkscript"
      }
      ]
}
```

Launching debug session results in 
```
GDB executable "arm-none-eabi-gdb" was not found.
Please configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath" correctly.
```

How to configure debug without hardcoding path to the toolchain?

dsabala, 2022-06-08:
Sorry for my issue-happiness. I solved it already, `armToolchainPath` is actually ARM toolchain path, not a path to bin directory, so setting the following path solved the problem:
`"armToolchainPath" : "/home/dsa/tools/gcc-arm/gcc-arm-none-eabi-10.3-2021.10",`
The mistake is caused by common pattern to set system PATH variable to point to `toolchain/bin` folder, not toolchain root folder...

closed 2022-06-08 by dsabala

haneefdm, 2022-06-08:
We don't use/specify any shell. It is what NodeJS (and VSCode) uses and I don't know what that is. I doubt it is any shell at all

https://nodejs.org/api/child_process.html#child_processspawncommand-args-options

dsabala, 2022-06-08:
@haneefdm Sorry once again but I was wrong, my solution didnt solved this problem, it look like a path have to be hardcoded

dsabala, 2022-06-08:
@haneefdm I got a solution: create symbolic link
```
cd /usr/bin 
sudo ln -s /home/dsa/tools/gcc-arm/gcc-arm-none-eabi-10.3-2021.10/bin/arm-none-eabi-gdb arm-none-eabi-gdb
```
It looks like it works now.

haneefdm, 2022-06-08:
Why not set the "cortex-debug.armToolchainPath": "/home/dsa/tools/gcc-arm/gcc-arm-none-eabi-10.3-2021.10/bin" in your settings.json file? User or Workspace settings. Not launch.json.

Checkout all the global settings in the VSCode settings editor for this extension.