Marus/cortex-debug #674: Please make stack-info-depth configurable

patrick-dojofive, 2022-06-09:
Really I'm facing two issues right now.
We are working with an elf file compiled with a functional safety version of armclang (v6.16LTS) and running into an issue where the stack trace does not terminate as expected.

In this scenario it looks like stack-info-depth defaults to 10k while other tools we've investigated with (SES and Ozone) will default to 100 frames.

If we can resolve the primary issue great, but a possible quick work around would be to make that 10k configurable.

haneefdm, 2022-06-09:
That is 10,000 frames is the limit. Are you saying gdb is taking too long? Meaning you have a corrupted stack. But what is the real problem?

FYI: Also, we do not control the actual number of frames requested. It is controlled by VSCode and how it wants to present the UI -- and it is not a very large number from what I recall.

Here is the API

https://github.com/microsoft/debug-adapter-protocol/blob/663bd36379980fb880733f0d9532cc7e78dc0d07/debugAdapterProtocol.json#L1933

We give no more levels than requested.

patrick-dojofive, 2022-06-09:
Hmm ok. What I'm observing is a few real frames and then the next 9995 (or whatever) all reference main. So gdb isn't detecting to stop backtracing.

haneefdm, 2022-06-09:
According to the ARM ABI, I thought it is the LR (Link Register) value of 0xffffffff that indicates end of stack. Nd, that is what I would expect gdb to be doing looking at all the old values of LR stored until it gets to the end.

closed 2022-06-09 by haneefdm