Marus/cortex-debug #675: SVD file not loaded / registers are not shown in VS Code UI

abarnas-summerrnd, 2022-06-09:
git lab issue:
I have the launch target configured as followed:
```
{ 
   "version": "0.2.0",
    "configurations": [
    {
    "name": "Cortex Debug",
    "type": "cortex-debug",
    "executable": "${workspaceRoot}/build/zephyr/zephyr.elf",
    "cwd": "${workspaceRoot}",
    "request": "launch",
    "servertype": "openocd",
    "interface": "swd",
    "configFiles": [
      "interface/stlink.cfg",
      "target/stm32h7x.cfg"
    ],
    "runToEntryPoint": "main",
    "svdFile": "${workspaceFolder}/boards/arm/higo/support/STM32H753.svd"
    }
    ]
}
```
Debugging works fine. OpenOCD output:
```
Open On-Chip Debugger 0.11.0+dev-00590-g2586fec92 (2022-03-07-12:32)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 50004 for tcl connections
Info : Listening on port 50005 for telnet connections
Info : clock speed 1800 kHz
Info : STLINK V2J37M26 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.240156
Info : stm32h7x.cpu0: Cortex-M7 r1p1 processor detected
Info : stm32h7x.cpu0: target has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 50003
Info : Listening on port 50003 for gdb connections
Info : accepting 'gdb' connection on tcp/50003
Info : Device: STM32H74x/75x
Info : flash size probed value 2048k
Info : STM32H7 flash has dual banks
Info : Bank (0) size is 1024 kb, base address is 0x08000000
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800307c msp: 0x24002880
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800307c msp: 0x24002880
Info : Padding image section 0 at 0x0800dbd6 with 2 bytes
Info : Padding image section 1 at 0x0800ff08 with 24 bytes (bank write end alignment)
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800307c msp: 0x24002880
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800307c msp: 0x24002880
```
But VS Code UI shows the message as follows:

"SVD: No active debug session or no SVD files specified."

SVD seems to load because if I change the svd path to incorrect then an error appears.

Is there any way to debug this behavior if there is no error in the console output?

abarnas-summerrnd, 2022-06-09:
System reboot have helped... Nothing changed. No idea why it is working now.

ChrisIdema, 2022-07-11:
> System reboot have helped... Nothing changed. No idea why it is working now.

I had the same. If I closed my first debug session any subsequent debug session didn't have register view. Restarting VsCode fixes that, but only for the first debug session. The solution is to use version 1.5.x of the extension (click switch to pre-release in extension tab)