Marus/cortex-debug #677: When closing VSCode after using cortex-debug, ubuntu crashes desktop environment

FL0WL0W, 2022-06-11:
**Describe the bug**
After using cortex-debug in vscode, when closing that instance of vscode, the desktop environment crashes and sends me back to the login screen. I am using an st-link debugger with openocd. I tried opening and closing openocd by itself, but that does not cause the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Start VSCode and open project
2. Start debugging device with Cortex-Debug
3. Stop debugging device
4. Close VSCode

**Expected behavior**
Not to crash my desktop environment

**Environment:**
 - VSCode Version 1.68
 - Ubuntu 20.04.4 LTS
 - 2x Intel E5-2667 V2 with 64GB of ram.
 - st-link v2
 - OpenOCD v0.10.0-esp32-20210902 (2021-09-02-09:38)
 - STM32F103C8T6

haneefdm, 2022-06-11:
Thanks for listing all the version numbers. But cortex-debug version is missing?

FYI: Whatever we/VSCode should not crash the desktop env. It is Unix after all and we don't interact with the desktop env. in anyway. VSCode on the other hand makes a ton of requests of the desktop env. VSCode Version 1.68 is very new, we have to wait and see. Did you try to revert VSCode to older version? Install an older version from https://code.visualstudio.com/updates/v1_68

Run vscode in verbose mode. From a terminal, run
```sh
code --verbose
```
You will see a ton of output in the terminal. Could you grab that and report back

The other thing to try is.
* You will the exact command line for both gdb and openocd in debug console
* Run both from separate terminals in the same dir. as what your launch.json says for `cwd`
* If you can, duplicate the gdb commands you see in vscode debug console.
* exit gdb
* exit openocd if it still hanging around.

I don't know about esp32 and their version of openocd. Seems super old. What you see here in `OpenOCD v0.10.0-esp32-20210902 (2021-09-02-09:38)` is the build date. We always recommend using openocd that ships with STM32CubeIDE.

haneefdm, 2022-06-11:
Ubuntu 20.4 is also special. News has been circulating that `oomd` is killing processes too early and not necessarily the right one or the right reasons. It (sometimes falsely) thinks system is running out of memory and kills processes. It is known to mercilessly kill VSCode, Firefox, etc and the desktop env. is no exception

https://lists.ubuntu.com/archives/ubuntu-devel/2022-June/042116.html
https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1972159

You can try disabling oomd

https://askubuntu.com/questions/1404888/how-do-i-disable-the-systemd-oom-process-killer-in-ubuntu-22-04

I have a feeling your SWAP size is too small.

haneefdm, 2022-06-11:
Check your system logs

See also, https://bugzilla.redhat.com/show_bug.cgi?id=1941170
> Seems like if the memory gets filled fast enough, systemd will even decide to kill Gnome

FL0WL0W, 2022-06-11:
Thank you for all of the replys. I updated the main post with the cortex-debug version. 

I have been having this problem since last year so i have tried multiple vscode versions and they all have this issue.

As far as the openocd version, I did notice that it was the esp32 version and remembered i had setup esp32 environment to automatically setup when opening new terminals. I removed this and built and installed the latest from the openocd git repository.

I don't think my memory is getting filled up considering i have 64GB and this crash happens even if vscode is the only thing open.

I also discovered that i do not have to close vscode to get it to crash. If i close the gdbserver terminal within vscode, the crash happens.

haneefdm, 2022-06-11:
Good it is not a memory thing. We have many many Ununtu users. Never heard of this issue.

> I also discovered that i do not have to close vscode to get it to crash. If i close the gdbserver terminal within vscode, the crash happens.

While a debug session is running? Also, another Window managed by VSCode, we only add text to it.

What other processes (VSCode terminal) are running at the time?

Nothing jumped out at me in the vscode log yet. The last few messages are not in our module (anything with 

**What about linux logs -- like syslog? If a display server is taken down, there should be messages in linux logs.**

haneefdm, 2023-04-14:
Stale. We don't thing this is a problem with our extension.

closed 2023-04-14 by haneefdm