Marus/cortex-debug #679: Using Cortex-Debug with VScode Remote-Container with JLink

basprins, 2022-06-17:
Hi, I am having a hard time figuring out how to debug when I setup my dev env in docker.

I am running ubuntu 22.04, with latest docker. Build my own docker image which roughly consists of:
- FROM ubuntu
- the gcc-arm-none-eabi 10.3 toolchain
- cmake (latest)
- jlink (latest)
- and a bunch of dependencies I need to run JLinkGDBServerCLExe in my container

I start my docker container with this command
```
sudo docker run -it --privileged --network host -v /dev/bus/usb/:/dev/bus/usb -v /run/udev:/run/udev:ro --mount type=bind,source="/home/bp/dev/iobox",target=/home/dev/iobox ubuntu-jlink
```

I can connect to the CPU from my docker container

```
root@bp-legion:/# JLinkExe -device MIMXRT1024XXX5A
SEGGER J-Link Commander V7.66c (Compiled Jun 15 2022 16:51:26)
DLL version V7.66c, compiled Jun 15 2022 16:51:11

Connecting to J-Link via USB...O.K.
```

But when I launch my project from VScode it stops without too much feedback.

[image]

My Cortex-Debug launch config :
```
    {
      "name": "iobox-1024",
      "type": "cortex-debug",
      "request": "launch",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceRoot}/iobox-1024/build/iobox-1024.axf",
      "serverpath": "/usr/bin/JLinkGDBServerCLExe",
      "windows": {
        "serverpath": "C:\\Program Files (x86)\\SEGGER\\JLink\\JLinkGDBServerCL.exe",
      },
      "osx": {
        "serverpath": "/Applications/SEGGER/JLink_V764e/JLinkGDBServerCLExe",
      },
      "servertype": "jlink",
      "device": "MIMXRT1024xxx5A",
      "preLaunchTask": "iobox-1024 - build",
      "interface": "swd",
      "runToEntryPoint": "main",
      "showDevDebugOutput": "raw",
      "showDevDebugTimestamps": true,
    },   
```

I found several other issues on this topic here, but whatever I try, I can't seem to get through. The only thing I did manage, is to connect `JLinkExe -ip <hostipaddress>` from my container to the JLinkRemoteServer. But I don't think that will get me much further. 

Last finding which might reveal something. When I run `JLinkGDBServerCLExe -device MIMXRT1024XXX5A -ip <hostip>` either from my docker container, or on the ubuntu host itself, it always complains about not being able to *open* port 2331. 

```
Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V11 compiled May 30 2022 15:54:09
Hardware: V11.00
S/N: 601011684
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 3.32 V
ERROR: Failed to listen at socket (Err = -1)
ERROR: Failed to open listener port 2331
Restoring target state and closing J-Link connection...
Shutting down...
Failed to open listener port 2331
```

Questions:
- Can I get more detailed logging of what exactly goes wrong when I start the debug session from VScode? The last line in the debug output I show above is `0000000018: 1-gdb-version`. 
- Do I need to rely on anything on the host? I'd expect that everything could be handled in my docker container. It can access the JLink probe fine. So I don't really need anything on the host then, right? 
- If the previous question is not true, am I correct that the only way forward is to host the `JLinkRemoteServer` on my host, and should Cortex-Debug fire up `JLinkGDBServerCLExe - ip <hostipaddress>` ? The gdb client Cortex-Debug fires up, where does that run exactly? Also in the container, right?

Many thanks in advance !!

haneefdm, 2022-06-17:
First, I have no experience with docker.

You can run gdb yourself (**exact** command) in the docker env. and run the same command we ran. It is not responding to the very first command we issue. We are asking gdb, what is your version? Gdb never responds and we continue to wait.

Note that while VSCode is running on Windows, it knows about docker and runs our debugger on the docker side. At least that is way it is on WSL2.

You can run a  remote access server (different from JLinkGDBServerCLExe) on the host that can serve other machines/virtual environments.

https://www.segger.com/products/debug-probes/j-link/tools/j-link-remote-server/
https://wiki.segger.com/Eclipse#Connecting_to_J-Link_via_J-Link_Remote_Server

Beyond that, I don't know. Search in the list of issues here. I saw someone post something as to how to use this kind of setup.

I recommend running everything using command-lines and then figure out how to configure cortex-debug/vscode.

haneefdm, 2022-06-17:
We started three external commands. objdump, nm and gdb. Looks like none of them are returning anything -- not sure where their stdout is going. Hopefully, the are not running on the host. They should be running in the remote (docker) environment. Ie, if you opened the Folder in VSCode as a Docker folder. VSCode know about docker and knows (for the most part) what runs on the host and what runs remotely. Our extension was classified by them as "remote".

basprins, 2022-06-18:
Ah should've though of that myself... Apologies. running arm-none-eabi-gdb from command line complains about the missing infamous libncurses.so.5. Easy fix. Works like a charm now.

Thx!!

closed 2022-06-18 by basprins

jnz86, 2023-10-27:
> Ah should've though of that myself... Apologies. running arm-none-eabi-gdb from command line complains about the missing infamous libncurses.so.5. Easy fix. Works like a charm now.
> 
> Thx!!

I'm sure this isn't "exactly" the right place to bring up this old topic... But I didn't follow entirely here. Was the problem that your debug extension was not inside the container? 

I'm setting exactly this up now. So if you found any other pain points, knowing about them would surely save me some time!