Marus/cortex-debug #680: STM32F1 Hangs waiting for GDB

gigaj0ule, 2022-06-19:
I had been using cortex-debug with the STM32F1x series MCU for a few years, but as of 2020 it seems to have ceased working. 

I've been puzzled for days trying to sort out what is wrong. My .json is nothing special...

`  
{
            // For the Cortex-Debug extension
            "type": "cortex-debug",
            "servertype": "openocd",
            "request": "launch",
            "name": "Debug F103 (STM32)",
            "executable": "${workspaceRoot}/__build/fw.elf",
            //"device": "STM32F103C8",
            "configFiles": [
                "interface/stlink.cfg",
                "${workspaceRoot}/system32/bootloader/_build_tools/openocd/stm32f1x.cfg"
            ],
            "cwd": "${workspaceRoot}",
            "showDevDebugOutput": true,
            "svdFile": "${workspaceRoot}/system32/_debug_tools/STM32F103.svd"
}
`

And yet, when I attempt to launch, openOCD hangs waiting for gdb

`
[2022-06-19T01:05:57.698Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate

Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.392777
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f1x.cpu on 50000
Info : Listening on port 50000 for gdb connections
[2022-06-19T01:05:57.955Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
`

The debug console is as such:

`
Reading symbols from '/mnt/disk0/Users/user/Nextcloud/PC-SYNC/Projects/_source_codes/github/stm42/__build/fw.elf'
Finished reading symbols
Please check TERMINAL tab (gdb-server) for output from openocd
Launching server: "openocd" "-c" "gdb_port 50000" "-c" "tcl_port 50001" "-c" "telnet_port 50002" "-s" "/mnt/disk0/Users/user/Nextcloud/PC-SYNC/Projects/_source_codes/github/stm42" "-f" "interface/stlink.cfg" "-f" "/mnt/disk0/Users/user/Nextcloud/PC-SYNC/Projects/_source_codes/github/stm42/system32/bootloader/_build_tools/openocd/stm32f1x.cfg" "-c" "reset_config none separate"
Launching GDB: "arm-none-eabi-gdb" "-q" "--interpreter=mi2"
 "/mnt/disk0/Users/user/Nextcloud/PC-SYNC/Projects/_source_codes/github/stm42/__build/fw.elf"1-gdb-set target-async on
2-interpreter-exec console "source /home/user/.vscode/extensions/marus25.cortex-debug-0.4.10/support/gdbsupport.init"
3-interpreter-exec console "set print demangle on"
4-interpreter-exec console "set print asm-demangle on"
5-target-select extended-remote localhost:50000
6-interpreter-exec console "monitor reset halt"
7-target-download
8-interpreter-exec console "monitor reset halt"
9-enable-pretty-printing
`

After I launch, fail, and then timeout, I need to manually kill the openocd process with 

`
killall openocd
`

I'm quite puzzled as I don't have any more visibility into what is failing. 

Has anyone encountered this before?

gigaj0ule, 2022-06-19:
On a hunch I reinstalled the plugin completely, and now it fails even sooner. I don't even get a terminal output, instead, just:

```
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /mnt/disk0/Users/user/Nextcloud/PC-SYNC/Projects/_source_codes/github/stm42/__build/fw.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /mnt/disk0/Users/user/Nextcloud/PC-SYNC/Projects/_source_codes/github/stm42/__build/fw.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 /mnt/disk0/Users/user/Nextcloud/PC-SYNC/Projects/_source_codes/github/stm42/__build/fw.elf
1-gdb-set target-async on
```

haneefdm, 2022-06-19:
Chances are you are missing libncurses on your system (if Linux). Run the exact gdb command in a terminal to see if it even runs.

As you can see, we are sending commands to gdb with zero responses.

gigaj0ule, 2022-06-19:
Wow, that was a quick fix.

```sudo apt install libncurses5```

Is the output of the Debug terminal captured and reported somewhere?

If I had seen 

```
arm-none-eabi-gdb: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
```

as an error it would have quickly helped me solve the problem. unfortunately, I had no idea where it was getting stuck.

haneefdm, 2022-06-19:
Thing is, we don't get the error (on stdout or stderr) as that is an OS thing. I wish we did get something. Every day, we get one of these issues from Linux folks. A bug should really be submitted to whoever builds/distributes this gdb package.

closed 2022-06-19 by haneefdm