Marus/cortex-debug #681: Can not load SVD file when start debugging

manadc0, 2022-06-19:
Hello,
The CORTEX PERIPHERALS tab usually can not show the list of register for MCU. I dont know why because sometime I can see it but almost SVD (xml) can not load
[image]

My launch.json and xml
`{
    "version": "0.2.0",
    "configurations": [
            {
            "cwd": "${workspaceRoot}",
            "executable": "./armgcc/debug/gpio_led_output.elf",
            "name": "Debug JLink",
            "request": "launch",
            "type": "cortex-debug",
            "serverpath": "C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe",
            "servertype": "jlink",
            "armToolchainPath": "C:/Compiler/gcc-arm-none-eabi-10.3-2021.10-win32/bin",
            "runToMain": true,
            "svdFile": "./MK64F12.xml",
            "device": "MK64FN1M0VLL12",
            "interface": "swd",
        }
    ]
}`
[MK64F12.zip](https://github.com/Marus/cortex-debug/files/8934726/MK64F12.zip)

haneefdm, 2022-06-19:
Please try version 1.5.x (available in VSCode as a pre-release in the extensions section).

manadc0, 2022-06-19:
> Please try version 1.5.x (available in VSCode as a pre-release in the extensions section).

Yeah, It was fixed with pre-release. Thank you!

closed 2022-06-19 by manadc0