Marus/cortex-debug #684: Unable to parse svd file: TypeError: Cannot read property 'map' of undefined

vasundhara-bhat, 2022-06-22:
Issue Type: <b>Bug</b>

Hello, 

I would like to request your support regarding an issue we are facing. In the debug session, the SVD file is not being parsed and displays the error: TypeError: Cannot read property 'map' of undefined.

This had worked before and it has stopped working recently. We are not sure if this could be due to the upgradation of VS code.

Thank you in advance.

Best Regards,
Vasundhara Bhat

Extension version: 1.2.2
VS Code version: Code 1.65.2 (c722ca6c7eed3d7987c0d5c3df5c45f6b15e77d1, 2022-03-10T14:33:55.248Z)
OS version: Windows_NT x64 10.0.19042
Restricted Mode: No

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-10210U CPU @ 1.60GHz (8 x 2112)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>rasterization: enabled<br>skia_renderer: enabled_on<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|7.78GB (1.48GB free)|
|Process Argv|--crash-reporter-id ad29f819-20ac-411c-aa08-dffb464c1104|
|Screen Reader|no|
|VM|0%|
</details>

vasundhara-bhat, 2022-06-23:
Hi, Thank you for your answer. 
We have tried the pre release version. We notice that the problem comes for a specific svd file(belongs to TI). Other svds have been parsed with no issue.
But we are not able locate the problem in this perticular svd file.

Thanks

haneefdm, 2022-06-23:
Can you compress and attach the svd file?

haneefdm, 2022-07-23:
No response in almost a month. Closing the issue as we are cleaning up the queue

closed 2022-07-23 by haneefdm