Marus/cortex-debug #685: Feature request: RTOS ChibiOS support

martantoine, 2022-06-24:
I have been using this extension for dev on stm32 with ChibiOS and it has been already a pleasure to debug compared to eclipse based IDEs. 
However there is a feature that I miss a lot which is threads stats.
Now there is a RTOS tab in Cortex-Debug, I am wondering how hard it would be to support ChibiOS and if you've planned to add a such feature.

haneefdm, 2022-06-24:
Because each OS requires intimate knowledge of the OS and its data structures itself, we are looking for volunteers. We need two things

1. A volunteer to add support for said OS. We have one such volunteer add support for uCOSII already and there are now two templates you can follow. Yes, knowledge of typescript is needed.
2. A test case using an STM32 board I own.

A reference implementation (like for Eclipse) in source form would be nice as well. For instance, source code for this

https://www.chibios.org/dokuwiki/doku.php?id=chibios:documentation:books:rt:kernel_debug#kernel_statistics

martantoine, 2022-06-24:
Unfortunately I don't have the required deep ChibiOS nor typescript knowledge
However, I too have stm32 boards with different debug protocols (STLink and BMP) so I could help with the testing.

Regarding the already supported RTOS, I don't understand how it works
For instance, how does the cortex extension knows thread priority, does it read MCU's memory at known location through GDB and then deduce thread stats ?

haneefdm, 2022-06-24:
We examine the RTOS data structures whenever the program pauses. Yes, we find out via gdb, where important data structures are, their format/fields, and bit by bit traverse them. Then we have to accommodate for missing fields, global variables, etc.

You might want to spend some time looking at the stuff. Even if you don't know Typescript/JavaScript, you will probably understand. I know it is not easy, but might be entertaining :-)

Look here: https://github.com/Marus/cortex-debug/tree/master/src/frontend/rtos I tried to make things as open as I can think of. But I need help. I am all alone.

If you look at the [FreeRTOS](https://github.com/Marus/cortex-debug/blob/master/src/frontend/rtos/rtos-freertos.ts), there are so many optional things in FW (#ifdefed out) so we have to guard ourselves from things not available.

A thought, you could ask the ChibiOS people to help out. I hate to reverse engineer their stuff and get it wrong. They could contribute, you know. Or help us, if they are truly open-source and free and interested.

See how one OS got integrated, I am sure the next one will be easier because of @PhilippHaefele  and his team.

https://github.com/Marus/cortex-debug/pull/642

vrepetenko, 2022-09-06:
Hi!
I can try to add ChibiOS support, need some time to look at FreeRTOS implementation and how to debug plug-in.

haneefdm, 2022-09-06:
See this thread

https://github.com/Marus/cortex-debug/issues/732

and if you have general questions, you can ask over there, so it can be helpful for both and others can answer as well.

haneefdm, 2022-10-21:
The RTOS functionality is moving to a new repo and @vrepetenko generously contributed ChibiOS to it. You will see a new extension published soon. Work is ongoing.

https://github.com/mcu-debug/rtos-views

I am closing this issue here.

closed 2022-10-21 by haneefdm