Marus/cortex-debug #687: Having trouble setting breakpoints while target is running

jwpwh, 2022-06-30:
Basic scenario: 
* Start gdbserver on the remote target
* Attach to gdbserver in Visual Studio Code with stop at main. (line 16)
* Stops at main. Hit continue
* Add a breakpoint at line 18 in main.cpp. I don't expect it to be hit but it should be planted.
  * Get a popup message in lower right corner: 
<pre>
Cannot execute this command while the target is running.
Use the "interrupt" command to stop the target and then try again. (from break-insert -f "/export/space/jpruitt/soc_camera_sw/ARM/camera_software/src/main.cpp:18")
</pre>
* Do "i b" in the debug window. No breakpoints are planted. 
* Visual Studio Code shows a breakpoint at being present.

Also have trouble asking to "Stop" while target is running. This leaves the remote gdbserver in a funny state. 

Since there have been similar issues in the past and the problems were resolved, I am hopeful that something can be done here. Maybe it is due to the remote gdbserver being used.  The debug log doesn't seem to show the commands being sent to the gdbserver but maybe the responses are enough. I thought I saw an issue where the extension was stopping and restarting the target but that doesn't seem to being done here..

Any suggestions would be appreciated.

Thanks

Information:

* Extension version: 1.5.1
* Visual Studio Code version: 

<pre>
Version: 1.68.1 (user setup)
Commit: 30d9c6cd9483b2cc586687151bcbcd635f373630
Date: 2022-06-14T12:48:58.283Z
Electron: 17.4.7
Chromium: 98.0.4758.141
Node.js: 16.13.0
V8: 9.8.177.13-electron.0
OS: Windows_NT x64 10.0.19043
</pre>

* remote gdbserver version

<pre>
GNU gdbserver (GDB) 8.0
Copyright (C) 2017 Free Software Foundation, Inc.
gdbserver is free software, covered by the GNU General Public License.
This gdbserver was configured as "arm-poky-linux-gnueabi"
</pre>

* launch.json

<pre>
{
    "configurations": [
     {
        "type": "gdb",
        "request": "attach",
        "name": "Attach to gdbserver",
        "executable": "ARM/camera_software/.libs/camera_application",
        "target": "localhost:10000",
        "stopAtEntry": true,
        "showDevDebugOutput": true,
        "setupCommands": [
            { "text": "set solib-search-path ARM/camera_software/src/Sensors/U35MGXS/.libs;ARM/camera_software/src/Sensors/CMV300/.libs"}
        ],
        "remote": true,
        "cwd": "${workspaceRoot}", 
        "gdbpath": "/opt/poky/2.4.3/sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gdb"
    }]
}
</pre>

* Debug log
[debug.log](https://github.com/Marus/cortex-debug/files/9020544/debug.log)

There are other popup warnings when starting the debug session and getting to main:

<pre>
Failed to get Stack Trace: Selected thread is running. (from stack-info-depth --thread 1)
</pre>

<pre>
Could not get threads: Cannot execute this command while the target is running.
Use the "interrupt" command to stop the target
and then try again. (from thread-info)
</pre>

haneefdm, 2022-06-30:
You are using this extension to debug a native app? This designed for embedded apps. cpptools fro microsoft.  is more appropriate for you. Your launch.json appears to be nor for our debugger but the some of the messages are. So, I am confused.

You may want to use `"showDevDebugOutput": "raw"` for better output with our debugger which needs `"type": "cortex-debug"

With a bit of massaging I am sure you can use our debugger for native apps. When we first attach to a program, we expect the program to halt after we issue commands to the server. Your can try to use `overrideAttachCommands` to make the program halt (I believe this is default, so you can put some dummy commands in `overrideAttachCommands` There is a log of customization you can do tailor our debugger to your gdbserver.

jprui, 2022-06-30:
This is not for a native app. It is a remote app. I will try your suggestions though and see if they help.

Thanks for the quick response.

haneefdm, 2022-06-30:
Sorry, "Native" as in runs on a desktop in a proper OS., Our debugger is used to debug stuff like  IoT, pacemakers, etc where programs run in a Microcontroller. Technically, all the stuff we debug is "remote" because none of the programs can run in an OS.

The following tells me that you are using [Microsoft's debugger](https://github.com/microsoft/vscode-cpptools)
```
        "type": "gdb",
```
That is not us.

haneefdm, 2022-06-30:
Many years ago, we started with WebFreak's version and adapted it for embedded stuff. Which is why some messages look familiar. We diverged so much, there is very little common now between us.

closed 2022-06-30 by haneefdm