Marus/cortex-debug #688: Unable to start a debug session using `"servertype": "external"`, running `ST-LINK_gdbserver.exe`

daantimmer, 2022-07-01:
After connecting (and it seems like flashing the MCU does work) but afterwards I get:
> Failed to launch GDB: Protocol error with Rcmd (from interpreter-exec console "monitor reset halt")

I've tried launching `ST-LINK_gdbserver.exe` in two ways:
```C:\\ST\\STM32CubeIDE_1.10.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_2.0.300.202203231527\\tools\\bin\\ST-LINK_gdbserver.exe -p 50000 -cp C:\\ST\\STM32CubeIDE_1.10.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.0.300.202205161323\\tools\\bin --swd --halt```
and
```C:\\ST\\STM32CubeIDE_1.10.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_2.0.300.202203231527\\tools\\bin\\ST-LINK_gdbserver.exe -p 50000 -l 1 -d -s -e -cp C:\ST\STM32CubeIDE_1.10.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.0.300.202205161323\tools\bin -m 0 -k ```

The first entry is the way that cortex-debug launches ST-LINk server, the second one is something random I found by googling.

I can see that cortex-debug connects to the server, it flashes the device, and then it gives the failed to launch error.

daantimmer, 2022-07-01:
We(/I) are in the process of moving all development directly from within Docker. That is, using the 'clone repository in docker' option from vscode.

From this remote container there is no way to access attached USB devices for programming and debugging. That's why we need to use a TCP/IP connection out of the container in to a GDB server.

haneefdm, 2022-07-01:
When it is "external", we have no idea what server/executable you are running. How would we? It could even be something we never knew about. We just let gdb connect to a socket that can even be another machine as well. We are not even involved in the actual connection process.

I edited the documentation to add the missing `-`. Thanks.

In our launch commands for ST-Link, we actually have four commands, the last one being `monitor halt` which may not be needed but we do it anyway. Just FYI

closed 2022-07-01 by haneefdm

daantimmer, 2022-07-01:
Perhaps an idea to allow setting external type somewhere? You (the
extension) might not know what the external server type is. But usually we
do.

haneefdm, 2022-07-01:
I will think about it.