Marus/cortex-debug #689: Debug session terminates unexpectedly when rapidly single-stepping

bssc2g, 2022-07-01:
Apologies if this is not a super clear cut bug report, still trying to figure out the trigger/gather information for helping to diagnose.

It seems that if you try to single step too quickly, the debug session will just terminate unexpectedly in some cases. I suspect it has something to do with trying to single step before the watch variables have all been updated. 

Debug Probe: Jlink
Host: Windows 10
Cortex Debug version: 1.5.1 pre
vscode version: 1.68.1
Target: NRF52840

Steps to recreate? 

- Launch debug session
- Add a "bunch" of variables to the watch window. In my case, I had ~10 variables plus an array of 256 uint32s in the watch window. 
- Single step rapidly or immediately after reaching a breakpoint before the watch panes update fully.

It doesn't happen in every single instance, sometimes I need to step many times in rapid succession for it to happen.

haneefdm, 2022-07-01:
Could you attach the contents of the Debug Console? Enable debug of course as shown in the Debug Console.

A debug session terminates unexpectedly for one of three reasons known to me
* gdb crashed
* JLink gdb server crashed
* cortex-debug debug adapter crashed.

Even if the Watch window updates did not finish, it should not crash. Some variables will look like 'not available' or some such message.

It helps to know what happened and the contents of the Debug Console.

haneefdm, 2022-07-12:
Nope thats. You can set to `showDevDebugOutput` and you will also see requests coming to us from VSCode and our responses. Can be very verbose.

In your %TEMP% or $TMPDIR  or $TMP (depends on system), there is also a file called `cortex-debug-server-exiting.log` You can see which processes are exiting -- at least the order in which they die/exit. If you don't see anyone exiting, then it will be us who died first.

bssc2g, 2022-07-13:
Here is a snippet from the most recent "crash" of the debug session, from the cortex-debug-server-exiting.log:

`[2022-07-13T20:46:13.591Z] ppid=35500 pid=18312 Debug Adapter killed by VSCode? code=1 signal=undefined
[2022-07-13T20:46:13.591Z] ppid=35500 pid=18312 GDBServer(37204): forcing an exit with kill()`

haneefdm, 2022-07-13:
The only time our extension runs anything is when VSCode makes a request. And, if we crash during that request, it is not fatal as every request is guarded by a try/catch -- I think.

`signal=undefined` -- means it did not get killed. It actually exited with exit code 1. I have no idea under what circumstances it can happen.

Scratching head....what was the last command(s) you saw in the Debug Console. If anything happened, it is probably because of that.

Can you post the last few lines of the Debug Console and the same for cortex-debug-server-exiting.log?

haneefdm, 2022-07-13:
When putting multi-line stuff, could you enclose them in a code block so that new lines are preserved?

https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

bssc2g, 2022-07-18:
Hey, sorry for the delay, it took a while for it to happen again (and I do have work I need to do in the meantime :) ) 

As requested, the end of the debug output and server-exiting.log.

```
-> 525^error,msg="-var-create: unable to create variable object"
watch -var-create: unable to create variable object (from var-create  watch_3115e5f65aeb12e3f805775eb9dae51d0c1b224d140891c80128eb569348ed28 @ "default_data")526-var-list-children --all-values watch_a661c02be3e4c8254730a0e619a411b294d637605ea85c070acda245bcaa78b9
-> 526^done,numchild="8",children=[child={name="watch_a661c02be3e4c8254730a0e619a411b294d637605ea85c070acda245bcaa78b9.key",exp="key",numchild="1",value="0x71f0000 \"\"",type="const char *"},child={name="watch_a661c02be3e4c8254730a0e619a411b294d637605ea85c070acda245bcaa78b9.id",exp="id",numchild="0",value="EP_RECORD_ID_RESERVED",type="earpiece_record_id_t"},child={name="watch_a661c02be3e4c8254730a0e619a411b294d637605ea85c070acda245bcaa78b9.scope",exp="scope",numchild="0",value="EP_RECORD_PUBLIC",type="earpiece_record_scope_t"},child={name="watch_a661c02be3e4c8254730a0e619a411b294d637605ea85c070acda245bcaa78b9.datatype",exp="datatype",numchild="0",value="(EP_RECORD_DATATYPE_BOOL | unknown: 0x28)",type="earpiece_record_datatype_t"},child={name="watch_a661c02be3e4c8254730a0e619a411b294d637605ea85c070acda245bcaa78b9.volatility",exp="volatility",numchild="0",value="(EP_RECORD_VOLATILE | unknown: 0x6)",type="earpiece_record_volatility_t"},child={name="watch_a661c02be3e4c8254730a0e619a411b294d637605ea85c070acda245bcaa78b9.permissions",exp="permissions",numchild="0",value="EP_RECORD_RONLY",type="earpiece_record_permission_t"},child={name="watch_a661c02be3e4c8254730a0e619a411b294d637605ea85c070acda245bcaa78b9.element_count",exp="element_count",numchild="0",value="0x0",type="unsigned int"},child={name="watch_a661c02be3e4c8254730a0e619a411b294d637605ea85c070acda245bcaa78b9.default_data",exp="default_data",numchild="5",value="{...}",type="union {...}"}],has_more="0"
527-var-list-children --all-values watch_5f9dc5e31c635a66324c61b82857aa7dfacf1b091f481bc33677d8df445e240a
-> 527^done,numchild="1",children=[child={name="watch_5f9dc5e31c635a66324c61b82857aa7dfacf1b091f481bc33677d8df445e240a.*&rec_desc->default_data.int_arr",exp="*&rec_desc->default_data.int_arr",numchild="1",value="0x0",type="uint32_t *"}],has_more="0"
528-stack-info-frame --thread 1 --frame 0
-> 528^done,frame={level="0",addr="0x0002c458",func="process_raw_impl",file="<redacted>_handler.c",fullname="<redacted>_kv_handler.c",line="253",arch="arm"}
529-exec-next --thread 1
-> 529^running
-> *running,thread-id="all"
mi2.status = running
530-stack-select-frame --thread 1 0
-> 530^error,msg="Selected thread is running."
531-stack-list-variables --thread 1 --frame 0 --simple-values
```

Below is the relevant output from cortex-debug-server-exiting.log:

```
[2022-07-18T17:40:31.171Z] ppid=20376 pid=3212 GDB started ppid=20376 pid=3212
[2022-07-18T17:43:08.892Z] ppid=20376 pid=3212 Debug Adapter killed by VSCode? code=1 signal=undefined
[2022-07-18T17:43:08.893Z] ppid=20376 pid=3212 GDBServer(17884): forcing an exit with kill()
```

haneefdm, 2022-07-18:
Thanks. I may have found the problem that caused our extension to crash. No one killed it, it just crashed.

I can give you a private build for you to try out. Are you willing? It will take a day or so to do some cleanup and provide you a build.

haneefdm, 2022-07-23:
Could you try the latest pre-release?
[image]

Can you enable the pre-release and see if this issue is still happening. Please report back and I can re-open the issue

closed 2022-07-23 by haneefdm