Marus/cortex-debug #690: Unable to start latest GDB Server

tapioreijonen, 2022-07-05:
When using [gcc-arm-none-eabi-10.3-2021.10-win32.exe](https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-win32.exe) the GDB debug session start as expected.

When using latest [gcc-arm-11.2-2022.02-mingw-w64-i686-arm-none-eabi.exe](https://developer.arm.com/-/media/Files/downloads/gnu/11.2-2022.02/binrel/gcc-arm-11.2-2022.02-mingw-w64-i686-arm-none-eabi.exe) the GDB debug sessio does not start. Here is copy/paste from the debug console:
Reading symbols from C:\Program Files (x86)\Arm GNU Toolchain arm-none-eabi\11.2 2022.02\bin\arm-none-eabi-objdump.exe --syms -C -h -w C:\Projects\git-aws\git-aws-clean\tier2-wxt-aqt\build-bmu801\gcc\projects\BMU801\tier2-wxt-aqt.elf
Reading symbols from c:/program files (x86)\arm gnu toolchain arm-none-eabi\11.2 2022.02\bin\arm-none-eabi-nm.exe --defined-only -S -l -C -p C:\Projects\git-aws\git-aws-clean\tier2-wxt-aqt\build-bmu801\gcc\projects\BMU801\tier2-wxt-aqt.elf
Launching GDB: "C:\\Program Files (x86)\\Arm GNU Toolchain arm-none-eabi\\11.2 2022.02\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2 "C:\\Projects\\git-aws\\git-aws-clean\\tier2-wxt-aqt\\build-bmu801\\gcc\\projects\\BMU801\\tier2-wxt-aqt.elf"
    Set "showDevDebugOutput": true in your "launch.json" to see verbose GDB transactions here. Helpful to debug issues or report problems
Launching gdb-server: "C:\\Program Files\\SEGGER\\JLink\\JLinkGDBServerCL.exe" -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device R7FA4M2AD
    Please check TERMINAL tab (gdb-server) for output from C:\Program Files\SEGGER\JLink\JLinkGDBServerCL.exe
undefined GDB Server Quit Unexpectedly. See gdb-server output for more details.

Debug Probe: Jlink (V7.62C)
Host: Windows_NT x64 10.0.19044
Cortex Debug version: 1.4.4
vscode version: 1.68.1 
Target: R7FA4M2AD

haneefdm, 2022-07-05:
You can see the exact GDB command line and the command line for JLinkGDBServerCL.

Looks like a bad/incomplete install of gdb.

Run them in a Terminal/Cmd.exe and see what you get.

tapioreijonen, 2022-07-05:
After more investigating of this issue I find this:

- GNU Arm Embedded Toolchain\10 2021.10\bin>"arm-none-eabi-gdb.exe" started as expected

C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin>"arm-none-eabi-gdb.exe"
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb)

- C:\Program Files (x86)\Arm GNU Toolchain arm-none-eabi\11.2 2022.02\bin>arm-none-eabi-gdb.exe
Does not start, this gdb reguires python27.dll ???

More, see https://community.arm.com/support-forums/f/compilers-and-libraries-forum/52585/python-2-7-dependency-for-arm-none-eabi-gdb-at-windows-hosts

tapioreijonen, 2022-07-05:
See my previous comment

closed 2022-07-05 by tapioreijonen

gregstarr, 2022-07-09:
I have the same problem but for openocd