Marus/cortex-debug #691: Support for multiple Segger RTT "terminals" ?

dsabala, 2022-07-05:
Hi

Is there support for Segger RTT "terminals"?

My `launch.json`
```
{
    "version": "0.2.0",
    "configurations": [
      {
      // "showDevDebugOutput": "both",
      "type": "cortex-debug",
      "request": "launch",
      "name": "Debug J-Link",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceRoot}/build-gcc-arm-Debug/ecu-app.elf",
      "serverpath": "JLinkGDBServerCLExe", 
      "servertype": "jlink",
      "device": "ATSAMV71",
      "interface": "swd",
      "rttConfig": {
        "enabled": true,
        "address": "auto",
        "decoders": [
            {
                "label": "/dev/stdout",
                "port": 0,
                "type": "console",
                "noprompt": true,
            },
        ]
      },
      "runToEntryPoint": "_start",
      "jlinkscript": "debug.jlinkscript"
      }
      ]
}
```

I would like to add second decoder like this
```
            {
                "label": "/dev/stderr",
                "port": 0,
                "type": "console",
                "noprompt": true,
            },
```
and then handle both stdout device and stderr device of my RTOS, using firmware function `SEGGER_RTT_SetTerminal`.
Is it possible to have two terminals handled by `cortex-debug` ? By the way, you are doing great work guys ;)

haneefdm, 2022-07-05:
Sorry, No. If you use terminals, you can see the terminals changing (when SEGGER_RTT_SetTerminal() is called) in the window but we don't have separate tabs for terminals. Creating dozens of windows in VSCode is complex and a lot of work. Even the API in FW is weird because it is modal. So, no not the way JLink viewers see it but you can see terminals switching in your regular output.

A better way is to use multiple channels but JLink only provides support for a single channel via the GDB server. With OpenOCD, multiple channels is not a problem.

dsabala, 2022-07-06:
@haneefdm 

Ok, thank you, I will consider using OpenOCD

closed 2022-07-06 by dsabala