Marus/cortex-debug #692: Cannot debug Nucleo-G491RE

ChrisIdema, 2022-07-05:
I've created a simple blinky project for the Nucleo-G491RE with STM32CubeMX for makefile target. I created a Cmake project in VsCode and I can build the blinky project and I can flash the elf file with STM32CubeProgrammer and it works. But I cannot seem to get debugging working. When I hit debug the debug button bar briefly appears and then quickly dissapears without any error message. This is the message in the Debug Console:

> Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/user/git/stm32-template/../build-stm32-template/cmake-blinky.elf
> Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/user/git/stm32-template/../build-stm32-template/cmake-blinky.elf
> Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 /home/user/git/stm32-template/../build-stm32-template/cmake-blinky.elf
> 1-gdb-set target-async on
> 

This is my settings file:

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug Nucleo openocd",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/../build-stm32-template/cmake-blinky.elf",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            // "preLaunchTask": "Build all",
            "device": "STM32G491RE",
            "showDevDebugOutput": "both",
            "servertype": "openocd",
            "interface": "swd",
            // svd file
            "configFiles": [
                "interface/stlink-v2-1.cfg",
                "target/stm32g4x.cfg"
            ],
            "swoConfig": {
                "enabled": false
            }
        },
        {
            "name": "Debug Nucleo st util",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/../build-stm32-template/cmake-blinky.elf",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            // "preLaunchTask": "Build all",
            "device": "STM32G491RE",
            "showDevDebugOutput": "both",
            "servertype": "stutil",
            "interface": "swd",
            // svd file
            "swoConfig": {
                "enabled": false
            }
        }
    ]
}
```

haneefdm, 2022-07-05:
Can you make sure you can run gdb from the command-line just like we are doing? Newer gdb versions doesn't come fully packaged and have missing DLLS/so's

haneefdm, 2022-07-06:
Aren't you running a different gdb executable from what you gave us? Debug Gonsole says
```
arm-none-eabi-gdb -q --interpreter=mi2 /home/user/git/stm32-template/../build-stm32-template/cmake-blinky.elf
```
Not the same program as `gdb-multiarch ~/git/build-stm32-template/cmake-blinky.elf`

If the program even starts, it is a big deal. You don't even have to do anything beyond that.

haneefdm, 2022-07-06:
Yup. Like I suspected. You have an incomplete install of gdb. You need to install libncursesw using whatever package manager you are using. Google it.

The trouble is this error is never seen by us or else we would have reported it. You can only see it if you run the command on the command-line. We have some gdb installs require python which may be missing.

I don't know why they are not dependencies that are auto-installed when you install gdb

closed 2022-07-06 by haneefdm