Marus/cortex-debug #693: SWO terminal not working in VsCode (it does work in the commandline)

ChrisIdema, 2022-07-06:
In the command line I can get SWO working:

My target has a CPU frequency of 170MHz. STMCubeIDE uses 12MHz for SWO, but it works without defining this frequency.
```
openocd -f "interface/stlink.cfg" -f "target/stm32g4x.cfg" -c "transport select hla_swd" -c "tpiu config internal - uart off 170000000" -c "itm ports on"
```

```
arm-none-eabi-gdb -q --interpreter=mi2 ~/git/stm32-template/../build-stm32-template/cmake-blinky.elf
target extended localhost:3333
```

I use https://github.com/robertlong13/SWO-Parser to view the SWO output:
```
python swo_parser.py
```

But in VsCode I cannot. The terminal remains empty. This is my configuration:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug Nucleo openocd",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/../build-stm32-template/cmake-blinky.elf",
            "request": "attach",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "device": "STM32G491RE",
            "showDevDebugOutput": "both",
            "servertype": "openocd",
            "interface": "swd",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32g4x.cfg"
            ],
            "swoConfig": {
                "enabled": true,
                "cpuFrequency": 170000000,
                "swoFrequency":   1200000,
                "source": "probe",
                "decoders":
                [
                    {
                        "label": "ITM port 0 output",
                        "type": "console",
                        "port": 0,
                        "showOnStartup": true,
                        "encoding": "ascii"
                    }
                ]
            }           
        }
    ]
}
```

Sometimes the program hangs in ITM_SendChar:
```
 while (ITM->PORT[0U].u32 == 0UL)
```

ChrisIdema, 2022-07-08:
TERMINAL:
> Waiting for gdb server to start...[2022-07-08T10:58:06.379Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/user/git/stm32-template -f /home/user/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl -f interface/stlink.cfg -f target/stm32g4x.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V3J9M3 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.319251
Info : stm32g4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32g4x.cpu on 50000
Info : Listening on port 50000 for gdb connections
Info : accepting 'gdb' connection on tcp/50000
target halted due to debug-request, current mode: Thread 
xPSR: 0x21000000 pc: 0x0800166a msp: 0x2001bfe0
Info : device idcode = 0x10016479 (STM32G49/G4Axx - Rev 'unknown' : 0x1001)
Info : flash size = 512kbytes
Info : flash mode : single-bank
Info : Unable to match requested speed 2000 kHz, using 1000 kHz
Info : Unable to match requested speed 2000 kHz, using 1000 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08002cbc msp: 0x2001c000
Info : Unable to match requested speed 2000 kHz, using 1000 kHz
Info : Unable to match requested speed 2000 kHz, using 1000 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08002cbc msp: 0x2001c000
Info : Unable to match requested speed 2000 kHz, using 1000 kHz
Info : Unable to match requested speed 2000 kHz, using 1000 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08002cbc msp: 0x2001c000
Info : Unable to match requested speed 2000 kHz, using 1000 kHz
Info : Unable to match requested speed 2000 kHz, using 1000 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08002cbc msp: 0x2001c000
  stm32g4x.cpu tpiu
    stm32g4x.cpu tpiu config (disable | ((external | internal (<filename> | <:port> | -)) (sync <port
              width> | ((manchester | uart) <formatter enable>))
              <TRACECLKIN freq> [<trace freq>]))
tpiu
  tpiu config (disable | ((external | internal (<filename> | <:port> | -)) (sync <port
            width> | ((manchester | uart) <formatter enable>)) <TRACECLKIN
            freq> [<trace freq>]))
**Error: invalid subcommand "init"
/home/user/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl:14: Error: 
in procedure 'CDSWOConfigure' 
at file "/home/user/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl", line 14**

DEBUG CONSOLE:

> Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/user/git/stm32-template/../build-stm32-template/cmake-blinky.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/user/git/stm32-template/../build-stm32-template/cmake-blinky.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 /home/user/git/stm32-template/../build-stm32-template/cmake-blinky.elf
    Set "showDevDebugOutput": true in your "launch.json" to see verbose GDB transactions here. Helpful to debug issues or report problems
Error: arm-none-eabi-nm failed! statics/global/functions may not be properly classified: Error: spawn arm-none-eabi-nm ENOENT
    Expecting `nm` next to `objdump`. If that is not the problem please report this.
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/user/git/stm32-template -f /home/user/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl -f interface/stlink.cfg -f target/stm32g4x.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 28 ms
Finished reading symbols from nm: Time: 27 ms
Reading symbols from /home/user/git/stm32-template/../build-stm32-template/cmake-blinky.elf...
0x08001634 in HAL_GetTick () at /home/user/git/stm32-template/STM32G491RE/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal.c:336
336	}
Program stopped, probably due to a reset and/or halt issued by debugger
Unable to match requested speed 2000 kHz, using 1000 kHz
[... 24 lines cut ...]
> 
> width> | ((manchester | uart) <formatter enable>)) <TRACECLKIN
> 
> freq> [<trace freq>]))
> **invalid subcommand "init"
> /home/user/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl:14: Error:
> in procedure 'CDSWOConfigure'
> at file "/home/user/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl", line 14**
> 
> Program
>  received signal SIGINT, Interrupt.
> Reset_Handler () at /home/user/git/stm32-template/STM32G491RE/startup_stm32g491xx.s:62
> 62	  ldr   r0, =_estack
> Warning: the current language does not match this frame.
> Note: automatically using hardware breakpoints for read-only addresses.
> 
> Temporary breakpoint 1, main () at /home/user/git/stm32-template/STM32G491RE/Core/Src/main.c:90
> 90	  HAL_Init();
> 

So apparently There is an error at line 14 of openocd-helpers.tcl:
```
    tpiu init; # we are allowed to call this multiple times. So, call it just in case
```
I have no idea what this command does.

haneefdm, 2022-07-08:
Instead of commenting it out, could you replace that line with the following
```TCL
    catch {tpiu init}; # we are allowed to call this multiple times. So, call it just in case
```

ChrisIdema, 2022-07-08:
I have commented the line out before and it didn't help. I got errors at other lines, which I also commented out until all errors were gone.

I now replaced it with your catch function and I get:

> Error: invalid command name "catch{tpiu"
in procedure 'CDSWOConfigure'

ChrisIdema, 2022-07-08:
I checked out a different variant of the exact same version. The openocd version from apt is very old so I had it build from source. But somehow that version is missing some functionality. So Instead I downloaded a binary from here: https://github.com/xpack-dev-tools/openocd-xpack/releases/tag/v0.11.0-4 and set "cortex-debug.openocdPath" to the extracted folder.
I now no longer get that error, but I get a different error:

> 
> xPack  OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2022-03-25-17:31)
> 
>  Licensed under GNU GPL v2
>  For bug reports, readd
>         http://openocd.org/doc/doxygen/bugs.html
> CDRTOSConfigure
> Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
> Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
> Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
> Info : Listening on port 50001 for tcl connections
> Info : Listening on port 50002 for telnet connections
> Info : clock speed 2000 kHz
> Info : STLINK V3J9M3 (API v3) VID:PID 0483:374E
> Info : Target voltage: 3.315431
> Info : [stm32g4x.cpu] Cortex-M4 r0p1 processor detected
> Info : [stm32g4x.cpu] target has 6 breakpoints, 4 watchpoints
> Info : starting gdb server for stm32g4x.cpu on 50000
> Info : Listening on port 50000 for gdb connections
> Info : accepting 'gdb' connection on tcp/50000
> target halted due to debug-request, current mode: Thread 
> xPSR: 0x61000000 pc: 0x08004b52 psp: 0x20000e50
> Info : device idcode = 0x10016479 (STM32G49/G4Axx - Rev 'unknown' : 0x1001)
> Info : RDP level 0 (0xAA)
> Info : flash size = 512kbytes
> Info : flash mode : single-bank
> Info : device idcode = 0x10016479 (STM32G49/G4Axx - Rev 'unknown' : 0x1001)
> Info : RDP level 0 (0xAA)
> Info : OTP size is 1024 bytes, base address is 0x1fff7000
> Info : Unable to match requested speed 2000 kHz, using 1000 kHz
> Info : Unable to match requested speed 2000 kHz, using 1000 kHz
> target halted due to debug-request, current mode: Thread 
> xPSR: 0x01000000 pc: 0x08005ea8 msp: 0x2001c000
> Info : Unable to match requested speed 2000 kHz, using 1000 kHz
> Info : Unable to match requested speed 2000 kHz, using 1000 kHz
> target halted due to debug-request, current mode: Thread 
> xPSR: 0x01000000 pc: 0x08005ea8 msp: 0x2001c000
> Info : Padding image section 0 at 0x080001d8 with 8 bytes
> Info : Unable to match requested speed 2000 kHz, using 1000 kHz
> Info : Unable to match requested speed 2000 kHz, using 1000 kHz
> target halted due to debug-request, current mode: Thread 
> xPSR: 0x01000000 pc: 0x08005ea8 msp: 0x2001c000
> Info : Unable to match requested speed 2000 kHz, using 1000 kHz
> Info : Unable to match requested speed 2000 kHz, using 1000 kHz
> target halted due to debug-request, current mode: Thread 
> xPSR: 0x01000000 pc: 0x08005ea8 msp: 0x2001c000
> /**home/user/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl: Error: Could not find TPIU/SWO names. Perhaps it hasn't been created?**

I will try to find where STM32Cube put its openocd version.

ChrisIdema, 2022-07-08:
I found where it is, but I don't know how I can make VsCode find all the cfg files.

> "cortex-debug.openocdPath": "/opt/st/stm32cubeide_1.8.0/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.linux64_2.1.0.202110201500/tools/bin/openocd",

I added the full paths of the cfg files:

> "configFiles": [
> "/opt/st/stm32cubeide_1.8.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.100.202111151130/resources/openocd/st_scripts/interface/stlink.cfg",
> "/opt/st/stm32cubeide_1.8.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.100.202111151130/resources/openocd/st_scripts/target/stm32g4x.cfg"
> ],

But then it cannot find:

> /opt/st/stm32cubeide_1.8.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.100.202111151130/resources/openocd/st_scripts/target/stm32g4x.cfg:6: Error: Can't find **target/swj-dp.tcl**
> in procedure 'script' 
> at file "embedded:startup.tcl", line 26
> at file "/opt/st/stm32cubeide_1.8.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.100.202111151130/resources/openocd/st_scripts/target/stm32g4x.cfg", line 6

haneefdm, 2022-07-08:
The xPack seems more up-to-date by it may still be missing/old config files.
```
/home/user/.vscode/extensions/marus25.cortex-debug-1.4.4/support/openocd-helpers.tcl: Error: Could not find TPIU/SWO names. Perhaps it hasn't been created?
```
So, the config files you used never created the TPIU HW interface. If I see the `target/stm32g4x.cfg` shipped with STM32CubeIDE, which are in a different dir. btw.

haneefdm, 2022-07-08:
This is what I have in my install for the script files
```
hdm@HaneefMac-2021 STM32CubeIDE.app % find . -name stm32g4x.cfg
./Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.100.202111151130/resources/openocd/st_scripts/target/stm32g4x.cfg
./Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.200.202202161333/resources/openocd/st_scripts/target/stm32g4x.cfg
```

There are so many versions of the truth, I cannot keep it straight.

ChrisIdema, 2022-07-08:
It can now find the cfg files, but now I get this error:
> ...
> in procedure 'swj_newdap' called at file "embedded:startup.tcl", line 238
> in procedure 'hla newtap' called at file "/opt/st/stm32cubeide_1.8.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.100.202111151130/resources/openocd/st_scripts/target/swj-dp.tcl", line 28
> in procedure 'swj_newdap' called at file "embedded:startup.tcl", line 238
> in procedure 'hla newtap' called at file "/opt/st/stm32cubeide_1.8.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.100.202111151130/resources/openocd/st_scripts/target/swj-dp.tcl", line 28
> in procedure 'swj_newdap' called at file "embedded:startup.tcl", line 238
> in procedure 'using_hla' called at file "/opt/st/stm32cubeide_1.8.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.100.202111151130/resources/openocd/st_scripts/target/swj-dp.tcl", line 27
> at file "embedded:startup.tcl", line 944
> [2022-07-08T17:01:05.160Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
> GDB server session ended. This terminal will be reused, waiting for next session to start...

I feel like I'm constantly chasing bugs as if my system configuration is very odd. Is ubuntu 20.04 not common?

ChrisIdema, 2022-07-08:
Yes I have STM32CubeIDE 1.8.0 installed. I normally use this for development, but we are now converting our projects to Cmake and will be using VsCode as an IDE. I'm used to STM32CubeIDE.

haneefdm, 2022-07-08:
If you have/create a STM32CubeIDE project with your device then you can open the Debug Configuration for OpenOCD and you will see the following
[image]
If you run the debug session even once, it will create the `.cfg` file. We will need that. It will have everything we need and you can use this file as the only config file in launch.json. You may want to make a copy of this file in case Eclipse overwrites it in the future.
Click the `Show Command Line` button and it will show you exact command-line ST uses for openocd. From here, we collect the following
* Path to the openocd executable
* All the arguments to openocd with the `-s` option. These are what goes into the `searchDir` in launch.json
[image]

haneefdm, 2022-07-08:
This is the official way ST does and that is what I do regardless of what the Internet tells me. But that is me. Everyone does it differently.

ChrisIdema, 2022-07-08:
I will try it. But I'm a bit skeptical as STM32CubeIDE crashes when debugging with openocd (at least with a different board we have).

haneefdm, 2022-07-08:
We just need it to start and create that cfg file.

ChrisIdema, 2022-07-08:
It works! The only thing I changed was I grabbed the generated cfg file "STM32G491RE_blinky Debug.cfg" and substituted that for my cfg files:
```
"configFiles": [
                "STM32G491RE_blinky Debug.cfg"
                // "interface/stlink.cfg",
                // "target/stm32g4x.cfg"
            ],
```

The file is quite small, so I can probably generate it myself for different targets (I use a python script to parse the ioc file and makefile generated by STM32CubeMX)
Here is the contents of the file:

```
# This is an NUCLEO-G491RE board with a single STM32G491RETx chip
#
# Generated by STM32CubeIDE
# Take care that such file, as generated, may be overridden without any early notice. Please have a look to debug launch configuration setup(s)

source [find interface/stlink-dap.cfg]

set WORKAREASIZE 0x8000

transport select "dapdirect_swd"

set CHIPNAME STM32G491RETx
set BOARDNAME NUCLEO-G491RE

# Enable debug when in low power modes
set ENABLE_LOW_POWER 1

# Stop Watchdog counters when halt
set STOP_WATCHDOG 1

# STlink Debug clock frequency
set CLOCK_FREQ 8000

# Reset configuration
# use hardware reset, connect under reset
# connect_assert_srst needed if low power mode application running (WFI...)
reset_config srst_only srst_nogate connect_assert_srst
set CONNECT_UNDER_RESET 1
set CORE_RESET 0

# ACCESS PORT NUMBER
set AP_NUM 0
# GDB PORT
set GDB_PORT 3333

# BCTM CPU variables

source [find target/stm32g4x.cfg]

# SWV trace
set USE_SWO 0
set swv_cmd "-protocol uart -output :3344 -traceclk 170000000"
source [find board/swv.tcl]
```

closed 2022-07-08 by ChrisIdema

haneefdm, 2022-07-08:
I just create sample projects once per device and use the generated cfg file. I sometimes comment stuff out that may conflict with the options I set, but so far it just works for me.

The `GDB_PORT` concerns me (but seems to be ignored). We (and ST as well) see that from the command-line so I don't understand why they set it here and in their other config files.