Marus/cortex-debug #695: How to set swd/jtag speed?

hongshui3000, 2022-07-12:
How to set swd/jtag speed? cppdbg can use "debugServerArgs" -speed to specify the swd speed for example:
```
"debugServerArgs": "-if swd -speed 4000 -endian little -port 3333 -device Cortex-M3 -x ${workspaceFolder}/.vscode/reset.gdb",
```
I use cortex debug, what should I do?

haneefdm, 2022-07-12:
Depends on the gdb-server. See `serverArgs`

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

closed 2022-07-12 by haneefdm

hongshui3000, 2022-07-12:
I added the configuration but it doesn't seem to work?

```
"serverArgs": "-speed 10000000",
```

```
Launching gdb-server: "C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe" -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device Cortex-M3 -select usb=59522101 -rtos GDBServer/RTOSPlugin_Zephyr.dll "-speed 10000000"
```
[image]

hongshui3000, 2022-07-13:
I changed it to the following configuration
But the effect is still the same
```
            "serverArgs": [
                "-speed 1000000",
              ],
Launching gdb-server: "C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe" -singlerun -nogui -if swd -port 50000 
-swoport 50001 -telnetport 50002 -device Cortex-M3 -select usb=59522101 -rtos GDBServer/RTOSPlugin_Zephyr.dll 
"-speed 10000000"
```

If I use -gui command without quotes
```
            "serverArgs": [
                "-gui",
              ],
Launching gdb-server: "C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe" -singlerun -nogui -if swd -port 50000 
-swoport 50001 -telnetport 50002 -device Cortex-M3 -select usb=59522101 、
-rtos GDBServer/RTOSPlugin_Zephyr.dll -gui
```

haneefdm, 2022-07-13:
No, that is not one argument, it is two arguments from an argc/argv point of view. Like two arguments to a command
```JSON
            "serverArgs": [
                "-speed",
                "1000000"
            ],
```

hongshui3000, 2022-07-13:
I get it, thank