Marus/cortex-debug #696: Run the main function directly by set the pc register to the main function addr

XinyuKhan, 2022-07-12:
I don't know if this feature is supported. Sometimes I need the debugger to bypass the bootloader and run the main function directly but not just wait the breakpoint, because the downloading operation usually broken the check mechanism in the bootloader or the bootloader may just not exist.

haneefdm, 2022-07-12:
Yes, people do that. But you better know your processor, the gdb-server (JLink, openocd, etc.), etc.

There are all kinds of `override...`s available in launch.json.

See https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

closed 2022-07-12 by haneefdm

XinyuKhan, 2022-07-12:
Thx for the rely. I'll do more research

XinyuKhan, 2022-07-12:
> Yes, people do that. But you better know your processor, the gdb-server (JLink, openocd, etc.), etc.
> 
> There are all kinds of `override...`s available in launch.json.
> 
> See https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

I can set PC reg by override the pre launch command, but as know I have to set the main address to the command manually, Is there any way that can automatically get the address of the main entry and set it to the pc reg? If there is not, It's possible to and such feature in the future? The stm32cubeide support this feature accroding its log.
[image]

haneefdm, 2022-07-12:
> is there any way that can automatically get the address of the main entry

Probably. You also have to be a gdb expert. Try one of the following
```
set $pc = main
set $pc = &main
```
in your overrideLaunchCommands. Don't do it in 'preLaunc..."

Note that for most ARM processors, you also have to set $sp and $lr, or your stack will be bogus. $lr should be -1

It will help if you read our Wiki https://github.com/Marus/cortex-debug/wiki/Cortex-Debug:-Under-the-hood to see in detail our startup sequence.

selectstriker2, 2022-07-12:
Rather than jumping straight to `main`, you might be better off jumping to your reset vector, which should be the first address in the image, something like `__reset_handler`. That will set up your stack pointer(s) and link register, initialize .bss region to 0, copy any read only data to RAM, etc. before entering `main()`

XinyuKhan, 2022-07-12:
> Probably. You also have to be a gdb expert. Try one of the following
>
> ```
> set $pc = main
> set $pc = &main
> ```
>
> in your overrideLaunchCommands. Don't do it in 'preLaunc..."
>
> Note that for most ARM [...]

Great! I will try your sugestion latter. Thank you so mush for your reply!