Marus/cortex-debug #697: Unable to get disassembly on Cortex-M0 using Atmel-ICE JLinkGDBServerCLExe

jameshume, 2022-07-12:
Hi,

I am setting up VSCode to debug a Cortex-M0 based device using the Atmel SAM-ICE.

Setup to use JLINK GDB server:
```
SEGGER J-Link GDB Server V7.66e Command Line Version
JLinkARM.dll V7.66e (DLL compiled Jun 28 2022 15:28:35)
```

Running on this version of VSCode:
```
Version: 1.69.0
Commit: 92d25e35d9bf1a6b16f7d0758f25d48ace11e5b9
Date: 2022-07-07T08:29:47.439Z
Electron: 18.3.5
Chromium: 100.0.4896.160
Node.js: 16.13.2
V8: 10.0.139.17-electron.0
OS: Linux x64 5.4.0-121-generic
```

With this version of cortex-debug:
```
Name: Cortex-Debug
Id: marus25.cortex-debug
Description: ARM Cortex-M GDB Debugger support for VSCode
Version: 1.4.4
Publisher: marus25
VS Marketplace Link: https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug
```

The error I am seeing is this:
```
Error: GDB failed: -data-disassemble: Mode argument must be 0, 1, 2, or 3. (from data-disassemble -s <snip> <snip> -- 5)
```

I looked this up and I can see that mode 5 requests mixed source and disassembly with raw opcodes and mode 3 seems to be the deprecated version that also reports to do the same thing [[ref]](https://docs.amd.com/bundle/ROCDebugger-User-Guide-v5.0/page/GDB_002fMI-Data-Manipulation.html).

I can see, in the source, that the following is hard coded:

```
// disasm.ts:521
cmd = `data-disassemble -s ${hexFormat(startAddress)} -e ${hexFormat(end)} -- 5`;
```

I'm a bit stuck on how to solve. I think my JLinkGDBServerCLExe is pretty recent, so perhaps it doesn't support this new mode and only supports the deprecated mode 3?

If this is the case, is there the possibility to provide an override?

PhilippHaefele, 2022-07-12:
The error comes from GDB (Client) and not the Segger GDB Server.

So most likely that one is outdated. As it seems that you're on Linux you maybe need to install it manually (e.g. from https://developer.arm.com/downloads/-/gnu-rm) as your package manager only installs a very old one.

haneefdm, 2022-07-12:
I think you are using version 7 of GDB. Many bugs/issues in that version. We are obsoleting anything older than version 9 this month. You should have seen a message in the Debug Console...

jameshume, 2022-07-25:
Ah, I see, thanks guys I'll have a go at upgrading the client.