Marus/cortex-debug #699: Cannot read cortex peripherals on RX130

8itbangr, 2022-07-12:
It seems to try, but gets zeroes every time:
```
85-data-read-memory-bytes "0x88684" 1
-> 85^done,memory=[{begin="0x00088684",offset="0x00000000",end="0x00088685",contents="00"}]
86-data-read-memory-bytes "0x88684" 1
-> 86^done,memory=[{begin="0x00088684",offset="0x00000000",end="0x00088685",contents="00"}]
```

8itbangr, 2022-07-12:
```
[2022-07-12T14:52:59.441Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/Users/richardbrown/git/cooking.copperhead/build/micro-rx130/JLinkGDBServer -singlerun -nogui -if FINE -port 50000 -swoport 50001 -telnetport 50002 -device R5F51303 -jlinkscriptfile /Users/richardbrown/git/cooking.copperhead/build/micro-rx130/debug.jlink -speed 1000
SEGGER J-Link GDB Server V6.80a Command Line Version

JLinkARM.dll V6.80a (DLL compiled May 29 2020 16:28:29)

Command line: -singlerun -nogui -if FINE -port 50000 -swoport 50001 -telnetport 50002 -device R5F51303 -jlinkscriptfile /Users/richardbrown/git/cooking.copperhead/build/micro-rx130/debug.jlink -speed 1000
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 /Users/richardbrown/git/cooking.copperhead/build/micro-rx130/debug.jlink
J-Link settings file:          none
------Target related settings------
Target device:                 R5F51303
Target interface:              FINE
Target interface speed:        1000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V11 compiled Apr 27 2021 16:36:21
Hardware: V11.00
S/N: 51001741
Feature(s): GDB
Checking target voltage...
Target voltage: 5.02 V
Listening on TCP/IP port 50000
Connecting to target...
Comparing flash   [....................] Done.
Erasing flash     [....................] Done.
Programming flash [....................] Done.
Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Reading 64 bytes @ address 0xFFFFFFC0
Reading 64 bytes @ address 0x00000000
Reading 64 bytes @ address 0xFFFFFF80
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0xFFFFFFFF)
Received monitor command: reset
Resetting target
Downloading 4096 bytes @ address 0xFFFF0000
Downloading 4096 bytes @ address 0xFFFF1000
Downloading 4096 bytes @ address 0xFFFF2000
Downloading 536 bytes @ address 0xFFFF3000
Downloading 2716 bytes @ address 0xFFFF3218
Downloading 1064 bytes @ address 0xFFFF3CB4
Downloading 128 bytes @ address 0xFFFFFF80
Comparing flash   [....................] Done.
Erasing flash     [....................] Done.
Programming flash [....................] Done.
Writing register (PC = 0xffff0040)
Reading 64 bytes @ address 0xFFFF0040
Reading 64 bytes @ address 0xFFFF0080
Received monitor command: reset
Resetting target
Reading 64 bytes @ address 0xFFFF0B40
[... 121 lines cut ...]
Read 4 bytes @ address 0x00000508 (Data = 0x0000050C)
Read 1 bytes @ address 0x0000050C (Data = 0x00)
Read 4 bytes @ address 0x00000508 (Data = 0x0000050C)
Read 1 bytes @ address 0x0000050D (Data = 0x00)
Read 4 bytes @ address 0x00000508 (Data = 0x0000050C)
Read 1 bytes @ address 0x0000050D (Data = 0x00)
Read 4 bytes @ address 0x00000508 (Data = 0x0000050C)
Read 1 bytes @ address 0x0000050E (Data = 0x00)
Read 4 bytes @ address 0x00000508 (Data = 0x0000050C)
Read 1 bytes @ address 0x0000050E (Data = 0x00)
Read 4 bytes @ address 0x00000508 (Data = 0x0000050C)
Read 1 bytes @ address 0x0000050F (Data = 0x00)
Read 4 bytes @ address 0x00000508 (Data = 0x0000050C)
Read 1 bytes @ address 0x0000050F (Data = 0x00)
Read 1 bytes @ address 0x00088684 (Data = 0x00)
Read 1 bytes @ address 0x00088684 (Data = 0x00)
Reading 64 bytes @ address 0xFFFF2340
Read 1 bytes @ address 0xFFFF23AF (Data = 0xFB)
Reading 64 bytes @ address 0xFFFF2680
Reading 64 bytes @ address 0xFFFF2600
Read 1 bytes @ address 0xFFFF268A (Data = 0x5D)
Reading 64 bytes @ address 0xFFFF2840
Reading 64 bytes @ address 0xFFFF27C0
Read 1 bytes @ address 0xFFFF2843 (Data = 0xED)
```

8itbangr, 2022-07-12:
Here's the debug console, as well:
```
Cortex-Debug: VSCode debugger extension version 1.5.1 git(89a84bd). Usaage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from /Users/richardbrown/git/cooking.copperhead/tools/kpit-rx/tc/bin/arm-none-eabi-objdump --syms -C -h -w /Users/richardbrown/git/cooking.copperhead/build/micro-rx130/micro-rx130.elf
Reading symbols from /Users/richardbrown/git/cooking.copperhead/tools/kpit-rx/tc/bin/arm-none-eabi-nm --defined-only -S -l -C -p /Users/richardbrown/git/cooking.copperhead/build/micro-rx130/micro-rx130.elf
Launching GDB: /Users/richardbrown/git/cooking.copperhead/tools/kpit-rx/tc/bin/arm-none-eabi-gdb -q --interpreter=mi2 /Users/richardbrown/git/cooking.copperhead/build/micro-rx130/micro-rx130.elf
1-gdb-version
Error: /Users/richardbrown/git/cooking.copperhead/tools/kpit-rx/tc/bin/arm-none-eabi-nm failed! statics/global/functions may not be properly classified: Error: spawn /Users/richardbrown/git/cooking.copperhead/tools/kpit-rx/tc/bin/arm-none-eabi-nm ENOENT
    Expecting `nm` next to `objdump`. If that is not the problem please report this.
Error: objdump failed! statics/globals/functions may not be properly classified: Error: spawn /Users/richardbrown/git/cooking.copperhead/tools/kpit-rx/tc/bin/arm-none-eabi-objdump ENOENT    ENOENT means program not found. If that is not the issue, please report this problem.Launching gdb-server: /Users/richardbrown/git/cooking.copperhead/build/micro-rx130/JLinkGDBServer -singlerun -nogui -if FINE -port 50000 -swoport 50001 -telnetport 50002 -device R5F51303 -jlinkscriptfile /Users/richardbrown/git/cooking.copperhead/build/micro-rx130/debug.jlink -speed 1000
    Please check TERMINAL tab (gdb-server) for output from /Users/richardbrown/git/cooking.copperhead/build/micro-rx130/JLinkGDBServer
Finished reading symbols from objdump: Time: 32 ms
Finished reading symbols from nm: Time: 29 ms
Python Exception <type 'exceptions.ImportError'> No module named gdb:

warning:
Could not load the Python gdb module from `/Users/richardbrown/git/cooking.copperhead/tools/kpit-rx/tc/bin/../share/gdb/python'.
Limited Python support is available from the _gdb module.
Suggest passing --data-directory=/path/to/gdb/data-directory.

-> =thread-group-added,id="i1"
-> ~"Reading symbols from /Users/richardbrown/git/cooking.copperhead/build/micro-rx130/micro-rx130.elf..."
-> ~"done.\n"
-> ~"GNU gdb (GDB) 7.8.2-20210419-Build_1\n"
-> ~"Copyright (C) 2014 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law.  Type \"show copying\"\nand \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-apple-darwin20.5.0 --target=rx-elf\".\nType \"show configuration\" for configuration details."
-> ~"\nFor bug reporting instructions, please see:\n"
-> ~"<http://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n<http://www.gnu.org/software/gdb/documentation/>.\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
WARNING: Cortex-Debug will deprecate use of GDB version 8 after July 2022. Please upgrade to version 9+
Reading symbols from /Users/richardbrown/git/cooking.copperhead/build/micro-rx130/micro-rx130.elf...done.
2-gdb-set target-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-interpreter-exec console "source /Users/richardbrown/.vscode/extensions/marus25.cortex-debug-1.5.1/support/gdbsupport.init"
-> 5^done
6-interpreter-exec console "source /Users/richardbrown/.vscode/extensions/marus25.cortex-debug-1.5.1/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 6^done
7-target-select extended-remote localhost:50000
-> =thread-group-started,id="i1",pid="42000"
-> =thread-created,id="1",group-id="i1"
-> ~"0xffffffff in fixedVectorTable ()\n"
0xffffffff in fixedVectorTable ()
-> *stopped,frame={addr="0xffffffff",func="fixedVectorTable",args=[]},thread-id="1",stopped-threads="all"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 7^connected
8-interpreter-exec console "monitor halt"
-> 8^done
9-interpreter-exec console "monitor reset"
-> @"Resetting target\r\n"
Resetting target
-> 9^done
10-target-download
-> 10+download,{section=".text",section-size="12824",total-size="441736"}
-> 10+download,{section=".rodata",section-size="2716",total-size="441736"}
-> 10+download,{section=".data",section-size="1064",total-size="441736"}
-> 10+download,{section=".fixedVectorTable",section-size="128",total-size="441736"}
-> 10^done,address="0xffff0040",load-size="16732",transfer-rate="26771200",write-rate="2390"
11-interpreter-exec console "monitor reset"
[... 170 lines cut ...]
-> 88^done,bkpt={number="5",type="breakpoint",disp="keep",enabled="y",addr="0xffff23af",func="ConfigurePwms",file="src/micro-rx130/Bsp/TinyDataSource_Pwm.c",fullname="/Users/richardbrown/git/cooking.copperhead/src/micro-rx130/Bsp/TinyDataSource_Pwm.c",line="139",thread-groups=["i1"],times="0",original-location="/Users/richardbrown/git/cooking.copperhead/src/micro-rx130/Bsp/TinyDataSo [... cut ...]
-> 96^done,bkpt={number="10",type="breakpoint",disp="keep",enabled="y",addr="0xffff2658",func="WritePeriod",file="src/micro-rx130/Bsp/TinyDataSource_Pwm.c",fullname="/Users/richardbrown/git/cooking.copperhead/src/micro-rx130/Bsp/TinyDataSource_Pwm.c",line="249",thread-groups=["i1"],times="0",original-location="/Users/richardbrown/git/cooking.copperhead/src/micro-rx130/Bsp/TinyDataSource_Pwm.c:249"}
97-break-delete
-> 97^done
98-target-disconnect
-> =thread-group-exited,id="i1"
-> 98^done
-gdb-exit
-> ^exit
```

haneefdm, 2022-07-12:
Don't know what to tell you. That is what JLink server returned
```
Read 1 bytes @ address 0x00088684 (Data = 0x00)
Read 1 bytes @ address 0x00088684 (Data = 0x00)
```

Can you use the memory viewer to see what is actually there in and around that address? Maybe it is unable to read 1 byte at a time? I don't know this device. It is strange that you have 1-byte registers (not illegal but we should be able read them then)

8itbangr, 2022-07-12:
If I put the address into the Watch window, it reads 0 there, as well.

haneefdm, 2022-07-12:
Then you have to contact your silicon vendor to ask why. Or maybe it is actually zero. Sometimes status registers (or write only registers) always read as zero so as to not cause a side effect.

I don't see a defect on our side, so I am closing it. Contact if you have reason to believe otherwise

closed 2022-07-12 by haneefdm