Marus/cortex-debug #706: Unable to launch debug under certain specific circumstances

AndriiLishchynskyi, 2022-07-20:
Unable to launch debug session under certain specific circumstances. The issue was initially observed with Cortex-Debug 1.4.4, but it also exist with the latest pre-release version - 1.5.1.

Assume you have a chained configuration: 'Multi-Core Debug --> Attach'. Both configurations located in different directories. In order to be able to launch 'Attach' configuration, user need to specify a path to the second 'Attach' config by using `folder` option.

Here we found a curious thing. In case `folder` points to the wrong location lacking 'Attach' configuration, launching debug session will fail to start `Attach` config and will show an error. Ok, that is expected. Later on, if you fix a path to the `Attach` config, debug will start properly. But here is a tricky point: you will not be able to launch `Attach` config standalone unless you restart VSCode.

When `Attach` launched standalone, I have noticed something strange in the log. `folder` option, for some reason, contain wrong path that is not actually present in the parent config, but the path that was specified at the time debugging failed.
"configFromParent": {
        "name": "Attach PSoC6 CM4 (KitProg3_MiniProg4)",
        "folder": "C:/Users\\lishchynskyi\\mtw-psoc6-15-07-2022-2\\Dual-CPU_Empty_PSoC6_App\\proj_cm3"
        "enabled": true,
        "delayMs": 0,
        "detached": false,
        "waitOnEvent": "postInit",
        "lifecycleManagedByParent": true
}

I have attached a folder structure that I have. My steps to reproduce looks this way:
1) launch `Multi-Core Debug (KitProg3_MiniProg4)` chained config to check it works properly
2) edit second config in a chain `Attach PSoC6 CM4 (KitProg3_MiniProg4)` - change `folder` option to point to some wrong location, e.g. "folder": "./proj_cm4" -> "folder": "./proj_cm3"
3) launch `Multi-Core Debug (KitProg3_MiniProg4)` chained config and observe an error about missing `Attach` config
4) fix path to the `Attach PSoC6 CM4 (KitProg3_MiniProg4)` config and relaunch `Multi-Core Debug (KitProg3_MiniProg4)`. Make sure both configs start.
5) Launch `Attach PSoC6 CM4 (KitProg3_MiniProg4)` standalone and get an error

VSCode-1.69.1

haneefdm, 2022-07-23:
I cannot duplicate this. This sounds like openocd did not start or it errored out.

haneefdm, 2022-07-24:
I could not duplicate because the workspace is missing in the zip file. Can you provide that as well?

The rule is that the VSCode API requires that the folder be any path name that iis a path to a workspace folder or it is workspace name. In the end, the API only expects the wsFolder. Here is the API
```TypeScript
export function startDebugging(folder: WorkspaceFolder | undefined, nameOrConfiguration: string | DebugConfiguration, parentSessionOrOptions?: DebugSession | DebugSessionOptions): Thenable<boolean>;
```
A `WorkspaceFolder` cannot be created by this extension. It should already exist and I convert a folder name you give me to a VSCode `WorkspaceFolder` by searching through all the workspace folders for a match. I will update the docs accordingly. Now that we know the rules better, perhaps it is easier to use the workspace folder name (not a path) but it is upto you. I provided the additional feature to convert a pathname to a `WorkspaceFolder` which perhapsj is not exhaustive.

Having said all that this still has something to do with openocd perhaps quitting too early or not creating all the TCP ports for multiple gdbs to connect to. We need to focus a bit on openocd here.

haneefdm, 2023-03-07:
Not sure this is still an issue. I will close it for now and we can reopen if this is still an issue.

closed 2023-03-07 by haneefdm