Marus/cortex-debug #709: Live variable debugging

Frisc98, 2022-07-25:
Has anyone had success with live variable debugging? Is it even supported or enabled somehow?

haneefdm, 2022-07-25:
In the roadmap but not yet implemented. We actually don't know how to. I have some ideas but they don't look good.

closed 2022-07-25 by haneefdm

julian-poidevin, 2022-07-27:
@haneefdm, maybe it’s off topic but I found this method : https://sysprogs.com/w/live-non-intrusive-debugging-of-real-time-arm-targets/

haneefdm, 2022-07-27:
Not off-topic. When expanding data structures, we have to do all the work gdb normal does. Very complex and error-prone. structs/unions/bitfields. If it is just monitoring some global scaler variables, I already know how to do that at least with OpenOCD -- which even allows two gdb connections which may be another/better way.

Getting DWARF info from objdump was an option but it ended up in output that was 100MB or more, so we stopped using that. One extreme case clocked in 2GB of output, and other times a crash (ran out of virtual memory).

https://openocd.org/doc/html/GDB-and-OpenOCD.html#gdbmeminspect

Not sure about JLink which is another popular adapter. Some (encouraging) info here

https://forum.segger.com/index.php/Thread/3905-SOLVED-GDB-server-read-data-structures-while-target-is-running/