Marus/cortex-debug #710: [disassembly view,  line 521 of  src/backend/disasm.ts] hexFormat will round address to 32bit width which will cause error for arm64 when it try to access memory larger than 4GB

mikewang01, 2022-07-27:
cmd = `data-disassemble -s ${hexFormat(startAddress)} -e ${hexFormat(end)} -- 5`; 

line 521 of file src/backend/disasm.ts
hexFormat will round address to 32bit width which will cause error for arm64 when it try to access memory larger than 4GB
I have done following modification for correcting this
`cmd = `data-disassemble -s 0x${startAddress.toString(16)} -e 0x${end.toString(16)} -- 5`;`

haneefdm, 2022-07-27:
This extension is for 32-bit `Cortex-M` class devices only. It says so as our Github repo description. As such, hexFormat is not our only problem - which btw, is used all over the place.

Typescript/javascript itself cannot handle 64-bit numbers. It runs out at 53 bits. There are many places I know where code will fail. We have to scrub the entire source line by line to move to a new numeric type `bigint` which was introduced in 2020. So, very recent and not all tools had support as of last year.

I also do not have any 64-bit boards with me to test anything. Maybe someday. For now, I will recommend using MS cpptools for your debugging. needs. I would not trust our extension.

Eventually, we may change our purpose and move away from Cortex-M devices only. Already, people are using it for Xtensa, RISC-V, etc. But all 32-bit. 64-bit is on the horizon but not right now.

closed 2022-07-27 by haneefdm

mikewang01, 2022-07-27:
> Thanks for  Ur feedback. Yes, As u explained, I understand that this tool was designed to target cortex m SOC. And I used this to debug some MCU like stm32 for some while. Recently, I am trying to bring up Cortex A55, so the first idea was to use this tool again. Anyway, most features of this tool functioned as expected with AARch64. 

Agree,  This was a good trending which means more and more people pay attention and make contributions to this tool I believe .Really hope to see a powerful tool which was able to cover all architecture someday

Again, really appreciate Ur cortex debug tool, this really boost our development efficient a lot.

haneefdm, 2022-07-27:
> Anyway, most features of this tool functioned as expected with AARch64

In general, it will work out. But I know of land mines which may silently fail you. A crash is preferable to wrong results IMO. Even if I had the `hexFormat` for larger than 32 bits, it would have failed for anything larger than 2^53 silently. Until we port all of our code to be `bigint` compliant, including all the other opensource stuff we use.

Thanks for liking our extension. One of these days (if I can get a solid 3-4 days away from work), I will attack the 64-bit issue when I know that all the other 3rd part tools and Javascript modules we rely on are also on board. I think they are by now, but not sure.