Marus/cortex-debug #711: How to debug "GDB Server Quit Unexpectedly"? Windows 11 problem?

tdjastrzebski, 2022-07-28:
Recently I had to reinstall my PC and updated to Windows 11. Since then, I am no longer able to debug my boards, launch.json configs which used to work no longer work. The error message I get is: "GDB Server Quit Unexpectedly." (pyOCD and ST-Util) or nothing happens (OpenOCD). All three GDB server seem to start fine from the CMD or from VS Code terminal window.
I have exactly the same problem on two Windows 11 laptops reinstalled recently. All versions are the latest, no matter if I use cortex-debug released or pre-preleased version.

What are the steps to try first before running cortex-debug under debugger? I have tried adding `"debugServer": 4711` config line as instructed [here](https://github.com/Marus/cortex-debug#how-to-debug) but it did not help. Error I get: `connect ECONNREFUSED 127.0.0.0:4711`  
Please advise.  

[image]
[image]
[image]
[image]
[image]
[image]

haneefdm, 2022-07-28:
> What are the steps to try first before running cortex-debug under debugger? I have tried adding "debugServer": 4711 config line as instructed [here](https://github.com/Marus/cortex-debug#how-to-debug) but it did not help. Error I get: connect ECONNREFUSED 127.0.0.0:4711

Our launch.json has multiple configurations. As the instructions says, you need select "Extension + Debug Server" for debugging both the frontend and the backend. You probably did not do that.

Since you are making sure that stuff on the command-line works, you should use the **exact** command-line as we use to start with and perhaps deviate from there in case that is the problem. We print the command we use with its args. We print every transaction between gdb and us. Why you server is quitting is beyond me. Error message says, the server closed the connection (unexpectedly). Maybe your USB is not working at all?

Maybe there is things going on in your launch.json that is causing issues.

tdjastrzebski, 2022-07-28:
Thanks @haneefdm, it is probably a dumb question but where do I select "Extension + Debug Server" option?
I does not seem like it is a USB problem. It is the same, well tested hardware, I can flash with no problems.
I used exactly the same GDB start commands (copy-paste) - no problems.

haneefdm, 2022-07-28:
In the Run and Debug toolbar. Mine is docked and looks like this.

[image]

We have four configurations in our launch.json and this one debug both frontend and backend at the same time

tdjastrzebski, 2022-07-29:
I see, this 'launch.json' entry I am missing.

haneefdm, 2022-07-29:
I have no idea what you are trying to do. The `debugServer` is used to debug the debugger which is our extension.

https://github.com/Marus/cortex-debug/blob/e98ea060579da3672cae817c57d98f42bd6f54b6/.vscode/launch.json#L47

tdjastrzebski, 2022-07-30:
Advanced code structure and my limited typescript skills do not help debugging. After two hours I can only say that problem happens the time GDBServer.connectToConsole() gets executed. I am unable to catch any exception. Likely it is a race condition, probably some promise is not awaited for.

haneefdm, 2022-07-30:
We must be getting an error. Maybe you already did this Put a breakpoint at this line and examine `e`

https://github.com/Marus/cortex-debug/blob/e98ea060579da3672cae817c57d98f42bd6f54b6/src/backend/server.ts#L202

Above means there was an error making a connection or someone closed the socket while in use. But, if you get to the following line then it was an orderly shutdown (if this.consoleSocket !== null).

https://github.com/Marus/cortex-debug/blob/e98ea060579da3672cae817c57d98f42bd6f54b6/src/backend/server.ts#L199

The server is in the backend. It is trying to send the messages coming from the gdb-server on stdout/stderr to the frontend for display on a socket. This socket was created here...

https://github.com/Marus/cortex-debug/blob/e98ea060579da3672cae817c57d98f42bd6f54b6/src/frontend/server_console.ts#L213

By default it uses the port `55878`, but it will find a free one if it is busy. When things go wrong, there are a bunch of messages that go to the Debug Console (there are multiple so make sure you select the right one. For the frontend, it should be "Extension")

On Windows, you can see if someone is using a port like this.

https://stackoverflow.com/questions/48198/how-do-i-find-out-which-process-is-listening-on-a-tcp-or-udp-port-on-windows

tdjastrzebski, 2022-07-30:
Socket error is handled by handler set up in GDBServerConsoleInstance.newBackendConnection() but probably it is not the root cause.

haneefdm, 2022-07-30:
No not the root cause. ECONNRESET means that the other end closed the connection.

I am wondering about your Windows security policy. Could it be that the Windows firewall is causing socket creations/connections to fail on certain programs? You said you recently updated to Win 11.

I am no Windows expert/IT person.

tdjastrzebski, 2022-07-30:
I am not an expert either but I think firewall should affect tcpip socket connections within the same host.
Probably the next step would be to start gdb server as node.js process and try to connect over tcpip socket but maybe the current class structure allows for such basic test to be easily written and placed where it belongs?

haneefdm, 2022-07-30:
Try this first. I want to know if things are okay with the server. Comment out this line

https://github.com/Marus/cortex-debug/blob/e98ea060579da3672cae817c57d98f42bd6f54b6/src/backend/server.ts#L62

Let us not use the console at all. After this line
https://github.com/Marus/cortex-debug/blob/e98ea060579da3672cae817c57d98f42bd6f54b6/src/backend/server.ts#L227

Add this
```typescript
        this.emit('output', data.toString());
        return;
```
Nothing will be sent to the console but your simulation should continue.

haneefdm, 2022-07-30:
The above proves that it is not the frontend-backend that is failing. We got an `ECONNRESET` on the frontend (server) side. This means the client (backend) closed it.  Means someone/something closed the socket on the client side - perhaps forcefully.

Put breakpoints in
```typescript
GDBServer.exit
GDBServer.onExit
GDBServer.onError
GDBServer.disconnectConsole
```
and on these lines
https://github.com/Marus/cortex-debug/blob/e98ea060579da3672cae817c57d98f42bd6f54b6/src/backend/server.ts#L254
https://github.com/Marus/cortex-debug/blob/e98ea060579da3672cae817c57d98f42bd6f54b6/src/backend/server.ts#L203

> but I think firewall should not affect tcpip socket connections

I think it can when creating the server. It can prevent servers from even starting. Has happened to me before even on Windows 10. Not sure what you mean by `gdb process`? You mean openocd?

I would first create two Nodejs programs for the server (front-end) and client (debug-adapter). Note that the gdb.

Starting the debug-adapter as a separate Nodejs process is super complicated as the entry point is actually in a vscode shim

tdjastrzebski, 2022-07-30:
@haneefdm, I found the problem. The [most official build](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads) of the **ARM GNU Toolchain** (latest version 11.2) requires retired and unsupported Python2.7. The latest [xPack build](https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/) of the same version does not have this issue.
Until I started to debug this extension, I did not fully realize that GDB and gdb-server are two different things and both commands had to be checked.  
Thanks for your kind support.  

[image]

haneefdm, 2022-07-31:
This is why, in the Debug Console we print the *exact* command line of external tools we use so you can run those yourself.

We try to document stuff but the Wiki is rarely visited.

https://github.com/Marus/cortex-debug/wiki/Cortex-Debug:-Under-the-hood

closed 2022-07-31 by haneefdm