Marus/cortex-debug #712: Openocd argument order

plzCZ, 2022-08-02:
I am trying to debug Raspberry Pico using jlink. The issue is that the openocd defaults to jtag. I can use `transport select` argument but only before or after `-f interface/jlink.cfg -f target/rp2040.cfg`(using `openOCDLaunchCommands` and `openOCDPreConfigLaunchCommands`). It works only when the `transport select ` is between interface and target. Like this `-f interface/jlink.cfg -c "transport select swd" -f target/rp2040.cfg`. All other combinations result in error. Is there a way to force this in launch.json?

haneefdm, 2022-08-02:
Why not use your own cfg file that forces any order you want and then use that as the only config file in launch.json. Config files are nothing special, they are just Tcl files. Whatever you do with a -c option, you can do that (on separate lines) in your config/tcl file.
```Tcl
# The following line is the equivalent of -f filename
source [find interface/jlink.cfg]
# The following line is the equivalent of -c command
transport select swd
source [find target/rp2040.cfg]
```

You can also undo some of the options set on the command line by adding some TCL code.

closed 2022-08-02 by haneefdm

plzCZ, 2022-08-02:
Exactly what i needed. Thank you very much.

surm92, 2024-10-12:
Another way is to use "openOCDPreConfigLaunchCommands" to select driver and interface before the target configuration.

```typescript
"openOCDPreConfigLaunchCommands":[
   "adapter driver jlink",
   "transport select swd"
],
"configFiles": [
   "${config:openOcdScriptsPath}/target/stm32f3x.cfg"
],
```