Marus/cortex-debug #713: No valid SVD filepath

hongshui3000, 2022-08-03:
[image]
No valid SVD filepath.
I'm using CORTEX debug and Microsoft's Embedded tool, but the embedded tool svd display has some issues, I don't know if it's a issue with cortex debug or its own problem.
launch config
```
            "svdFile":"./STM32L0x0.svd",
            "svdPath":"./STM32L0x0.svd",
```
cortex debug svd works fine

haneefdm, 2022-08-03:
I don't know, but you should file an issue with them. It worked for me and I didn't do anything special. Not in our extension, or theirs, or my launch.json other than just using their method of specifying it.

Contrary to what people think, the SVD is a beast of a thing to parse and make sense of, mainly because it changes all the time. Maybe it is stable now. We are several years behind and the format moved on with tons of new stuff. I don't have an army of people to keep track. So, I was glad MS took up the mantle. Isn't that nice? So, please help them.

closed 2022-08-03 by haneefdm

hongshui3000, 2022-08-04:
[image]
[image]
Can these two functions currently support hiding? Because it is a bit repetitive with other functions. After hiding, the debugging interface will become more concise

haneefdm, 2022-08-04:
Yes, you can right-click and hide any panel. Also, click on the there dots `...` on top right.