Marus/cortex-debug #714: `rtt_start_retry` doesn't seem to work

werediver, 2022-08-10:
I switched to v1.5.5 (pre-release) in hope `rtt_start_retry` option can help with enabling RTT via OpenOCD automatically (stopping after RTT initialization and manually issuing `monitor rtt start` works), but it doesn't seem to have any effect.

I understand this is an unstable feature, so just providing feedback.

haneefdm, 2022-08-11:
Thank you for trying it out. I had a 70% shot at it so I took it. Could you give me some details of what device and which openocd (xpack, from ST, etc.) There are so many versions of custom openocds so I want to have some idea.

I can create some logging data so we can debug in your setup why it is not working. But first, I will try to duplicate it myself with the info above.

werediver, 2022-08-11:
I was trying to set up RTT for Rasperry Pi Pico / RP2040 target using another Pico with [raspberrypi/picoprobe](https://github.com/raspberrypi/picoprobe) as a debugger. The picoprobe, unfortunately, is not a CMSIS-DAP compatible debugger and requires a custom [raspberrypi/openocd](https://github.com/raspberrypi/openocd) build.

If I stop the firmware after RTT initialization and manually issue `monitor rtt start` command, RTT is attached and the plug-in works as expected. Without doing this manual maneuver, though, RTT doesn't get attached, as OpenOCD is searching for a control block before the target is initialized.

I understand it's more of an OpenOCD issue, so today I tried switching to [ciniml/rust-dap](https://github.com/ciniml/rust-dap) (a CMSIS-DAP compliant debugger running on Pico) and [pyocd/pyOCD](https://github.com/pyocd/pyOCD). pyOCD supports RTT (via `pyocd rtt` command), but `monitor rtt start` doesn't work with it:

```
60-interpreter-exec console "monitor rtt start"
-> @"Error: unrecognized command 'rtt'\n"
Error: unrecognized command 'rtt'
-> 60^done
```

Again, this is an issue with pyOCD and not the plug-in, unless we start talking about custom pyOCD RTT support, but that would be a topic for a separate ticket.

I could use a CMSIS-DAP debugger with the vanilla OpenOCD too, but I assume the vanilla OpenOCD has the same issue with RTT initialization.

haneefdm, 2022-08-11:
Glad pyocd now supports RTT but support for it is not in Cortex-Debug as there were no requests and not sure how that one works. Only (latest?) OpenOCD or JLink for now.

There is no standardization on what monitor (aka. gdb-server) commands a given gdb-server supports.  Anything after the word monitor is sent to the gdb-server to act on. Syntax/methods/command-set are different. In OpenOCD, it might be `rtt start` but in pyOCD, it can be something totally different.

Also, CMSIS-DAP may not be related. All that needs to work is SWD and a gdb-server providing the additional service for RTT.

werediver, 2022-08-11:
> Also, CMSIS-DAP may not be related. All that needs to work is SWD and a gdb-server providing the additional service for RTT.

Sure, the difference CMSIS-DAP makes is with a CMSIS-DAP compliant debugger the vanilla upstream OpenOCD can be used, no outdated custom builds.

I should try a Pico with a CMSIS-DAP compliant firmware and the vanilla upstream OpenOCD.

Another possible combination is a CMSIS-DAP debugger with [Black Magic Debug App](https://black-magic.org/#black-magic-debug-app-bmda). I'm inclined to try the latter first, especially given that I should receive BMP v2.3a soon.

In the meantime, I've opened https://github.com/pyocd/pyOCD/issues/1439 to check on pyOCD `monitor rtt` support.

haneefdm, 2022-08-11:
RTT is very custom and unrelated to CMSIS-DAP. RTT is a layer above basic SWD. Not many gdb-servers support it. Until recently OpenOCD didn't. Only SEGGER, the creators of RTT did it with JLink and it had nothing to do with CMSIS. RTT even works with non ARM processors.

haneefdm, 2022-08-11:
Servers like JLink/openocd/pyocd need to know about RTT, its protocol and open TCP ports for clients to communicate with. Unrelated to CMSIS. JLink for instance only opens one port for one channel. But OpenOCD can open multiple TCP ports -- one per each channel.

werediver, 2022-08-11:
You're right on everything and know more details on the debugging protocols and software then me, but my point still holds:

- with https://github.com/raspberrypi/picoprobe I _have_ to use the outdated https://github.com/raspberrypi/openocd
- if I upload a CMSIS-DAP compliant firmware on the same hardware, I can choose from at least the following:
  - upstream OpenOCD
  - PyOCD
  - BMPA

Don't you agree CMSIS-DAP does make a difference? Even if unrelated directly to RTT.

haneefdm, 2022-08-12:
I have not seen pyOCD's RTT implementation. Your issue over there may be misplaced. JLink for instance does not even need `monitor rtt start` and nor does it have such a command. pyOCD implemented RTT very recently and I have not had a chance to look at it so chances of it working are near zero with our extension. And nothing to do with CMSIS -- JLink who invented this proves that you do not need CMSIS.

Anyway. If you want to work with me we can make this work with your setup.

werediver, 2022-08-12:
I tried some more combinations here, but in the end I'm still interested in making RTT retry logic (`rtt_start_retry`) work with OpenOCD. I'd appreciate your help.

From what I see in the log, the retry doesn't seem to ever happen, only the initial attempt. Below is the log filtered by "rtt":

```
23-interpreter-exec console "monitor rtt setup 0x2003fba0 10 {SEGGER RTT}"
24-interpreter-exec console "monitor rtt server start 60001 0"
-> @"Listening on port 60001 for rtt connections\n"
Listening on port 60001 for rtt connections
25-interpreter-exec console "monitor rtt start"
-> @"rtt: Searching for control block 'SEGGER RTT'\n"
rtt: Searching for control block 'SEGGER RTT'
-> @"rtt: No control block found\n"
rtt: No control block found
```

<details>
<summary>
Full log
</summary>

```
Cortex-Debug: VSCode debugger extension version 1.5.5 git(e98ea06). Usaage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Users/ramanf/Dropbox/Projects/escale/escale_fw_rs/target/thumbv6m-none-eabi/debug/escale
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Users/ramanf/Dropbox/Projects/escale/escale_fw_rs/target/thumbv6m-none-eabi/debug/escale
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /Users/ramanf/Dropbox/Projects/escale/escale_fw_rs -f /Users/ramanf/.vscode/extensions/marus25.cortex-debug-1.5.5/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/rp2040-custom.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 14 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GDB) 11.2\n"
-> ~"Copyright (C) 2022 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=aarch64-apple-darwin21.5.0 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
GNU gdb (GDB) 11.2
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=aarch64-apple-darwin21.5.0 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source /Users/ramanf/.vscode/extensions/marus25.cortex-debug-1.5.5/support/gdbsupport.init"
-> 6^done
[... 168 lines cut ...]
openocd -> 'type target_event event gdb-halt'
openocd -> 'type target_event event gdb-end'
openocd -> 'type target_event event halted'
openocd -> 'type target_event event step-end'
openocd -> 'type target_event event resume-start'
openocd -> 'type target_event event resumed'
openocd -> 'type target_state state running'
openocd -> 'type target_event event resume-end'
openocd -> 'type target_event event gdb-start'
```

</details>

The set-up is
- a CMSIS-DAP probe ([ciniml/rust-dap](https://github.com/ciniml/rust-dap) on a Pico)
- [raspberrypi/openocd](https://github.com/raspberrypi/openocd) 0.11.0-g228ede43d
  - the target config is insignificantly modified to work with a CMSIS-DAP probe instead of the intended [raspberrypi/picoprobe](https://github.com/raspberrypi/picoprobe)
  - behaves exactly the same with picoprobe
- the latest pre-release (v1.5.5) Cortex Debug plug-in in VS Code

I suppose, I could try debugging the plug-in locally. Will take some time to figure out the process, though, because I'm not typically working with NodeJS, Electron, and JS/TS.

Looking forward for any hints, advice, or extra info requests.

On a side note. I've learned that BMP got RTT support recently (not included in v1.8.2): [blackmagic-debug/blackmagic/blob/main/UsingRTT.md](https://github.com/blackmagic-debug/blackmagic/blob/main/UsingRTT.md). I assume, Cortex Debug doesn't support RTT via BMP yet. It may be a cool future addition too 👀

haneefdm, 2022-08-13:
Please hang on...I may have some good news shortly. We have to focus on OpenOCD. Probe/other-SW does not matter.

haneefdm, 2022-08-13:
Can you please try this build I created manually (not published to marketplace). You have ot install it manually

https://github.com/Marus/cortex-debug/releases/tag/v1.5.6-pre1

Give me feedback. I create a launch.json to log all transactions to a file and we can enable that if it is still not working.

werediver, 2022-08-14:
@haneefdm In a few tests I made, [v1.5.6-pre1](https://github.com/Marus/cortex-debug/releases/tag/v1.5.6-pre1) worked perfectly! I tried:

- Running with debugging, without breakpoints (no `runToEntryPoint` either)
- Running with debugging with a breakpoint _before_ RTT initialization

The plug-in manages to attach RTT in both cases.

RTT doesn't work, if I use "Run Without Debugging", but, I suppose, that's to be expected.

Thanks a lot for implementing the fix! Looking forward to the next release.

haneefdm, 2022-08-14:
Thank you for the feedback. I will make a release (pre) soon. I need to remove some of the debug messages that are still in there.

Yes, `Run Without Debugging` does not do many of the things we normally do....technically, some things are possible but require a ton of changes and may break the normal flow.

closed 2022-08-14 by haneefdm