Marus/cortex-debug #715: Start JLinkGDBServer with different RTT port

robertcragie, 2022-08-12:
I am running two targets in parallel and whilst everything else works fine, I cannot set two different RTT ports for logging. Looking at the code, it seems that whilst there is support for RTT in the configuration, there is no way to set any port other than the default (19021). Segger provides the command line option -RTTTelnetPort <port> which can be used to set a different RTT port for each instance of JLinkGDBServer. It would be really useful to have this support in.

robertcragie, 2022-08-12:
I tried using serverArgs but this didn't seem to work presumably because it put the argument in quotes: "-RTTTelnetPort 19022"

haneefdm, 2022-08-12:
You said "-RTTTelnetPort 19022". That is not one arg. That is two args. JLink SW will not split that into two. 

We support RTT directly via launch.json if you are interested. Just start typing "RTT" and hit tab and Intellisense should give you a template. If you use our RTT client, then we will pick a TCP port for your second target if the first one took the default PORT. Basically, we look to see if 19021 is busy and if so, we pick another one.

There have been some changes recently to our RTT client, so I recommend the pre-release available from the Extensions tab in VSCode. If you are using JLink's RTT client, then you don't need ours.

https://github.com/Marus/cortex-debug/wiki/SEGGER-RTT-support

closed 2022-08-12 by haneefdm

robertcragie, 2022-08-14:
> You said "-RTTTelnetPort 19022". That is not one arg. That is two args. JLink SW will not split that into two.

That's what I thought, however it doesn't work. I checked JLinkGDBServerCLExe on the command line using a different port and telnet'ed to it and that did work. So no idea what is going on.

I will try the built-in RTT client instead.

haneefdm, 2022-08-14:
How about attaching what you see in the Debug Console and the gdb-server terminal windows?

Also the output of how you made it work from the command-line.

So we can see the differences.

robertcragie, 2022-08-15:
In the following, I am deliberately using the "next" RTT telnet port (19022). If I use:
```
            "serverArgs": ["-rtttelnetport 19022"],
```
in `launch.json`, I get the following on the debug console:
```
Launching gdb-server: /usr/bin/JLinkGDBServerCLExe -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device nRF52840_xxAA -select usb=683220351 "-rtttelnetport 19022"
```
If I try to telnet to port 19022, I get:
```
telnet localhost 19022
Trying 127.0.0.1...
telnet: Unable to connect to remote host: Connection refused
```
If I use:

```
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "decoders": [
                    {
                        "label": "",
                        "port": 0,
                        "type": "console"
                    }
                ]
            },
```
in `launch.json`, I get the following on the debug console:
```
Launching gdb-server: /usr/bin/JLinkGDBServerCLExe -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device nRF52840_xxAA -rtttelnetport 19022 -select usb=683220351
```
The RTT terminal appears and I can see the output. If I try to telnet to port 19022, I get:
```
telnet localhost 19022
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
ERROR: Connection refused - There already is an active connection.
Connection closed by foreign host.
```
as I would expect.

I am happy to use the RTT terminal in VSCode so no need for anything further from my perspective.