Marus/cortex-debug #716: Allow loading ipAddress in launch.json from inputs variable

mwerner96, 2022-08-13:
I have configured my `launch.json` to interface with a jLink-pro via ethernet. Inside the `launch.json`, I have hardcoded it's `ipAddress` field to the correct value. However, since I frequently switch between the office and remote work, I constantly have to change the `ipAddress` field.

Would it be possible to not only accept IP-addresses for this field but also VS Code input variables? For example, this works for the `cwd` field:

    "configurations": [
        {
            "cwd": "${input:test}",
            "servertype": "jlink",
            "ipAddress": "192.168.xxx.yyy",
            ...
        }
    ],
    "inputs": [
        {
            "id": "test",
            "type": "pickString",
            "options": [
                "/home/username/projects/project"
            ],
            "description": "important description"
        }
    ]

If this could work for the `ipAddress`, I could write a script which automagically detects the jLinks IP and hand it to the `${input:...}` with the [shell input extension](https://marketplace.visualstudio.com/items?itemName=augustocdias.tasks-shell-input).

haneefdm, 2022-08-13:
Totally at a loss of what this is. None of this I recognize and I maintain this extension.

@mwerner96 Are you sure you are working with our extension? I would love to know which one you are actually using.

Can someone help?

haneefdm, 2022-08-13:
If someone doesn't respond, I will close this as I do not understand at all, how this issue is related to cortex-debug.

Please post the entire launch.json for others to decipher and help. I am 99% sure it has nothing to do with us.

Wild guess, you are using Microsoft's cpptools extension Not ours.

mwerner96, 2022-08-13:
Hi, I am sorry that I didn't clearly communicate my intent. I am quite certain that I use your extension. This is the `launch.json` I am currently using:

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "executable": "build/LPC55S06/executable.axf",
            "preLaunchTask": "Build executable.axf",
            "name": "Debug LPC55S06",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "serverpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe",
            "armToolchainPath": "/home/mwerner/gcc-arm-none-eabi-10.3-2021.10/bin",
            "device": "LPC55S06",
            "interface": "swd",
            "ipAddress": "192.168.xxx.yyy",
            "svdFile": "libs/LPC55S06/LPC55S06.xml",
            "showDevDebugOutput": "both",
            "runToEntryPoint": "main",
        }
    ]
}
```

What I would like to be able to do is the following, where `find_jlink_ip` is some shell script I provide to identify the jlinks IP-address:

```
{
    "version": "0.2.0",
    "configurations": [
        {
            ...
            "ipAddress": "${input:jlink-ip}",
            ...
        }
    ],
    "inputs": [
        {
            "id": "jlink-ip",
            "type": "command",
            "command": "shellCommand.execute",
            "args": {
                "command": "find_jlink_ip",
                "cwd": "${workspaceFolder}",
            }
        }
    ]
}
```

I hope this is actually relevant for the `cortex-debug` extension and I am not completely off track, otherwise please close this issue.

haneefdm, 2022-08-13:
Oh, I see what you are trying to do. Sorry, it is outside our extension's purview.  Entire launch.json is first filtered/processed by VSCode and we only see the final result of the one section that applies to the current debug session.

It may be possible, what you are trying to do. Best to ask on [stack overflow](https://stackoverflow.com/) or with Microsoft.

closed 2022-08-13 by haneefdm