Marus/cortex-debug #719: Corrupting memory on Cortex-A8

nonarkitten, 2022-08-19:
When set up to run on my Cortex-A8, it seems to corrupt some memory locations. I suspect this is because the chip's built in bootloader doesn't automagically scan the SRAM for a program and continues, messing up the program. How do we halt the debugger immediately after load and then specify an entry point?

haneefdm, 2022-08-19:
There is `breakAfterReset` or `runToEntryPoint` or you can use a gdb command in your `preLaunchCommands` to set a breakpoint at any address.

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

If you are comfortable with gdb, it is all gdb under the hood. You can see each and every command and response from gdb in the Debug Console if you enable that chatter. It can be verbose but you can tell exactly what is going on.

Once we issue a reset via your gdb-server we just wait for things to happen. If your bootloader loads something, that is fine, we just wait until  a breakpoint is hit somewhere or if you force a pause

haneefdm, 2022-08-19:
See instructions in `Debug Console` to enable the extra chatter.

nonarkitten, 2022-08-19:
Thanks. Had to try a few different options, the pre/post ones didn't seem to work, but the override did. This seems to be working fairly reliably so far and is nice and quick.
```
            "breakAfterReset": true,
            "overrideLaunchCommands": [
                "monitor reset halt",
                "load",
                "monitor reset halt",
                "break main",
                "jump Reset_Handler"
            ],
```

closed 2022-08-19 by nonarkitten