Marus/cortex-debug #721: GDB Suddenly Stopped Working - GDB Server Quits Unexpectedly

apollolabsdev, 2022-08-23:
I've been working with the GDB Debugger and had a working setup with OpenOCD and Cortex-M debug that stopped working all of a sudden and I'm at a loss why. The GDB Server session abruptly ends as shown below. I didn't update OpenOCD or GDB however I did do a VSCode update.

```
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
[2022-08-23T18:17:28.657Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```
Heres also my launch.json

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug (OpenOCD)",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}",
            "executable": "./target/thumbv7em-none-eabihf/debug/stm32f4xxgpio",
            "preLaunchTask": "build",
            "device": "STM32F401RET6U",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32f4x.cfg"
            ],
            "openOCDLaunchCommands": [
                "init",
                "reset init"
            ],
            "svdFile": "${workspaceRoot}/STM32F401.svd",
            "runToEntryPoint": "main",
        }
    ]
}
```

haneefdm, 2022-08-23:
We also had an update yesterday. The contents of the `DEBUG CONSOLE` are very important. If you can run all the commands we launch from a terminal, that would be great. There have been so many releases of various tools and it is not clear what is breaking.

Since OpenOCD is quitting, try to launch that one first -- use exact same command-line. Also make sure gdb can even run. Sometimes, improper installation with no fault of yours of these tools can cause these issues. They are being packaged badly with missing dependencies.

We have had no other reports among the thousands of users out there. Alo, this same version was available as a pre-release for a month or more.

apollolabsdev, 2022-08-24:
So it turns out I somehow had several versions of OpenOCD installed. I figured this out as I started uninstalling the versions I knew about and somehow OpenOCD was still launching. At one point the setup started working all over again when I thought I had uninstalled everything. Not sure what triggered the issue all of a sudden though. I found all instances of OpenOCD on my device and cleaned up everything and started with a clean slate. It might be worth noting that I did have an issue with OpenOCD and GDB using STLink with an older version of OpenOCD. I think it was the 0.11.0 stable homebrew release and it was a known issue. It seems that somehow my system was reverting back to that version but didn't occur to me that it's still lingering in my system.

haneefdm, 2022-08-24:
With ST, the only OpenOCD I trust is the one they package and ship with their IDE. You have run a `find` to see what directory openocd is in, also do the same for the scripts search path.

To me, the version number printed by OpenOCD means very little as there are many sub-versions and custom builds/mods

closed 2022-08-24 by haneefdm