Marus/cortex-debug #722: Add `.vsix` artifacts to releases on GitHub

abougouffa, 2022-08-23:
Hello and thank you for this extension,

I'm using `cortex-debug` with Emacs (via `dap-mode`), I want to automate the download process so I can update it from withing Emacs, however, the VSCode workplace do not provide direct links.

Can you please provide the `.vsix` with the release files (like in [v1.5.6-pre1](https://github.com/Marus/cortex-debug/releases/tag/v1.5.6-pre1) release)?

I'm working on adding the `cortex-debug` support to the upstream [`dap-mode`](https://github.com/emacs-lsp/dap-mode) so Emacs users can use it also.

Thank you again

haneefdm, 2022-08-23:
Hehehe, I thought I was the last human using Emacs.

Could you not download it from the Marketplace? It may even have an API to download any version while they only show the last few. I only publish on github releases for making a tag and if someone (including myself) want to recreate the source at that point and build/deploy. The build to create a vsix is just a few commands.

[image]

abougouffa, 2022-08-23:
> Hehehe, I thought I was the last human using Emacs.
😆 🤣

In fact, VS Marketplace is blocking download requests, direct links works fine from the browser, however, if you try to make a script which automate downloading, it get rejected.

For example, I was facing the same issue with the `cpptools` extension, which give me sometimes this error:

```json
{
  "$id":"1",
  "innerException":null,
  "message":"Request was blocked due to exceeding usage of resource 'Count' in namespace 'AnonymousId'. For more information on why your request was blocked, see the topic \"Rate limits\" on the Microsoft Web site (https://go.microsoft.com/fwlink/?LinkId=823950).",
  "typeName":"Microsoft.TeamFoundation.Framework.Server.RequestBlockedException, Microsoft.TeamFoundation.Framework.Server",
  "typeKey":"RequestBlockedException",
  "errorCode":0,
  "eventId":3000
}
```

And sometimes this one:

```json
{
  "$id":"1",
  "innerException":null,
  "message":"TF400898: An Internal Error Occurred. Activity Id: e15a6fba-431f-4173-9cc3-cc75cfc27a0c.",
  "typeName":"System.NullReferenceException, mscorlib",
  "typeKey":"NullReferenceException",
  "errorCode":0,
  "eventId":0
}
```

Hopefully, the developper is providing `.vsix` files on GitHub, so [I patched the Emacs package to download from it](https://github.com/emacs-lsp/dap-mode/blob/61c8825fd5743139f4b7d8dc93ffbdbf3b2a592b/dap-cpptools.el#L46-L57).

Well, the Marketplace works fine as long as you are using VS Code!

haneefdm, 2022-08-24:
Okay, I have no issues publishing the VSIX file via github. Except, it is a bunch of extra error-prone work. I am not a git or github expert doing it via the web interface is painful.

Can you/someone provide a script that I can use to publish a release (pre-release or release) on github

1.   It should create a tag from my package.json
2.   Publish the release on github that includes the source zip/tar of the source and the vsix artifacts. This is the part I need REAL HELP with. It should mirror what I would have done normally/manually. I am sure github has an API to do this but...**help needed**

I can insert/add my commands to create a vsix

I want it to be a process that people can rely on. Not based on dozens of manual steps that I can screw up on.

abougouffa, 2022-08-24:
Yep, I can feel the pain!

Unfortunately, I'm not a DevOps expert too, otherwise, I would be happy to do it.

Do not mind, I will find a way to build the extension from source from withing Emacs.

For people who are experienced with GitHub actions/workflow, they can find an inspiration [in this file from WebFreak's debug extension.](https://github.com/WebFreak001/code-debug/blob/master/.github/workflows/release.yml)

PhilippHaefele, 2022-08-24:
Some time ago i found https://dev.to/sourishkrout/the-github-action-you-need-to-publish-vs-code-extensions-1n8 but didn't had the time to go through it. Maybe this helps with this point and also publishing the releases to the marketplaces

haneefdm, 2022-08-24:
Thanks, I will take a look

haneefdm, 2023-03-22:
Btw, you can download most of our extensions now from openvsx.org. I will be publishing to both VSCode and OpenVSX simultaneously. But releases only. OpenVSX does not understand pre-releases

haneefdm, 2023-03-22:
I do not wish to make releases as a github action as popular as that may be. I think I can use the GH CLI to make a release.

https://cli.github.com/manual/gh_release_create

haneefdm, 2025-01-17:
Unless someone is willing to help, I am closing this issue

PhilippHaefele, 2025-01-19:
I'm willing to do something here.
@haneefdm As you don't like the GitHub action Release triggering, should I add it to the release script?

I would use the https://www.npmjs.com/package/gh-release to create the release with all the needed options and vsix asset upload. Any concerns against that?

haneefdm, 2025-01-19:
That script is exactly what I wanted. Some things go wrong during the release process and I want to those failures on my computer and interact with tools directly. 

I have 3-5 personal access tokens (azure devops, github, marketplace, Open VSX, etc.), that keep expiring and I have to renew them manually. I also don't want to store my tokens on github. Call me parnoid.

Even `vsce` gives (false) errors sometimes and I have to deal with it.

Assigning to you. No hurry, but I want to try to make some draft/pre-releases