Marus/cortex-debug #723: VS Code with Dev-containers: gdb server launches and exits right away

sidmodi-mw, 2022-08-24:
I have been using the Cortex-Debug extenstion with the VS Code Remote Containers extension without any issues for a few months now - I have a launch.json file that has successfully let me debug my firmware. Until I have version 1.4.4, I had no problems. However, I rebuilt my container today and the extension got updated to 1.6.0 and I noticed that I wasn't able to debug anymore - the moment I tried to launch the debugger, the server would launch and exit right away:

```

haneefdm, 2022-08-24:
Thanks for reporting this.

Can you run the exact command line you see in the Debug Console manually in a shell? Do that for both gdb and jlinkgdbserver. Use the same `cwd` as well for launching those tools.

haneefdm, 2022-08-24:
Also, the contents of the `Debug Console`.

sidmodi-mw, 2022-08-24:
This is the output from the Debug Console with `showDevDebugOutput` set to `raw`:

```
Cortex-Debug: VSCode debugger extension version 1.6.0 git(3026299). Usaage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from /usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-objdump --syms -C -h -w /workspaces/vhealth_application/_pca10056/s140/armgcc/_build/nrf52840_xxaa.out
Reading symbols from /usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-nm --defined-only -S -l -C -p /workspaces/vhealth_application/_pca10056/s140/armgcc/_build/nrf52840_xxaa.out
Launching GDB: /usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Launching gdb-server: JLinkGDBServer -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device nrf52840_xxaa -rtttelnetport 19021 -select usb=51021094
    Please check TERMINAL tab (gdb-server) for output from JLinkGDBServer
Finished reading symbols from objdump: Time: 115 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git\n"
-> ~"Copyright (C) 2019 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-linux-gnu --target=arm-none-eabi\".\n"
```

sidmodi-mw, 2022-08-24:
This the the command I ran for `JLinkGDBServer`:

`JLinkGDBServer -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device nrf52840_xxaa -rtttelnetport 19021 -select usb=51021094`

Output:
```
SEGGER J-Link GDB Server V7.58b Command Line Version

JLinkARM.dll V7.58b (DLL compiled Nov 16 2021 15:04:27)

Command line: -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device nrf52840_xxaa -rtttelnetport 19021 -select usb=51021094
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 nrf52840_xxaa
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

RTT Telnet Port set to 19021
Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V11 compiled Nov 16 2021 11:59:06
Hardware: V11.00
S/N: 51021094
Feature(s): GDB
Checking target voltage...
Target voltage: 1.98 V
Listening on TCP/IP port 50000
Connecting to target...
Connected to target
Waiting for GDB connection...
```

sidmodi-mw, 2022-08-24:
This is the command I ran for `gdb`:

`/usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gdb -q --interpreter=mi2`

Output:
```
=thread-group-added,id="i1"
(gdb)
```

haneefdm, 2022-08-24:
> After I ran the above command, I did not see any output in the terminal the server was running in.

That is because gbd needs to connect to the server and that requires commands and then JLink will respond as needed. But not needed right now. It shows that both programs are installed properly and able to start.

In your system tmp dir (/tmp in Linux and $TMPDIR on Mac, etc.) there is a file `cortex-debug-server-exiting.log` that show what happened during the last few sessions. We could see which program quit and perhaps why/how (crash, killed, exited normally, etc.)

sidmodi-mw, 2022-08-24:
I just did a fresh attempt, here are the contents of `cortex-debug-server-exiting.log`

```
[2022-08-24T21:16:19.336Z] ppid=524 pid=524 nRF52 App (_pca10056): ******* Starting new session request type="launch"
[2022-08-24T21:16:19.398Z] ppid=524 pid=534 GDB started ppid=524 pid=534
```

haneefdm, 2022-08-24:
Hmmm. I don't see anything quitting/crashing... I only see things starting. Could things be buffered and not flushed to the file? Cannot be. But something clearly died and you should see a message even if our extension is the one that (almost) died. That log should look something like.
```
[2022-08-24T20:08:08.240Z] ppid=13477 pid=13477 Debug with OpenOCD: ******* Starting new session request type="launch"
[2022-08-24T20:08:08.254Z] ppid=13477 pid=13522 GDB started ppid=13477 pid=13522
[2022-08-24T20:18:19.973Z] ppid=13477 pid=13522 Debug with OpenOCD: Begin disconnectRequest
[2022-08-24T20:18:20.085Z] ppid=13477 pid=13522 GDBServer(13523): exited code=0 signal=null
[2022-08-24T20:18:20.101Z] ppid=13477 pid=13522 Debug with OpenOCD: disconnectRequest sendResponse 2
[2022-08-24T20:18:20.333Z] ppid=13477 pid=13522 GDB: exited
[2022-08-24T20:18:20.344Z] ppid=13477 pid=13522 Debug with OpenOCD: quitEvent: sending VSCode TerminatedEvent
```

sidmodi-mw, 2022-08-24:
No luck - every time I try it, I just get the same messages:

```
root@eba51253ed3e:/workspaces/application# cat /tmp/cortex-debug-server-exiting.log 
[2022-08-24T23:30:59.210Z] ppid=1006 pid=1006 nRF52 App (vhealth_rev_a): ******* Starting new session request type="launch"
[2022-08-24T23:30:59.233Z] ppid=1006 pid=1015 GDB started ppid=1006 pid=1015
[2022-08-24T23:36:22.197Z] ppid=2073 pid=2073 nRF52 App (vhealth_rev_a): ******* Starting new session request type="launch"
[2022-08-24T23:36:22.223Z] ppid=2073 pid=2082 GDB started ppid=2073 pid=2082
[2022-08-24T23:39:10.219Z] ppid=2553 pid=2553 nRF52 App (vhealth_rev_a): ******* Starting new session request type="launch"
[2022-08-24T23:39:10.245Z] ppid=2553 pid=2562 GDB started ppid=2553 pid=2562
[2022-08-24T23:39:26.910Z] ppid=2611 pid=2611 nRF52 App (vhealth_rev_a): ******* Starting new session request type="launch"
[2022-08-24T23:39:26.917Z] ppid=2611 pid=2618 GDB started ppid=2611 pid=2618
[2022-08-24T23:39:50.731Z] ppid=2685 pid=2685 nRF52 App (vhealth_rev_a): ******* Starting new session request type="launch"
[2022-08-24T23:39:50.742Z] ppid=2685 pid=2692 GDB started ppid=2685 pid=2692
[2022-08-24T23:41:01.352Z] ppid=2878 pid=2878 nRF52 App (vhealth_rev_a): ******* Starting new session request type="launch"
[2022-08-24T23:41:01.361Z] ppid=2878 pid=2885 GDB started ppid=2878 pid=2885
[2022-08-24T23:41:26.607Z] ppid=2959 pid=2959 nRF52 App (vhealth_rev_a): ******* Starting new session request type="launch"
[2022-08-24T23:41:26.622Z] ppid=2959 pid=2968 GDB started ppid=2959 pid=2968
[2022-08-24T23:41:28.259Z] ppid=2974 pid=2974 nRF52 App (vhealth_rev_a): ******* Starting new session request type="launch"
[2022-08-24T23:41:28.274Z] ppid=2974 pid=2984 GDB started ppid=2974 pid=2984
```

haneefdm, 2022-08-25:
I get the feeling that gdb is not starting. In your earlier post showing the Debug Console contents, we executed one command and I did not see any other commands come from us.

Can you start gdb just like how we did and execute this command
```
1-gdb-version
```
That is all we were able to do. I also see a few lines missing from the `gdb-version` command output. That is not the full version info. We couldn't even get to the point of starting the gdb-server. Note that if you are using gdb version less than 9, then the program will exit but then you should see a message indicating that.

After querying the version, the very next thing we do is
```
2-gdb-set mi-async on
```
If we can't figure out what is going on, we may have to debug this extension. I hope we don't have to go there. 

Where did you download/get your version of gdb?

boraozgen, 2022-08-25:
I have the same problem with GDB version 9-2020-q2-update. Using version 10.3-2021.10 fixes it.

haneefdm, 2022-08-25:
Yes, gdb version 8 no longer works. It needs to be 9+. But you should have seen an error message if it is <= 8. We had been printing a warning for almost a year before we finally removed support for version 8. It had too many issues and required workarounds from our end that were becoming hard to maintain. I messed the following line that was posted above.

```
-> ~"GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git\n"
```
I have to see why we are not seeing an error message and a proper exit...

haneefdm, 2022-08-25:
Btw, we use MS version numbering convention. `X.Y.Z`. Where odd `Y` numbers are pre-releases and even ones are final releases. You can subscribe to pre-releases if you like. Regardless, unless you disable auto-updates, VSCode auto-updates all extensions when they are updated.

I don't like the versioning system but have no choice and the best way for us to share pre-releases.

haneefdm, 2022-08-26:
I have improved the error message you get when an incompatible version of gdb is used.

sidmodi-mw, 2022-08-26:
Awesome, thank you! It only gave a warning and not an error last time I ran it so I didn't upgrade yet. I will upgrade it now.