Marus/cortex-debug #725: _SEGGER_RTT no longer found automatically

jwbmwv, 2022-08-25:
I have been successfully using the RTT feature since it was released with  "address": "auto". With the release of v1.6.0 this appears to have broken.

Here is the error in the DEBUG CONSOLE window (edited for brevity):

Cortex-Debug: VSCode debugger extension version 1.6.0 git(3026299). Usaage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from C:\Users\U170623\.vscode\extensions\chipcode-nl.gcc-arm-windows-1.0.1\bin\arm-none-eabi-objdump.exe --syms -C -h -w C:/Users/USERNAME/Projects/LRF/ULF/IAR/Debug/MyProg/Exe/MyProgout
Reading symbols from c:/users/USERNAME/.vscode/extensions/chipcode-nl.gcc-arm-windows-1.0.1/bin/arm-none-eabi-nm.exe --defined-only -S -l -C -p C:/Users/USERNAME/Projects/LRF/ULF/IAR/Debug/MyProg/Exe/MyProg.out
Launching GDB: "C:\\Users\\USERNAME\\.vscode\\extensions\\chipcode-nl.gcc-arm-windows-1.0.1\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Launching gdb-server: "C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe" -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32H725VG -rtttelnetport 19021 -gui -speed 50000
    Please check TERMINAL tab (gdb-server) for output from C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe
Finished reading symbols from objdump: Time: 470 ms
Could not find symbol '_SEGGER_RTT' in executable. Make sure you compile/link with debug ON or you can specify your own RTT address

Here is the output of a manual objdump:

$ "C:\Users\USERNAME\.vscode\extensions\chipcode-nl.gcc-arm-windows-1.0.1\bin\arm-none-eabi-objdump.exe" --syms -C -h -w  ./MyProg.out | grep SEGGER_RTT
00000000 l    df *ABS*  00000000 c:\Users\U170623\Projects\LRF\ULF\ThirdParty\SEGGER\RTT\SEGGER_RTT.c
08033b90 l     F P1 ro  00000084 _INTERNAL_12_SEGGER_RTT_c_1275f953::_WriteNoCheck(SEGGER_RTT_BUFFER_UP*, char const*, unsigned int)
08033b04 l     F P1 ro  0000008c _INTERNAL_12_SEGGER_RTT_c_1275f953::_WriteBlocking(SEGGER_RTT_BUFFER_UP*, char const*, unsigned int)
08033c14 l     F P1 ro  0000001a _INTERNAL_12_SEGGER_RTT_c_1275f953::_GetAvailWriteSpace(SEGGER_RTT_BUFFER_UP*)
08033a90 l     F P1 ro  00000074 _INTERNAL_12_SEGGER_RTT_c_1275f953::_DoInit()
00000000 l    df *ABS*  00000000 SEGGER_RTT_ASM_ARMv7M
00000000 l    df *ABS*  00000000 c:\Users\USERNAME\Projects\LRF\ULF\ThirdParty\SEGGER\RTT\SEGGER_RTT_printf.c
08033c2e g     F P1 ro  00000082 SEGGER_RTT_WriteNoLock
08033cb0 g     F P1 ro  00000034 SEGGER_RTT_Write
08033ce4 g     F P1 ro  0000006c SEGGER_RTT_PutChar
2000698c g     O P4 zi  000000a8 _SEGGER_RTT

Here is the contents of my launch

        {
            "name":             "MyProg Debug (Jlink)",
            "cwd":              "${workspaceFolder}",
            "request":          "launch",
            "type":             "cortex-debug",
            "servertype":       "jlink",
            "serverArgs": [
                "-gui",
                "-speed", "50000"
            ],
//          "serialNumber":       "",     // If you have more than one J-Link probe, add the serial number here.
            "interface":        "swd",
            "device":           "STM32H725VG",
            "svdFile":          "${workspaceFolder}/ThirdParty/SVD/STM32/STM32H725.svd",
            "executable":       "${workspaceFolder}/IAR/Debug/MyProg/Exe/MyProg.out",
            "runToEntryPoint":  "main",
            "rttConfig":
            {
                "enabled": true,
                "address": "auto",
                // "clearSearch": false    // OpenOCD users may have to un-comment this
                "decoders": [
                    {
                        "type":             "console",
                        "label":            "RTT Trace Output",
                        "showOnStartup":    true,
                        "port":             0,
                        "timestamp":        true,
                    }
                ]
            },
        },

haneefdm, 2022-08-25:
Hmmm. It does not look like the output of the standard objdump but should have worked.
```
2000698c g O P4 zi 000000a8 _SEGGER_RTT
```
No idea what P4 is. What I get is something like this
```
200000a8 g     O .bss   000000a8 _SEGGER_RTT
```
Even spaces and tabs matter. Because everything is specified positional and only each char means something. Copy/paste probably messed it up. Please redirect output of objdump to a file and attach the same. That way nothing will be lost. The format of the header matters as well.  See also

https://linux.die.net/man/1/objdump

Even if you use someone else's compiler, you can still use the standard objdump/gdb since it is an elf file. Perhaps the following still works  (from ARM).

https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

Btw, do you see any globals and statics in the Variables window? And what about _SEGGER_RTT. It should show up there.

jwbmwv, 2022-08-25:
As an FYI here's the version info for objdump

GNU objdump (GNU Arm Embedded Toolchain 10-2020-q4-major) 2.35.1.20201028
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.

And yes, C & P does mess up whitespace!

haneefdm, 2022-08-25:
Would you mind attaching the entire output. You can strip out entire lines that may be secret stuff or create a small dummy application?

jwbmwv, 2022-08-25:
using an IAR compiler introduces a few twists vs GCC

P4 is a memory section

Sections:
Idx Name                Size      VMA       LMA       File off  Algn  Flags
  0 A0                  000002cc  08000000  08000000  00000034  2**2  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 P1 ro               0003a04c  080002cc  080002cc  00000300  2**3  CONTENTS, ALLOC, LOAD, READONLY, CODE
  2 P1 ui               0000000c  0803a318  0803a318  0003a34c  2**0  ALLOC, READONLY
  3 P1 b0               0000034f  0803a324  0803a324  0003a34c  2**0  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 P4 s0               00000634  20000000  20000000  0003a69c  2**2  ALLOC
  5 P4 zi               00006e34  20000634  20000634  0003a69c  2**2  ALLOC
  6 P4 ui               000085f8  20007468  20007468  0003a69c  2**3  ALLOC
  7 P5                  0000e800  24000000  24000000  0003a69c  2**3  ALLOC

haneefdm, 2022-08-25:
This is the line that parses the output of objdump...

https://github.com/Marus/cortex-debug/blob/8df2957a250000edbd2652c0b4b4ff5e033fab93/src/backend/symbols.ts#L17

This hasn't changed at all since 1.4.4. So, something else is broken. I am using RTT all the time myself.

The P4 part is fine but what I saw in the C & P was that it was in the wrong column.

jwbmwv, 2022-08-25:
Using objdump from cygwin

GNU objdump (GNU Binutils) 2.39
Copyright (C) 2022 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.

$ objdump --syms -C -h -w ./SPLIT-ATOM.out | grep SEGGER_RTT
00000000 l    df *ABS*  00000000 c:\Users\USERNAME\Projects\xxx\xxx\ThirdParty\SEGGER\RTT\SEGGER_RTT.c
08034059 l     F P1 ro  00000084 _INTERNAL_12_SEGGER_RTT_c_1275f953::_WriteNoCheck(SEGGER_RTT_BUFFER_UP*, char const*, unsigned int)
08033fcd l     F P1 ro  0000008c _INTERNAL_12_SEGGER_RTT_c_1275f953::_WriteBlocking(SEGGER_RTT_BUFFER_UP*, char const*, unsigned int)
080340dd l     F P1 ro  0000001a _INTERNAL_12_SEGGER_RTT_c_1275f953::_GetAvailWriteSpace(SEGGER_RTT_BUFFER_UP*)
08033f59 l     F P1 ro  00000074 _INTERNAL_12_SEGGER_RTT_c_1275f953::_DoInit()
00000000 l    df *ABS*  00000000 SEGGER_RTT_ASM_ARMv7M
00000000 l    df *ABS*  00000000 c:\Users\USERNAME\Projects\LRF\ULF-dev\ThirdParty\SEGGER\RTT\SEGGER_RTT_printf.c
080340f7 g     F P1 ro  00000082 SEGGER_RTT_WriteNoLock
08034179 g     F P1 ro  00000034 SEGGER_RTT_Write
080341ad g     F P1 ro  0000006c SEGGER_RTT_PutChar
20004a6c g     O P4 zi  000000a8 _SEGGER_RTT

jwbmwv, 2022-08-25:
> Would you mind attaching the entire output. You can strip out entire lines that may be secret stuff or create a small dummy application?

Unfortunately I cannot send that to you. I also don't have a "simple" means to create a sample app. Maybe tonight I'll try to re-create the issue on a sample app I'm working on at home.

Thanks

ptahpeteh, 2022-08-30:
I have the same problem. After upgrade to version 1.6.2 I started getting the error "Could not find symbol '_SEGGER_RTT' in executable.". If I downgrade the extension to 1.4.4 it works again. So it's only the extension change that causes the issue.

My output from objdump --syms -C -h -w  file.elf | grep _SEGGER_RTT:
```
2001c6cc g     O .bss   00000078 _SEGGER_RTT
```
And from nm --defined-only -S -l -C -p file.elf | grep _SEGGER_RTT
```
2001c6cc 00000078 B _SEGGER_RTT /firmware/sdk/external/segger_rtt/SEGGER_RTT.c:254
```

haneefdm, 2022-08-30:
I cannot duplicate what you all are seeing. No one wants to share the output of objdump and nm. A lot has changed since 1.4.x and I have used RTT with multiple devices although always with GCC. @ptahpeteh, you seem to be using GCC?

My hands are tied with nothing to debug with. While those two lines seem okay, maybe something happened before that or even after that is causing the issue.

ptahpeteh, 2022-08-31:
@haneefdm I'm using GCC on Linux. I don't think it is a problem with objdump because it does not changed on my side and I can switch from working to not-working by changing the cortex-debug version leaving the rest of the system unchanged. I attach output of my objdump. Is there anything else I can do/check on my side (like gather debug traces from the cortex-debug)?
[objdump.txt](https://github.com/Marus/cortex-debug/files/9459307/objdump.txt)

haneefdm, 2022-08-31:
Found the problem. There are two symbols at the same address

2001c6cc l       .bss	00000000 $d
2001c6cc g     O .bss	00000078 _SEGGER_RTT

We don't know how to handle duplicates. The fact that it worked before, is happenstance. I think we can safely ignore the one with size zero, but needs more research. Of course objdump is dumping what it sees, but I want to know what these $d, $t symbols are. They are also not objects (data) or functions (text) or files

There are a TON of conflicts here. Does your compiler/linker say what these are?

ptahpeteh, 2022-08-31:
These are not conflicts. It's ok if two or more symbols refer to the same address. They are aliases. I don't see any problem here. I don't know how you handle duplicate right not but I think you need kind of associative array with symbol name as the key in the array and symbol address as the value.

As to $ symbols I found this info:
```
The ARM ELF specification requires that special symbols be inserted into object files to mark certain features:
$a
    At the start of a region of code containing ARM instructions.
$t
    At the start of a region of code containing THUMB instructions.
$d
    At the start of a region of data. 
```
They are just kind of tags.

ptahpeteh, 2022-08-31:
This probably the source of the problem: https://github.com/Marus/cortex-debug/pull/648/commits/34f6ec794cf594a861f3a79c91658603e6220146

jwbmwv, 2022-08-31:
> Unfortunately I cannot send that to you. I also don't have a "simple" means to create a sample app. Maybe tonight I'll try to re-create the issue on a sample app I'm working on at home.
>
> Thanks

I'm glad someone else sees this if even for my own sanity!
I've tried to recreate the problem on a project I can share, but have not succeeded. It sounds like the reason for the error has been identified.

Thank You

haneefdm, 2022-08-31:
@ptahpeteh I am glad you found the docs. Not sure why I have never seen these symbols before. I have many executables -- all from GNU tools of course and I have never seen these symbols.

Yes, multiple symbol files were one thing, but the bigger thing was that I needed to find quickly function symbols by address for disassembly. For that, I needed a hash table (there were linear searches before and C++ produces an ungodly number of symbols) and we could not have dups, and even if we did, did not know how to interpret them and which one to use when searching.

Anyways, thanks for the objdump output. I had been scratching my head.

Could you tell me where you found this ARM spec. And, I wonder why the versions of GCC I used have zero symbols that start with a $. But I have seen those as $ is not an illegal char in some languages

haneefdm, 2022-08-31:
Both ARM and IAR also insert many special symbols to help out their own linkers and debuggers. Especioally when you enable link time optimizations.

I wish GDB gave us this information. It has only partials and sometimes even for partial, the output can be super large (10s of MB) or GDB simply crashing. So, we stopped asking GDB. Now, we have to use a combination of objdump and nm to get all the symbol information.

ptahpeteh, 2022-08-31:
I found the information about $d in the as assembler manual, for example here: https://docs.adacore.com/live/wave/binutils-stable/html/as/as.html#ARM-Mapping-Symbols. Just now I found also the arm specification for ELF: 
https://refspecs.linuxbase.org/elf/ARMELFA08.pdf. It also mention $d in Mapping Symbols chapter.
I guess the symbols starting with $ can be safely ignored?

haneefdm, 2022-08-31:
Thanks for that doument

>> I guess the symbols starting with $ can be safely ignored?

No, can't do that blindly. Bit more to it.

spoorcc, 2022-09-01:
> There is a custom build available for testing. You have to download and install from VSIX
> 
> https://github.com/Marus/cortex-debug/releases/tag/v1.6.3-pre3

I've been also experiencing this issue with IAR ARM 7.80.4 and this solves the problem for me, thank you 👍

haneefdm, 2022-09-01:
Thanks for the feedback, you all

I hate to ask for this, but could you do me a favor and look at this and vote for the issue on the MS Website if you agree.

https://github.com/Marus/cortex-debug/issues/729#issuecomment-1233293715

closed 2022-10-21 by haneefdm