Marus/cortex-debug #726: Loading raw flash images

cmcquinn, 2022-08-25:
I am working on a project that uses the [wolfBoot](https://github.com/wolfSSL/wolfBoot) bootloader on STM32 microcontrollers. The wolfBoot tools produce a "factory image" `.bin` file that is written directly to the microcontroller flash. The bootloader code lives at the start of the image, and then the application code is located at an offset later in the image, directly after a small header containing a signature used by the bootloader to validate the application code. See the [wolfBoot docs](https://github.com/wolfSSL/wolfBoot/blob/master/docs/flash_partitions.md) for more info.

I want to program the target microcontroller using this image, and then load the elf files for the bootloader and the application to allow debugging. I added the elf files to the `symbolFiles` property of my `launch.json` and added the `.bin` image to the `loadFiles` property of my `launch.json`, but this causes GDB to exit with the error: `not in executable format: file format not recognized (from file-exec-file "image.bin")`. I think this error results from GDB not recognizing the `.bin` file as containing executable code. How do I load the `.bin` file onto the MCU as part of the launch process?

My `launch.json` config is as follows:
```
"name": "blink",
"type": "cortex-debug",
"request": "launch",
"servertype": "openocd",
"cwd": "${workspaceRoot}",
"executable": "${workspaceRoot}/build/stm32h7/apps/blink/boards/stm32h7/blink",
"device": "STM32H753VI",
"interface": "swd",
"configFiles": [
    "/usr/local/share/openocd/scripts/board/st_nucleo_h743zi.cfg",
    "/usr/local/share/openocd/scripts/interface/stlink.cfg"
],
"loadFiles": [
    "${workspaceFolder}/build/stm32h7/apps/blink/boards/stm32h7/stm32h7_blink_factory.bin"
],
"symbolFiles": [
    {
        "file": "${workspaceFolder}/build/stm32h7/platforms/stm32h7/soc/stm32h753/wolfboot_platform_stm32h753"
    },
    {
        "file": "${workspaceRoot}/build/stm32h7/apps/blink/boards/stm32h7/stm32h7_blink"
    }
],
"svdFile": "${workspaceRoot}/.config/stm32_svd_files/STM32H753.svd",
```

haneefdm, 2022-08-25:
Did you try converting to a intel hex format to and use that in loadFiles?

haneefdm, 2022-08-25:
I just tried it and it works. I converted the bin to hex file
```
arm-none-eabi-objcopy -I binary -O ihex STM32F429ZI_NUCLEO_144.bin STM32F429ZI_NUCLEO_144.hex
```
and used the resulting file for loadFiles.
```
11-file-exec-file "/Users/hdm/STM32CubeIDE/workspace_1.1.0/UART_Printf/SW4STM32/STM32F429ZI_NUCLEO_144/Debug/STM32F429ZI_NUCLEO_144.hex"
-> 11^done
12-target-download
-> 12+download,{section=".sec1",section-size="10100",total-size="10100"}
-> 12+download,{section=".sec1",section-sent="10100",section-size="10100",total-sent="10100",total-size="10100"}
-> 12^done,address="0x00000000",load-size="10100",transferred-bits="80800",write-rate="10100"
```

closed 2022-08-25 by haneefdm

cmcquinn, 2022-08-26:
I just tried your suggestion and it downloads as you showed, but now the process ends with the error: `Failed to launch GDB: Load failed (from target-download)`.

haneefdm, 2022-08-26:
If you can't make it work with any gdb commands, neither can we. You can bypass gdb and cortex-debug and do your programming directly with openocd/jlink. To do that

1. Set `loadFiles` to an empty array so zero programming is done by us
2. Use `preLaunchCommands` to program the device using OpenOCD commands. Any command that starts with the word `monitor` goes directly to the gdb-server.

You can see all the monitor commands by doing `monitor help` in the case of OpenOCD. See also section 15.4 of this page

https://openocd.org/doc/html/General-Commands.html