Marus/cortex-debug #727: Can’t get variables values in Rust

pwnorbitals, 2022-08-26:
The "variables" and "watch" panels are always empty when trying to debug Rust code, I can’t get them to work... Adding watch values always result in a `not available` value

haneefdm, 2022-08-26:
Are you compiling/debugging with optimizations turned off and symbols saved? If not, you will see nothing. By default gcc does not for C/CPP programs unless you use the -g flag and hopefully -O0 or -Og. Maybe something similar for Rust.

Btw, the watch window is always empty until you add something to it but no symbols means you cannot add anything. We get the local variables from gdb. If gdb does not see any variables, then there are likely no symbols in your executable. You use a tool like `nm` to check (we also use the same tool).

haneefdm, 2022-08-26:
[image]

https://doc.rust-lang.org/rustc/codegen-options/index.html#debuginfo

[image]

https://doc.rust-lang.org/rustc/codegen-options/index.html#opt-level

Use opt-level=0 and dbginfo=2 and make sure you do use such flags while linking as well. A lot of people forget you have to do that during the linking step as well.

closed 2022-08-26 by haneefdm