Marus/cortex-debug #728: Debugger no longer starts with v1.6.0

pearson, 2022-08-26:
Since updating to v1.6.0, the debugger no longer starts for me. All that seems to happen is the following in the terminal pane:
`Waiting for gdb server to start...[2022-08-26T23:01:38.277Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
[2022-08-26T23:01:38.321Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...`

Reverting to v1.2.2 solves the issue.

I am using the remote development option with VS Code, running the IDE under Windows 11 with server running on WSL2 and the Ubuntu 20.04 distribution.

My `launch.json` contains the following configuration:
```json
{
	"type": "cortex-debug",
	"request": "launch",
	"name": "Debug (J-Link)",
	"cwd": "${workspaceRoot}",
	"preLaunchTask": "Build firmware",
	"executable": "${workspaceRoot}/build/zephyr/zephyr.elf",
	"servertype": "jlink",
	"device": "nRF52832_xxAA",
	"interface": "swd",
	"serialNumber": "",
	"runToMain": false,
	"svdFile": "${workspaceFolder}/../nrf52.svd",
},
```

I don't know if this is a bug due to the remote development environment, or a configuration issue on my part. Suggestions would be appreciated.

Thanks!

haneefdm, 2022-08-27:
Please make sure you are using version 9 or higher of gdb. We deprecate version <= 8. There was supposed to be a nicer error message but that never appears. We have been printing a deprecation notice over the last year or so that it would happen in July 2022

Let us know what version of gdb you are using...

pearson, 2022-08-27:
The version of gdb on the machine is 9.2. The version of the J-Link is v7.60g. (I'm not certain what interactions there are there.)

haneefdm, 2022-08-27:
You should be good then. We did not change anything that relates to remote development one way or another. What is in the Debug Console? We have to run each of commands displayed in the Debug Console manually to see which command is failing and why there is no indication

It appears JLink didn't even start. Version 1.2.2 is so old so I cannot remember what could have changed.

haneefdm, 2022-08-27:
Please check to make sure you can run your gdb from the command line. There have been many bad updates of GDB going around. Frustrating.

skfjirim, 2022-08-28:
For me version 1.6.0 is problematic too:

```
Error: Unable to start GDB. Make sure you can start gdb from the command-line and run any command like "echo hello".
Failed to generate gdb commands: Error: Could not start gdb, no response from gdb
Error: Could not start gdb, no response from gdb
    at Timeout._onTimeout (c:\Users\...\.vscode\extensions\marus25.cortex-debug-1.6.0\dist\debugadapter.js:2:144413)
    at listOnTimeout (node:internal/timers:557:17)
    at processTimers (node:internal/timers:500:7)
```
It was showing with gdb 8.1, now after I read your suggestion, it is showing same with gdb 11.2.

Downgrade of cortex-debug to 1.4.4 solved the issue for me.

haneefdm, 2022-08-28:
@skfjirim  Where did you get your 11.2 version of GDB, so I can download the same? Can you run gdb from the command line? If not you got a bad installation. We have been inundated with bad GDB packages.

skfjirim, 2022-08-28:
Yes, OC I tested reachability. I unstalled it via scoop with package mingw.

skfjirim, 2022-08-28:
Sorry, I checked again the settings of our compilation, we use specific version of ARM toolchain, which defined inside path, which is version 8.0.50.20171128-git, so as I thinked I changed anything, my version of gdb is fixed, and this was cause of the issue for me.

tknapstad, 2022-08-29:
I'm facing similar issues with the Zephyr SDK, presumably because they injected the Zephyr SDK version into the GDB version string:
```
$ /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb --version
GNU gdb (Zephyr SDK 0.15.0) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

Maybe this version check can be changed from an error to a warning?

haneefdm, 2022-08-29:
@tknapstad But then random things will fail

> Maybe this version check can be changed from an error to a warning?

It has been a warning  for almost a year and we were maintaining if-then-elses accounting for issues and missing/non-working features in gdb 8

There is a newer version (1.6.2) being pushed out by Microsoft Marketplace that will address the way Zephyr build outputs the version number. Btw, people should not be messing with gdb version string. A lot of tools check to see what version it is. The actual version number appears after `(...)` and we don't care what is in the parens. But after the parens, the gdb version was too short. So, in this case, Zephyr did not do anything wrong. We were being too aggressive. Corrected.

closed 2022-08-29 by haneefdm