Marus/cortex-debug #729: Announcement: Better memory viewer on the way

haneefdm, 2022-08-28:
Hi Folks,

We may have a better memory viewer soon. It is in preview right now and will be integrated into Cortex-Debug soon. I did it as a separate extension so it can work with all debuggers that need to deal with memory. Not limited to embedded Cortex-M. Even with it not integrated you can use it as is -- just a different command from the Command Palette

https://marketplace.visualstudio.com/items?itemName=haneefdm.memview
https://github.com/haneefdm/memview

I think it will be better than what we have right now, supports very large (but artificially limited to 1MB) memories while fetching only what is needed. The memory views also survive a VSCode restart.

You may ask "what about [Hex Edittor](https://marketplace.visualstudio.com/items?itemName=ms-vscode.hexeditor)" which is already integrated into VSCode (especially into Variable and Watch windows). While it may be great for files, it does not play nice with a debugger with memory contents always changing and [other issues documented here](https://github.com/Marus/cortex-debug/wiki/Memory-Viewer). If you like `Hex Editor`, by all means, keep at it. If you [see their issues list](https://github.com/microsoft/vscode-hexeditor/issues), working well with a debugger is not a priority

This document also captures my design intent and why I am doing it in the first place. I will move this document to the `memview` extension shortly.

If you like this, come help me with trying it out. More importantly, I am going to be asking Microsoft to integrate this type of memory view by providing a setting. Right now, for Variable and Watch windows, they hard coded it to the `Hex Editor` and we should have a choice. They also use a URI mechanism but not one that is friendly to other extensions.

haneefdm, 2022-08-29:
Come to think of it, `memview` is perhaps a bad name. There seems to be some other software named similar but it is not a VSCode extension. 

Suggestions welcome. In the market place, it is called `MemoryView`

hongshui3000, 2022-08-29:
>NOTE: Editing features are not yet ready for use. You can edit but the edits will not be saved to your program

This is a very important function, I hope this function can be supported in the future

haneefdm, 2022-08-31:
Hi people, please vote for native extension supporting a user settable memory viewer. 

https://github.com/microsoft/vscode/issues/155597

If they don't see 20 votes, they will just close it. I found out today, but that was posted a long time ago. Click on the thumbs up reaction for the first post.

Sadly, we have less than two weeks.

thegecko, 2022-09-01:
Sorry, I only see a link to this:

https://github.com/microsoft/vscode/issues/155597

closed 2022-11-19 by haneefdm