Marus/cortex-debug #731: SVD File Parsing Failed

EGOKM, 2022-08-30:
Hello,
I got the Error Message "Unable to parse SVD file /home/eg/DisplayR7/Config/max32625.svd: TypeError: Cannot read properties of undefined (reading 'map')" when I start debugging - I've checked "max32625.svd" with ARM's SVDConv - no Errors here.
So, I want to ask 1) can I check in more detail what the Problem is ? and 2) Is there any Solution probably for this ?
THX in advance, Enrico

EGOKM, 2022-08-30:
I forgot to tell the Version of the Plugin - its the latest 1.6.2, also tried 1.4.4 - same thing :-(

EGOKM, 2022-08-30:
vscode is:
Version: 1.70.2
Commit: e4503b30fc78200f846c62cf8091b76ff5547662
Date: 2022-08-16T05:36:25.715Z (2 wks ago)
Electron: 18.3.5
Chromium: 100.0.4896.160
Node.js: 16.13.2
V8: 10.0.139.17-electron.0
OS: Linux x64 5.0.9

haneefdm, 2022-08-30:
Only way to check what is wrong is to debug the extension itself. I can take a look but, I will need the SVD file. Can you compress and attach it?

You can also use Microsoft's version of SVD view. In your launch.json add an entry `svdPath` instead of `svdFile`. You can even try both at the same time. You have to install the following though.

https://marketplace.visualstudio.com/items?itemName=ms-vscode.vscode-embedded-tools

Our debugger and their tools work with each other.

EGOKM, 2022-09-01:
I've replied to the eMail from github - but it seems it doesnt work :-(, so pls see the SVD's attached here,  I've attached the "faulty" SVD ( max32625.svd ) , and a working one ( max32660.svd ) , beside this I will check Your other thoughts ;-),
[SVDs.zip](https://github.com/Marus/cortex-debug/files/9466990/SVDs.zip)

EGOKM, 2022-09-01:
Microsofts SVD View also didnt work:
[image]

haneefdm, 2022-09-01:
Please file an issue with them as well. We are hoping that in the future we can just rely on them for SVD which is too expensive for a one man show here to maintain.

I will take a look this weekend.

Could you do me a favor and look at this and vote for the issue on the MS Website if you agree.

https://github.com/Marus/cortex-debug/issues/729#issuecomment-1233293715

EGOKM, 2022-09-02:
upvoted right now ;-), THX for taking some time to look at this "special" SVD from MAXIM - the major Problem here is the nearly useless error message, its nearly same as telling me "Error 5" :-) , a more detailed Error Description would be fine, at least the Line # because SVD's are *very* large. Because ARM's svd checker shows me no Errors there is probably a Problem with the SVD - engine thinking MAXIM did made a correct SVD ;-)

haneefdm, 2022-09-02:
Found the problem. There were two files. One file was parsed fine. But the file `max32625.svd` had a whole bunch of empty enumerations which was not expected. Perhaps not illegal
```xml
              <enumeratedValues></enumeratedValues>
```
Please note that XML (which SVD is) is not really a language. It is data so our parser is not built like a compiler looking for problems and it probably made some bad assumptions 3-5 years ago. Normally it looks like this
```xml
       <enumeratedValues>
        <enumeratedValue>
         <name>en</name>
         <description>Slave clock stretching enabled.</description>
         <value>0</value>
        </enumeratedValue>
        <enumeratedValue>
         <name>dis</name>
         <description>Slave clock stretching disabled.</description>
         <value>1</value>
        </enumeratedValue>
       </enumeratedValues>
```
I already updated the code but we are not releasing anything until another issue is resolved. Perhaps this weekend.

Hopefully, you filed an issue with the MS Embedded tools as well. We would love to ditch our SVD support and let someone big like MS handle it.

EGOKM, 2022-09-02:
ok, THX a lot ;-), I will clean up the SVD from those empty strings and check again ...

EGOKM, 2022-09-02:
I will also open an issue to MS Embedded tools ;-), I've observed that installing both ( MS and Yours ) is not working very well - the ARM- Register View is not correct ( as it was before installing MS ) - but I need to verify this ...

EGOKM, 2022-09-02:
well, empty enums are not the Problem ;-), I've removed all that Stuff - not working, so I tried to validate the SVD here --> https://www.freeformatter.com/xml-validator-xsd.html - I've used the latest CMSIS-SVD.xsd ( attached ) and max32625.svd --- surprisingly it shows me many. many Errors (!), so, to verify it I tried max32660.svd --- ALL fine here :-),
that means at the End that I need to clean up the huge max32625.svd so that the Validator is happy with it , then try again -- this I will do also on Weekend because its seems to be a long lasting task :-)
[CMSIS-SVD.zip](https://github.com/Marus/cortex-debug/files/9477158/CMSIS-SVD.zip)

haneefdm, 2022-09-02:
Hmmm, it parsed just fine for me after I allowed empty enums to just be ignored. I will check again.

One was to debug this is to remove all peripherals and add them one at a time back in until you find the problem one. Then start removing stuff from that problem peripheral until find the offending XML-node and deduce a pattern/construct that is causing the issue.

Btw, that validator just says that the XML has the right syntax and matches the schema. Does not say that that it makes sense. Given ASRMs history on this, I don't even know if their xsd is correct..

I thought you were using this utility from Keil (unfortunately, Windows only so that is not easy for me)

https://www.keil.com/pack/doc/CMSIS/SVD/html/svd_SVDConv_pg.html
https://www.keil.com/pack/doc/CMSIS/SVD/html/svd_validate_file_pg.html

If they could just use the xml-validator, they would have. Again, that is an XML validator. Not and SVD validator.

EGOKM, 2022-09-05:
Hello again, well, I've checked max32625.svd against the Schema-File CMSIS-SVD.xsd as "xmllint --schema CMSIS-SVD.xsd max32625.svd" - it shows me a lot of errors :-(, so did use xmllint's error Messages tix them - mostly the Tag- Order was wrong ( like this "<description>whatever></description> was followed by <value>whatever</value>" - I interchanged those lines and xmllint was happy :-), so, the Schema-File obviously sets an order, I don't know why but it seems to be ...
After correcting all those Errors and some more ones that xmllint tells me - I've got an SVD that works smoothly :-), its parses fine now - if all those Stuff is finally correct I cant say but it works so far ,,,
Because SVDConv is Windows-Only I've used xmllint , but SVDConv shows me NO errors when I used the original SVD, thinking the SVD is ok but the Parser is not. So SVDConv did not do the right thing, xmllint does - at least it checks the svd against the xsd - this finally helped me a lot to solve the Problem - as it shows, the Problem was NOT vscode, NOT Your extension but a poorly written SVD from MAXIM (!) ...
Thx a lot for investigating Your time, pls find the correct SVD attached ...
[max32625.svd.zip](https://github.com/Marus/cortex-debug/files/9486909/max32625.svd.zip)

EGOKM, 2022-09-05:
oops, the online editor removed some important chars thinking its html :-(, here is the right statement:
... "xmllint's error Messages to fix them - mostly the Tag- Order was wrong ( like this \<description\>whatever\</description\> was followed by \<value\>whatever\</value\> - that is WRONG so I interchanged those lines and xmllint was happy :-)" ...

haneefdm, 2022-09-05:
I have a custom build you can try and give me feedback on. Especially look at the `CLKMAN` block to make sure all the registers are there. Also, check a couple of other blocks. I can't because I don't have this device/board.

https://github.com/Marus/cortex-debug/releases/edit/v1.6.3-pre3

Btw, when you type in markdown, use `code blocks` for code and it won't be touched, formatted or interpreted as HTML. Optionally, specify a language so it can color code as well.
```xml
<description>whatever</description>
```
https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

haneefdm, 2022-09-05:
Oops I had the wrong link earlier... 

https://github.com/Marus/cortex-debug/releases/tag/v1.6.3-pre4

It should say 1.6.3-pre4 if you check after install.

EGOKM, 2022-09-05:
ok, I did it ( installed v1.6.3-pre4) - it works so far ;-)
[image]

haneefdm, 2022-09-05:
I am going to close this. The fix will be in the next release in a day or so...

closed 2022-09-05 by EGOKM

skiphansen, 2023-12-07:
Just FYI: I ran into this same problem with the version 9/13/23 of Cortex-Debug and the svd file for and nrf52840.  

I made a local copy of the svd file and after changing all of the <dim>{hex value} and <dimIncrement>{hex value} to decimal it worked.  Apparently Segger had the same issue which is why I tried editing the file.

skiphansen, 2023-12-08:
@haneefdm Thanks for the pointer, I'll do just that.